wmbusmeters/wmbusmeters #844: Add new device - (ZRI) ZENNER International, Germany (0x6a49) type: Heat Cost Allocator (0x08) ver: 0xfd

henning84, 2023-02-08:
is it possible to add support for this device.?

(ZRI) ZENNER International, Germany (0x6a49) type: Heat Cost Allocator (0x08) ver: 0xfd

ID 82551121
key 296B4F5E07EEA2C95DFAE9195AD9484F

telegram

(meter) test: meter detection could not find driver for id: 82551121 mfct: (ZRI) ZENNER International, Germany (0x6a49) type: Heat Cost Allocator (0x08) ver: 0xfd
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|5E44496A21115582FD087A2CC050052F2F_0B6E030100426CDF2C4B6EFFFFFF82046CE1228B046E6200008D04EE132C3BFEFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF2F2F2F2F|+141
test    82551121        2023-02-08 10:47.00
telegram=|5E44496A21115582FD087A2DC050052F2F_0B6E030100426CDF2C4B6EFFFFFF82046CE1228B046E6200008D04EE132C3BFEFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF2F2F2F2F|+321
test    82551121        2023-02-08 10:50.00
telegram=|5E44496A21115582FD087A2EC050052F2F_0B6E030100426CDF2C4B6EFFFFFF82046CE1228B046E6200008D04EE132C3BFEFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF2F2F2F2F|+501
test    82551121        2023-02-08 10:53.00
telegram=|5E44496A21115582FD087A2FC050052F2F_0B6E030100426CDF2C4B6EFFFFFF82046CE1228B046E6200008D04EE132C3BFEFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF2F2F2F2F|+681
test    82551121        2023-02-08 10:56.00

BIBOLV, 2023-02-12:
Support added - please check.

henning84, 2023-02-12:
It was quick :)
Just checked and now it can read the device.
Thanks for super fast support and a great job you do with wmbusmeters.

Keep rocking.

closed 2023-03-16 by BIBOLV