wmbusmeters/wmbusmeters #845: Add new device - (DME)DIEHL Metering, Germany (0x11a5) type: Heat volume at inlet meter (0x0c) ver: 0x41

henning84, 2023-02-08:
### Type of request

New device support

### Meter description

(DME)DIEHL Metering, Germany (0x11a5) type: Heat volume at inlet meter (0x0c) ver: 0x41

### Logged telegrams from meter in decrypted format

```shell
henning@raspberrypi:~ $ wmbusmeters --logtelegrams rtlwmbus:c1 test auto 61842650 51728910E66D83F851728910E66D83F8
Started config rtlwmbus[00000001] listening on c1
(meter) test: meter detection could not find driver for id: 61842650 mfct: (DME)DIEHL Metering, Germany (0x11a5) type: Heat volume at inlet meter (0x0c) ver: 0x41
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|3E44A51150268461410C7AA80030052F2F_0C06105104000C13093835020B3B9401000C2B342600000A5A12060A5E91040AA61800004C0641460400426CFF21|+40
test    61842650        2023-02-08 11:04.37
telegram=|3E44A51150268461410C7AA80030052F2F_0C06105104000C13133835020B3B9801000C2B112700000A5A11060A5E91040AA61800004C0641460400426CFF21|+104
test    61842650        2023-02-08 11:05.41
telegram=|3E44A51150268461410C7AA80030052F2F_0C06105104000C13163835020B3B9501000C2B622600000A5A11060A5E89040AA61800004C0641460400426CFF21|+168
test    61842650        2023-02-08 11:06.45
telegram=|3E44A51150268461410C7AA80030052F2F_0C06115104000C13203835020B3B0102000C2B452700000A5A12060A5E90040AA61800004C0641460400426CFF21|+232
test    61842650        2023-02-08 11:07.49
telegram=|3E44A51150268461410C7AA80030052F2F_0C06115104000C13243835020B3B0502000C2B512800000A5A11060A5E85040AA61800004C0641460400426CFF21|+297
test    61842650        2023-02-08 11:08.54
```

BIBOLV, 2023-02-12:
Support added - please check.

henning84, 2023-02-12:
It was quick :)
Just checked and now it can read the device.
Thanks for super fast support and a great job you do with wmbusmeters.

Keep rocking.

closed 2023-03-16 by BIBOLV