wmbusmeters/wmbusmeters #846: cannot install

qwert4u, 2023-02-08:
### Type of request

Bug

### HAOS version

Home Assistant 2023.2.3

### wmbusmeters version

1.11.0

### Your message goes here

when I click on install it takes a while, then the following message appears...:

"Add-on konnte nicht installiert werden
Unknown error, see supervisor logs"

when i restart the raspberry pi 3b then the addon is installed, but when i start the addon i get the following message

"Starten des Add-on fehlgeschlagen
Image cada3e3b/aarch64-addon-wmbusmeters:1.11.0 does not exist for addon_cada3e3b_wmbusmeters"

i have tried several times... once I had the following message see picture:
[image]

DNS server I tested 8.8.8.8 and 9.9.9.9 (there was also an error) but it did not work.

BIBOLV, 2023-02-10:
It is more of HA issue rather than wmbusmeters. On slow hardware sometime building of addon fails due to different reasons. Never the less it is in plans to create prebuild addon containers so that installation would be much faster and there would be no need for them to be build locally. 
It will take some time, so you will need to wait for that, but in meantime you could try installation again few times more when there is less load on system and it might build.

mellaiz, 2023-02-28:
same here... it just stuck on installing

BIBOLV, 2023-03-10:
Please try https://github.com/wmbusmeters/wmbusmeters-ha-addon

thecem, 2023-03-11:
not working:

[image]

for the meters problem I created a PR https://github.com/wmbusmeters/wmbusmeters/pull/910 & https://github.com/wmbusmeters/wmbusmeters-ha-addon/pull/5 pls merge

BIBOLV, 2023-03-11:
Please try to install edge version. Stable version will be built on next wmbusmeters major release.

BIBOLV, 2023-03-13:
HA add-on has moved to separate repo - https://github.com/wmbusmeters/wmbusmeters-ha-addon
It has now pre-built container images and config interface changes - please test and close this issue if no other questions.

closed 2023-06-12 by github-actions[bot]