wmbusmeters/wmbusmeters #849: Wmbusmeters not starting

VoyteckPL, 2023-02-10:
### Type of request

Bug

### OS version

Home assistant addon

### wmbusmeters version

Latest

### Your message goes here

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[05:11:42] INFO: Syncing wmbusmeters configuration ...
[05:11:42] INFO: Registering meters ...
[05:11:42] INFO: Adding meter-0001 ...
[05:11:42] INFO: Generating MQTT configuration ... 
[05:11:44] INFO: Broker 192.168.1.9 will be used.
MQTT Discovery ...
 Adding meter: wodomierz ...
  Add/update topic: homeassistant/sensor/wmbusmeters/02360965_rssi_dbm/config
  Add/update topic: homeassistant/sensor/wmbusmeters/02360965_timestamp/config
  Add/update topic: homeassistant/sensor/wmbusmeters/02360965_total_m3/config
MQTT Discovery cleanup...
[05:11:49] INFO: Running wmbusmeters ...
(config) "device" "rtlwmbus:CMD(rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a)"
(wmbusmeters) version: 1.11.0-67-g7f0ff31
(config) store meter files in: "/config/wmbusmeters/logs/meter_readings/"
(config) using device: rtlwmbus:CMD(rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a) 
(config) number of meters: 1
(serial) expecting devices to work
(serial) expecting devices to work
(lookup) with cmd "rtlwmbus:CMD(rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a)"
(main) opening rtlwmbus:CMD(rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a)
Started config rtlwmbus listening on t1 using CMD(rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a)
(rtlwmbus) using command: rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a
(rtlwmbus) opening cmd_0
(bgshell) exec background "/bin/sh"
(serial) EVENT thread interrupted
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a"
(serialcmd) opened /bin/sh pid 202 fd 3 (rtlwmbus)
(main) regular reset of rtlwmbus  using CMD(rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a) will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) MAIN thread interrupted
(bgshell) 202 exited with return code 0
(serial) closing non working fd=3 "cmd_0"
(wmbus) disconnected cmd_0 rtlwmbus
(serialcmd) closed /bin/sh pid=202 fd=3 (rtlwmbus)
(serial) no devices working emergency exit!
(serial) stopping manager
(serial) non working devices found, exiting.
(serial) event loop stopped!
(serial) TIMER thread interrupted
(wmbus) deleted rtlwmbus
/run.sh: line 64:   198 Segmentation fault      (core dumped) /wmbusmeters/wmbusmeters --useconfig=$CONFIG_DATA_PATH
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

weetmuts, 2023-02-10:
Can you build wmbusmeters with DEBUG=true ? and rerun, then we will get a stack trace from the crash in wmbusmeters?

VoyteckPL, 2023-02-10:
My config in HA:

data_path: /config/wmbusmeters
enable_mqtt_discovery: true
conf: >-
  loglevel=debug

  debug=true

  device=rtlwmbus:CMD(rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p
  s -a)

  listento=t1

  logtelegrams=false

  format=json

  logfile=/dev/stdout

  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"

  meterfiles=/config/wmbusmeters/logs/meter_readings/

  meterfilesaction=overwrite
meters:
  - |-
    name=wodomierz
    driver=apator162
    id=xxxxxxxx
    key=00000000000000000000000000000000
mqtt:
  host: 192.168.1.x
  port: 1883
  user: xxxxx
  password: xxxxx

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[08:06:17] INFO: Syncing wmbusmeters configuration ...
[08:06:18] INFO: Registering meters ...
[08:06:19] INFO: Adding meter-0001 ...
[08:06:19] INFO: Generating MQTT configuration ... 
[08:06:20] INFO: Broker 192.168.1.9 will be used.
MQTT Discovery ...
 Adding meter: wodomierz ...
  Add/update topic: homeassistant/sensor/wmbusmeters/02360965_rssi_dbm/config
  Add/update topic: homeassistant/sensor/wmbusmeters/02360965_timestamp/config
  Add/update topic: homeassistant/sensor/wmbusmeters/02360965_total_m3/config
MQTT Discovery cleanup...
[08:06:37] INFO: Running wmbusmeters ...
(config) "debug" "true"
(wmbusmeters) version: 1.11.0-67-g7f0ff31
(config) store meter files in: "/config/wmbusmeters/logs/meter_readings/"
(config) using device: rtlwmbus:CMD(rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a) 
(config) number of meters: 1
(serial) expecting devices to work
(serial) expecting devices to work
(lookup) with cmd "rtlwmbus:CMD(rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a)"
(main) opening rtlwmbus:CMD(rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a)
Started config rtlwmbus listening on t1 using CMD(rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a)
(rtlwmbus) using command: rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a
(rtlwmbus) opening cmd_0
(bgshell) exec background "/bin/sh"
(serial) EVENT thread interrupted
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a"
(serialcmd) opened /bin/sh pid 203 fd 3 (rtlwmbus)
(main) regular reset of rtlwmbus  using CMD(rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a) will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) MAIN thread interrupted
(bgshell) 203 exited with return code 0
(serial) closing non working fd=3 "cmd_0"
(wmbus) disconnected cmd_0 rtlwmbus
(serialcmd) closed /bin/sh pid=203 fd=3 (rtlwmbus)
(serial) no devices working emergency exit!
(serial) stopping manager
(serial) MAIN thread interrupted
(serial) TIMER thread interrupted
(serial) non working devices found, exiting.
(serial) event loop stopped!
(wmbus) deleted rtlwmbus
/run.sh: line 64:   199 Segmentation fault      (core dumped) /wmbusmeters/wmbusmeters --useconfig=$CONFIG_DATA_PATH
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

weetmuts, 2023-02-10:
No, I meant when wmbusmeters is built when the HA-addon is installed, can you go into the build directory and build it with DEBUG=true?  

@BIBOLV do you know how to generate a DEBUG build with the HA continer setup?

BIBOLV, 2023-02-10:
AFAIK it is not possible straight forward. Would need to investigate what are possibilities to achieve that.  
Currently only idea is to create additional debug version of addon.

thecem, 2023-02-10:
you have to edit, the [https://github.com/weetmuts/wmbusmeters/blob/master/ha-addon/Dockerfile](https://github.com/weetmuts/wmbusmeters/blob/master/ha-addon/Dockerfile)
But, this is for all, if @VoyteckPL would clone it, add it to addon repo, change some links and add it, so this would be an option but not easy.
Other way around is to have a second add-on here with the DEBUG=true option and as dev version.

VoyteckPL, 2023-02-10:
Guys, I don't understand. I just use it as addon in HA... Can you help?

weetmuts, 2023-02-10:
Yes @VoyteckPL it seems like a crash inside wmbusmeters and we want to find a fix it asap.
However we do not know where it happens. A DEBUG=true build of the wmbusmeters software will
tell us. However it seems like it is related to the fact that it cannot start rtl_sdr/rtl_wmbus....

Can you test rtls_dr/rtl_wmbus separately? I.e. run the command:
rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a
from the command line?

VoyteckPL, 2023-02-10:
I know my answer won't make you happy :/ Unfortunately I have no access to command line

thecem, 2023-02-10:
Install portainer as an add-on. Find the wmbusmeters container, then open the cmd line.

this really graphic menu.

be sure to have a starting minimal config, even with not working radio.
The connect to the cmd and run above command.

Add-on repo:
https://github.com/alexbelgium/hassio-addons
Portainer:
https://github.com/alexbelgium/hassio-addons/tree/master/portainer

VoyteckPL, 2023-02-10:
[image]
Nothing happends :/

BIBOLV, 2023-02-10:
@thecem - great idea how to connect to cmd, could you create manual in https://github.com/wmbusmeters/wmbusmeters-wiki how to get in cmd from both console and graphical interface? That would benefit both docker container and ha-addon users.
@VoyteckPL - what hardware do you use for home assistant? 
Try running `rtl_test` and post what is output from that. If device is found then try `rtl_sdr -f 868.95M -s 1600000 -` and you should see lot of data in terminal. If it is so then we can consider that rtl_sdr part is working at least at high level.

thecem, 2023-02-11:
Will do when I have time…

VoyteckPL, 2023-02-16:
I updated addon to latest version and I still have problem:

`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[07:43:25] INFO: CONFIG_CONF ...
[07:43:25] INFO: {"loglevel":"normal","device":"auto:t1","donotprobe":"/dev/ttyAMA0","logtelegrams":false,"format":"json","logfile":"/dev/stdout","shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[07:43:25] INFO: CONFIG_METERS ...
[07:43:25] INFO: {"name":"wodomierz","driver":"apator162","id":"02360965","key":"00000000000000000000000000000000"}
[07:43:25] INFO: Syncing wmbusmeters configuration ...
[07:43:25] INFO: Registering meters ...
[07:43:25] INFO: Adding meter-0001 ...
**parse error: Invalid numeric literal at line 2, column 0**
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped`

BIBOLV, 2023-02-16:
This is another issue, global ha-addon issue that we are fixing currently. Will let you know when it will be fixed and you will be able to continue your tests.

VoyteckPL, 2023-02-17:
Fixed.You can close. :)

closed 2023-02-17 by BIBOLV