wmbusmeters/wmbusmeters #851: How decode compressed telegrams from one-shot CLI ?

rhuss, 2023-02-13:
### Type of request

Question

### OS version

Raspbian

### wmbusmeters version

1.11.0-72-ga2925a9

### Your message goes here

Im tyring to decode telegrams that I receive with `rtl_433` from the CLI like in 

```
./wmbusmeters 1f442d2c508248631b168d20fb21ad902415baec3c88445efc4683d6be6742dc watermeter multical21 63455555 12345679...
479
(meter) newly created meter (watermeter 63455555 multical21) did not handle telegram!
```

and analyze for the same telegram yields

```
./wmbusmeters --analyze=auto:12345679... 1f442d2c508248631b168d20fb21ad902415baec3c88445efc4683d6be6742dc

Auto driver  : multical21
Best driver  : unknown 00/00
Using driver : multical21 00/00
000   : 1f length (31 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 2d2c dll-mfct (KAM)
004   : 50824863 dll-id (63455555)
008   : 1b dll-version
009   : 16 dll-type (Cold water meter)
010   : 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
011   : 20 ell-cc (slow_resp sync)
012   : fb ell-acc
013   : 21ad9024 sn (AES_CTR)
017   : 80b7 payload crc (calculated 80b7 OK)
019   : 79 tpl-ci-field (EN 13757-3 Application Layer with Compact frame (no tplh))
020   : 0e7c format signature (unknown)
022 CC: 31A1000081590B001201 compressed and signature unknown

{
    "media":"cold water",
    "meter":"multical21",
    "name":"",
    "id":"63455555",
    "timestamp":"1970-01-01T00:00:00Z"
}
```

From running with debug I understand that this is a compressed frame that only can be uncompressed when a full frame has been received:

```
(wmbus) ignoring compressed telegram since format signature hash 0x7c0e is yet unknown.
     this is not a problem, since you only need wait for at most 8 telegrams
     (8*16 seconds) until an full length telegram arrives and then we know
     the format giving this hash and start decoding the telegrams properly.
Newly created meter ( 63488250 ultrimis) did not handle telegram!
```

And indeed, if I wait for that full length telegram, I can decode it.

How can I deal with the compressed telegrams when I use `wmbusmeters` from the CLI afresh in an external process to decrypt those telegrams? I mean, when running as a daemon, I can keep the state and afaiu the compressed telegrams coming in after a full telegram can be decoded. But how could I share the state information needed for decrypting that format between subsequent calls to `wmbus`?

Thanks a lot for this great software and the great dedication that you put into it. Highly appreciated!

weetmuts, 2023-02-13:
You can temporarily in your build add the format signature (which you can see in the --debug log)
to the function findFormatBytesFromKnownMeterSignatures.

I am working on the new driver format which loads the drivers from config files instead of keeping them compiled in source.

I will add something that will automatically pick the compressed formats from the example telegrams in the driver
so that it is possible to immediately decode a telegram as long as the test data is complete.

rhuss, 2023-02-13:
Thanks, let me try that!

rhuss, 2023-02-13:
@weetmuts, thanks for the hint, it works like a charme. See the attached PR (you might consider merging until you have the "full" solution.

PR #853 (2023-03-03): Add 0x7c0e hash to list of known meter format signatures
Adds an additional hash, format pair to the list of hardcoded formats.

```
./build/wmbusmeters --analyze=auto:12345678... 1f442d2c508248631b168d202eb1a99024efd3759b107ec1105a63b0766a5c60
Auto driver  : multical21
Best driver  : flowiq2200 08/08
Using driver : multical21 00/00
000   : 1f length (31 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 2d2c dll-mfct (KAM)
004   : 50824863 dll-id (63488250)
008   : 1b dll-version
009   : 16 dll-type (Cold water meter)
010   : 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
011   : 20  [... cut]
closed 2023-03-03 by weetmuts