wmbusmeters/wmbusmeters #855: Add-on store doesn't contain repository

dula34, 2023-02-14:
### Type of request

Bug

### HAOS version

2023.2.4

### wmbusmeters version

Latest

### Your message goes here

I have add-on repository in repository list, but add-on store doesn't contain section with this add-on

thecem, 2023-02-15:
if you have this:

[image]

the hit update and then clear the cache. Reload the page with strg+F5.

This is a home assistant problem, not related to wmbusmeters.

dula34, 2023-02-15:
[image]
I tried without success.
I also add another repository and worked, but this one not

thecem, 2023-02-15:
I will test this today night, if this is working on a new system.

dula34, 2023-02-15:
thx, btw I did clean install latest version HA on RPI3 with same result.

chpego, 2023-02-15:
[image]

dula34, 2023-02-15:
It works after merge PR, thanks all for quick fix

closed 2023-02-15 by dula34