wmbusmeters/wmbusmeters #859: parse error: Invalid numeric literal

dusan-ivanco, 2023-02-15:
### Type of request

Bug

### HAOS version

2023.2.4

### wmbusmeters version

1.11.0-1

### Your message goes here

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[22:00:54] INFO: CONFIG_CONF ...
[22:00:54] INFO: {"loglevel":"normal","device":"auto:t1","donotprobe":"/dev/ttyAMA0","logtelegrams":true,"format":"json","logfile":"/dev/stdout","shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[22:00:54] INFO: CONFIG_METERS ...
[22:00:54] INFO: {"name":"MainWater","driver":"izar","id":"13100924","key":"NOKEY"}
[22:00:54] INFO: Syncing wmbusmeters configuration ...
[22:00:55] INFO: Registering meters ...
[22:00:55] INFO: Adding meter-0001 ...
parse error: Invalid numeric literal at line 2, column 0
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

Murcin86, 2023-02-15:
I`ve got exactly the same problem... :) and trying to find since 4 hour any information in internet and couple minutes ago I decided to created issue here... you`re first :)

thecem, 2023-02-15:
Could you post your config as yaml?

thecem, 2023-02-15:
Like this:

```YAML
data_path: /config/wmbusmeters
enable_mqtt_discovery: true
conf:
  loglevel: normal
  device: >-
    /dev/serial/by-id/usb-Silicon_Labs_WiMOD_iM871A-usb_012345678-if00-port0:im871a:c1,t1
  donotprobe: /dev/ttyAMA0
  logtelegrams: true
  format: json
  logfile: /dev/stdout
  shell: /wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"
meters:
  - name: ODGMulti303
    driver: kamheat
    id: "82788990"
    key: 123075786777676676887888778878877
  - name: EGMulti303
    driver: kamheat
    id: "82813990"
    key: 123075786777676676887888778878877
mqtt: {}

```

convicte, 2023-02-15:
@thecem: 
https://github.com/weetmuts/wmbusmeters/issues/860

Here is my config that even with only one meter gives the same error.

I am guessing people that have `NOKEY` and alphanumerical `id` are suffering from this problem.

dusan-ivanco, 2023-02-16:
```
data_path: /config/wmbusmeters
enable_mqtt_discovery: true
conf:
  loglevel: normal
  device: auto:t1
  donotprobe: /dev/ttyAMA0
  logtelegrams: true
  format: json
  logfile: /dev/stdout
  shell: /wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"
meters:
  name: MainWater
  driver: izar
  id: "13100924"
  key: NOKEY
mqtt: {}
```
i cannot add more meters - if i have it as array i get:
```
Failed to save add-on configuration, Invalid dict for option 'meters' in Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters). Got {'data_path': '/config/wmbusmeters', 'enable_mqtt_discovery': True, 'conf': {'loglevel': 'normal', 'device': 'auto:t1', 'donotprobe': '/dev/ttyAMA0', 'logtelegrams': True, 'format': 'json', 'logfile': '/dev/stdout', 'shell': '/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"'}, 'meters': [{'name': 'MainWater', 'driver': 'izar', 'id': '13100924', 'key': 'NOKEY'}], 'mqtt': {}}
```
thank you very much

thecem, 2023-02-16:
I changed with PR https://github.com/weetmuts/wmbusmeters/pull/861 key to optional pls delete the key parameter and double check. 

@convicte @dusan-ivanco : 
Could you post the file `config/wmbusmeters/etc/wmbusmeters.d/meter-0001`

@KrzysztofHajdamowicz could you double check if the alphanumerics key´s and id´s will be right translatet in to the meter Ffile?

KrzysztofHajdamowicz, 2023-02-16:
> ```
> data_path: /config/wmbusmeters
> enable_mqtt_discovery: true
> conf:
>   loglevel: normal
>   device: auto:t1
>   donotprobe: /dev/ttyAMA0
>   logtelegrams: true
>   format: json
>   logfile: /dev/stdout [...]

Your meters section doesn't follow schema. Your config declared single object, where array of objects is required.
Look at the screenshot
[image]

thecem, 2023-02-16:
try:

```YAML
data_path: /config/wmbusmeters
enable_mqtt_discovery: true
conf:
  loglevel: normal
  device: auto:t1
  donotprobe: /dev/ttyAMA0
  logtelegrams: true
  format: json
  logfile: /dev/stdout
  shell: /wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"
meters:
  - name: MainWater
    driver: izar
    id: "13100924"
    key: NOKEY
mqtt: {}
```

dusan-ivanco, 2023-02-16:
> I changed with PR #861 key to optional pls delete the key parameter and double check.
> 
> @convicte @dusan-ivanco : Could you post the file `config/wmbusmeters/etc/wmbusmeters.d/meter-0001`
> 
> @KrzysztofHajdamowicz could you double check if the alphanumerics key´s and id´s will be right translatet in to the meter file?

meter-0001 is empty

dusan-ivanco, 2023-02-16:
> ```yaml
> data_path: /config/wmbusmeters
> enable_mqtt_discovery: true
> conf:
>   loglevel: normal
>   device: auto:t1
>   donotprobe: /dev/ttyAMA0
>   logtelegrams: true
>   format: json
>   logfile: /dev/stdout [...]

this doesnt work

```json
Invalid dict for option 'meters' in Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters). Got {'data_path': '/config/wmbusmeters', 'enable_mqtt_discovery': True, 'conf': {'loglevel': 'normal', 'device': 'auto:t1', 'donotprobe': '/dev/ttyAMA0', 'logtelegrams': True, 'format': 'json', 'logfile': '/dev/stdout', 'shell': '/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"'}, 'meters': [{'name': 'MainWater', 'driver': 'izar', 'id': '13100924', 'key': 'NOKEY'}], 'mqtt': {}}
```

thecem, 2023-02-16:
and with:

```YAML
meters:
  - name: MainWater
    driver: izar
    id: "13100924"
    key: NOKEY
```

or 

```YAML
meters:
  name: MainWater
  driver: izar
  id: "13100924"
  key: 123075786777676676887888778878877
```

dusan-ivanco, 2023-02-16:
> and with:
> 
> ```yaml
> meters:
>   - name: MainWater
>     driver: izar
>     id: "13100924"
>     key: NOKEY
> ```
> 
> or
> 
> ```yaml
> meters:
>   name: MainWater
>   driver: izar
>   id: "13100924"
>   key: 123075786777676676887888778878877
> ```

nothink from this is working

Murcin86, 2023-02-16:
here is my errors - maybe that helps:
`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[11:12:44] INFO: CONFIG_CONF ...
[11:12:44] INFO: {"loglevel":"debug","device":"rtlwmbus:868M:t1","donotprobe":"all","logtelegrams":false,"format":"json","logfile":"/config/wmbusmeters/wmbusmeters.log","shell":"/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME \"$METER_JSON\""}
[11:12:44] INFO: CONFIG_METERS ...
[11:12:44] INFO: {"name":"wodomierz","driver":"iperl","id":"21066284","key":"NOKEY"}
[11:12:44] INFO: Syncing wmbusmeters configuration ...
[11:12:44] INFO: Registering meters ...
[11:12:44] INFO: Adding meter-0001 ...
parse error: Invalid numeric literal at line 2, column 0
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped`

here is configuration in YAML:
`data_path: /config/wmbusmeters
enable_mqtt_discovery: false
conf:
  loglevel: debug
  device: rtlwmbus:868M:t1
  donotprobe: all
  logtelegrams: false
  format: json
  logfile: /config/wmbusmeters/wmbusmeters.log
  shell: /wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters:
  name: wodomierz
  driver: iperl
  id: "21066284"
  key: NOKEY
mqtt: {}
`
even when I set default pass there is error durning safe:
[image]

VoyteckPL, 2023-02-16:
Same problem here...

KrzysztofHajdamowicz, 2023-02-16:
I've created #863 
Schema that went into this repo was different I was using as my test build.
https://github.com/KrzysztofHajdamowicz/wmbusmeters/blob/master/ha-addon/config.json

BIBOLV, 2023-02-16:
Fix was pushed with https://github.com/weetmuts/wmbusmeters/pull/864 - please confirm that everything works now

BIBOLV, 2023-02-16:
Firstly please check that you are running latest version of addon - 1.11.0-2
Then compare with test config I used and it worked for me:
```
data_path: /config/wmbusmeters
enable_mqtt_discovery: false
conf:
  loglevel: normal
  device: auto:t1
  donotprobe: /dev/ttyAMA0
  logtelegrams: false
  format: json
  logfile: /dev/stdout
  shell: /wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"
meters:
  - name: water_meter
    driver: izar
    id: 4141d24a
    key: NOKEY
  - name: water_meter2
    driver: izar
    id: 4141d242
    key: NOKEY
mqtt: {}

```

Murcin86, 2023-02-16:
yes - version is latest (update couple minutes ago)
[image]

but there is "-" sign needed :)
[image]

so now looks working !! :)

dusan-ivanco, 2023-02-17:
with version 1.11.0-2 is it working now - thank you for you great work

closed 2023-02-17 by BIBOLV