wmbusmeters/wmbusmeters #860: Unable to set up multiple meters or documentation incorrect/inconsistent

convicte, 2023-02-15:
### Type of request

Question

### HAOS version

2023.02.04

### wmbusmeters version

1.11.0-1

### Your message goes here

@KrzysztofHajdamowicz @BIBOLV 

Since you've worked together on https://github.com/weetmuts/wmbusmeters/pull/838 I allowed myself to seek your assistance:"

For the life of me I cannot create a correct configuration under the new scheme with two meters having `NOKEY`.
```yaml
data_path: /config/wmbusmeters
enable_mqtt_discovery: true
conf:
  loglevel: normal
  device: rtlwmbus:CMD(/usr/bin/nc -lku 9011)
  donotprobe: /dev/ttyUSB0
  logtelegrams: false
  format: json
  logfile: /dev/stdout
  shell: /wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"
  meterfiles: /config/wmbusmeters/logs/meter_readings
meters:
  - name: water_meter
    driver: izar
    id: 4141d24a
    key: NOKEY
  - name: garden_water_meter
    driver: izar
    id: 2141503d
    key: NOKEY
mqtt: {}
``` 
I've tried both with passing the `id` and `key` as string and not.
This is the exact layout indicated here: https://github.com/weetmuts/wmbusmeters/blob/master/ha-addon/DOCS.md

convicte, 2023-02-15:
That's the error output:

```Failed to save add-on configuration, Invalid dict for option 'meters' in Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters). Got {'data_path': '/config/wmbusmeters', 'enable_mqtt_discovery': True, 'conf': {'loglevel': 'normal', 'device': 'rtlwmbus:CMD(/usr/bin/nc -lku 9011)', 'donotprobe': '/dev/ttyUSB0', 'logtelegrams': False, 'format': 'json', 'logfile': '/dev/stdout', 'shell': '/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"', 'alarmshell': '/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$ALARM_TYPE $ALARM_MESSAGE"', 'meterfiles': '/config/wmbusmeters/logs/meter_readings', 'meterfiles...
```

BIBOLV, 2023-02-15:
Could you add quotation marks for id and key of second meter and try?
```
meters:
  - name: water_meter
    driver: izar
    id: "4141d24a"
    key: "NOKEY"
  - name: garden_water_meter
    driver: izar
    id: "2141503d"
    key: "NOKEY"
```

convicte, 2023-02-15:
Tried all possible combinations! 
With and without on both and either - no dice.

BIBOLV, 2023-02-15:
@KrzysztofHajdamowicz - what is your take on this issue?

BIBOLV, 2023-02-15:
Found and started my dusty old test HA instance and my finding so far is that current config.json can work only with one meter.
When following config is used then there is no error and I could not find a way to add more:
```
meters:
  name: water_meter
  driver: izar
  id: 4141d24a
  key: NOKEY
```

convicte, 2023-02-15:
Thank you for checking!

But I have to say that not exactly 'work' - because this https://github.com/weetmuts/wmbusmeters/issues/859

KrzysztofHajdamowicz, 2023-02-16:
Ok, I analyzed config.json file and came to conclusion that schema has to change a little. `key` key has to be optional. I will figure out how to do it.

BIBOLV, 2023-02-16:
@KrzysztofHajdamowicz - @thecem created PR https://github.com/weetmuts/wmbusmeters/pull/861 for this. Can you review and confirm that that would fix the issue?

thecem, 2023-02-16:
@KrzysztofHajdamowicz if you add a "?" to "str" so "str?" will be an optional schema.

BIBOLV, 2023-02-16:
Fix was pushed with https://github.com/weetmuts/wmbusmeters/pull/864 - please confirm that everything works now

chpego, 2023-02-17:
> Fix was pushed with #864 - please confirm that everything works now

LGTM !

casper255, 2023-02-17:
The latest version of the add-on is installed.  It is not possible to save the configuration with multiple meters.  Same message as above.

chpego, 2023-02-17:
> The latest version of the add-on is installed. It is not possible to save the configuration with multiple meters. Same message as above.

can you post your configuration?

casper255, 2023-02-17:
This is a working configuration on version 1.11.0

data_path: /config/wmbusmeters
enable_mqtt_discovery: false
conf: |-
  loglevel=normal
  device=auto:c1
  meterfiles=/config/wmbusmeters/logs/meter_readings/
  meterfilesaction=overwrite
  logtelegrams=true
  format=json
  logfile=/dev/stdout
  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters:
  - |-
    name=grzejnik_salon_imp
    driver=fhkvdataiii
    id=05743828
  - |-
    name=grzejnik_kuchnia_imp
    driver=fhkvdataiii
    id=05743835
  - |-
    name=grzejnik_jadalnia_imp
    driver=fhkvdataiii
    id=05743829
  - |-
    name=grzejnik_julia_imp
    driver=fhkvdataiii
    id=05743794
  - |-
    name=grzejnik_sypialnia_imp
    driver=fhkvdataiii
    id=05743837
mqtt:
  host: 192.168.0.xxx
  port: 1883
  user: xxx
  password: xxx

The configuration I'm trying to save on version 1.11.0-2

data_path: /config/wmbusmeters
enable_mqtt_discovery: false
conf:
  loglevel: normal
  device: auto:c1
  meterfiles: /config/wmbusmeters/logs/meter_readings/
  meterfilesaction: overwrite
  logtelegrams: true
  format: json
  logfile: /dev/stdout
  shell: /wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters:
  - name: grzejnik_salon_imp
    driver: fhkvdataiii
    id: 05743828
  - name: grzejnik_kuchnia_imp
    driver: fhkvdataiii
    id: 05743835
  - name: grzejnik_jadalnia_imp
    driver: fhkvdataiii
    id: 05743829
  - name: grzejnik_julia_imp
    driver: fhkvdataiii
    id: 05743794
  - name: grzejnik_sypialnia_imp
    driver: fhkvdataiii
    id: 05743837
mqtt:
  host: 192.168.0.xxx
  port: 1883
  user: xxx
  password: xxx

thecem, 2023-02-17:
could you do us a favor and post it in correct way:

Mark the config
Click "<>"
Write after ''' YAML

```YAML
data_path: /config/wmbusmeters
enable_mqtt_discovery: false
conf:
loglevel: normal
device: auto:c1
meterfiles: /config/wmbusmeters/logs/meter_readings/
meterfilesaction: overwrite
logtelegrams: true
format: json
logfile: /dev/stdout
shell: /wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters:

    name: grzejnik_salon_imp
    driver: fhkvdataiii
    id: 05743828
    name: grzejnik_kuchnia_imp
    driver: fhkvdataiii
    id: 05743835
    name: grzejnik_jadalnia_imp
    driver: fhkvdataiii
    id: 05743829
    name: grzejnik_julia_imp
    driver: fhkvdataiii
    id: 05743794
    name: grzejnik_sypialnia_imp
    driver: fhkvdataiii
    id: 05743837
    mqtt:
    host: 192.168.0.xxx
    port: 1883
    user: xxx
    passwor
```d: xxx

casper255, 2023-02-17:
```
data_path: /config/wmbusmeters
enable_mqtt_discovery: false
conf: |-
  loglevel=normal
  device=auto:c1
  meterfiles=/config/wmbusmeters/logs/meter_readings/
  meterfilesaction=overwrite
  logtelegrams=true
  format=json
  logfile=/dev/stdout
  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters:
  - |-
    name=grzejnik_salon_imp
    driver=fhkvdataiii
    id=05743828
  - |-
    name=grzejnik_kuchnia_imp
    driver=fhkvdataiii
    id=05743835
  - |-
    name=grzejnik_jadalnia_imp
    driver=fhkvdataiii
    id=05743829
  - |-
    name=grzejnik_julia_imp
    driver=fhkvdataiii
    id=05743794
  - |-
    name=grzejnik_sypialnia_imp
    driver=fhkvdataiii
    id=05743837
mqtt:
  host: 192.168.0.187
  port: 1883
  user: mqtt_user
  password: misiek255

```

casper255, 2023-02-17:
```
data_path: /config/wmbusmeters
enable_mqtt_discovery: false
conf:
  loglevel: normal
  device: auto:c1
  meterfiles: /config/wmbusmeters/logs/meter_readings/
  meterfilesaction: overwrite
  logtelegrams: true
  format: json
  logfile: /dev/stdout
  shell: /wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters:
  - name: grzejnik_salon_imp
    driver: fhkvdataiii
    id: 05743828
  - name: grzejnik_kuchnia_imp
    driver: fhkvdataiii
    id: 05743835
  - name: grzejnik_jadalnia_imp
    driver: fhkvdataiii
    id: 05743829
  - name: grzejnik_julia_imp
    driver: fhkvdataiii
    id: 05743794
  - name: grzejnik_sypialnia_imp
    driver: fhkvdataiii
    id: 05743837
mqtt:
  host: 192.168.0.187
  port: 1883
  user: mqtt_user
  password: misiek255

```

chpego, 2023-02-17:
> id: 05743829

you have to put double quotations for the id values

convicte, 2023-02-17:
@casper255 
Are you certain this MQTT indent is working for you in 1.11.0? It looks to be set up incorrectly.

Also when you copy any code into git, please use three slashes ` before and after, to indicate to GIT it's code.
It will make it easier for people to help you.
You can also add the language after the slashes (like this 
[image]
to make it even better).
It will result in this:
```yaml
YOUR CODE HERE
```

As for your config, please try this:

```yaml
data_path: /config/wmbusmeters
enable_mqtt_discovery: false
conf:
   loglevel: normal
   device: auto:c1
   meterfiles: /config/wmbusmeters/logs/meter_readings/
   meterfilesaction: overwrite
   logtelegrams: true
   format: json
   logfile: /dev/stdout
   shell: /wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters:
   name: grzejnik_salon_imp
   driver: fhkvdataiii
   id: "05743828"
   name: grzejnik_kuchnia_imp
   driver: fhkvdataiii
   id: "05743835"
   name: grzejnik_jadalnia_imp
   driver: fhkvdataiii
   id: "05743829"
   name: grzejnik_julia_imp
   driver: fhkvdataiii
   id: "05743794"
   name: grzejnik_sypialnia_imp
   driver: fhkvdataiii
   id: "05743837"
mqtt:
   host: 192.168.0.xxx
   port: 1883
   user: xxx
   password: xxx
 ```
In addition, if you are using Mosquito  you do not need to indicate the MQTT section just `{}` will be fine

KrzysztofHajdamowicz, 2023-02-17:
> ```yaml
> meters:
>    name: grzejnik_salon_imp
>    driver: fhkvdataiii
>    id: "05743828"
>    name: grzejnik_kuchnia_imp
>    driver: fhkvdataiii
>    id: "05743835"
>    name: grzejnik_jadalnia_imp
>    driver: fhkvdataiii
>    id: "05743829"
>    name: grzejnik_julia_imp
>    driver: fhkvdataiii
>    id: "05743794"
>    name: grzejnik_sypialnia_imp
>    driver: fhkvdataiii
>    id: "05743837"
> mqtt:
>    host: 192.168.0.xxx
>    port: 1883
>    user: xxx
>    password: xxx
> ```

Correct format is:
```YAML
meters:
   - name: grzejnik_salon_imp
     driver: fhkvdataiii
     id: "05743828"
   - name: grzejnik_kuchnia_imp
     driver: fhkvdataiii
     id: "05743835"
   - name: grzejnik_jadalnia_imp
     driver: fhkvdataiii
     id: "05743829"
   - name: grzejnik_julia_imp
     driver: fhkvdataiii
     id: "05743794"
   - name: grzejnik_sypialnia_imp
     driver: fhkvdataiii
     id: "05743837"
```

casper255, 2023-02-17:
Thank you.  The setup worked.  The indentation in the mqtt section is correct.  It was entered in the configuration and the yaml is generated by the add-on.  It seems that the problem was the lack of quotation marks around the id.  The configuration has been saved and the plugin is working.

thecem, 2023-02-17:
So if this is solved, pls close

casper255, 2023-02-17:
The thread is not mine, so I can't close it.

convicte, 2023-02-17:
Before we close, million-dollar questions is if the same schema will work for ``` donotprobe: ```.
In other words, will ```donotprobe``` accept an array, because this is something at least I am using right now, to prevent it from talking to other SDRs and USB devices, causing conflicts.

For the moment the ```donotprobe``` key can be listed several times, but that is not possible in the new schema

See below for the 1.10.0 schema config:

```yaml
data_path: /config/wmbusmeters
enable_mqtt_discovery: true
conf: >-
  loglevel=normal

  #device=rtlwmbus[77771111153705700]:c1 

  device=rtlwmbus:CMD(/usr/bin/nc -lku 9011)

  donotprobe=/dev/ttyUSB0

  donotprobe=/dev/bus/usb/001/005

  donotprobe=/dev/bus/usb/002/002

  donotprobe=/dev/bus/usb/001/004

  logtelegrams=false

  format=json

  logfile=/dev/stdout

  meterfiles=/config/wmbusmeters/logs/meter_readings

  meterfilesaction=append

  meterfilesnaming=name

  meterfilestimestamp=day

  shell=/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"

  alarmshell=/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$ALARM_TYPE
  $ALARM_MESSAGE"

  ignoreduplicates=true

  alarmexpectedactivity=mon-sun(00-23)

  alarmtimeout=3h
meters:
  - |-
    name=water_meter
    driver=izar
    id=4141d24a
    key=NOKEY
  - |-
    name=garden_water_meter
    driver=izar
    id=2141503d
    key=NOKEY
mqtt: {}

```
This is the layout I would envision it working under, but please let me know if that makes sense:

```yaml

data_path: /config/wmbusmeters
enable_mqtt_discovery: true
conf:
  loglevel: normal
  device: rtlwmbus:CMD(/usr/bin/nc -lku 9011)
  donotprobe: 
     - /dev/ttyUSB0
     - /dev/bus/usb/001/005
     - /dev/bus/usb/002/002
     - /dev/bus/usb/001/004
  logtelegrams: false
  format: json
  logfile: /dev/stdout
  shell: /wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"
  alarmshell: /wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$ALARM_TYPE $ALARM_MESSAGE"
  meterfiles: /config/wmbusmeters/logs/meter_readings
  meterfilesaction: append
  meterfilesnaming: name
  meterfilestimestamp: day
  ignoreduplicates: true
  alarmexpectedactivity: mon-sun(00-23)
  alarmtimeout: 3h
meters:
   - name: water_meter
     driver: izar
     id: "4141d24a"
     key: NOKEY
   - name: garden_water_meter
     driver: izar
     id: "2141503d"
     key: NOKEY
mqtt: {}

Finally, should I drop the `key` key, when `NOKEY` is used for the old schema or should we keep the `NOKEY` to ensure the variable is called correctly?

convicte, 2023-02-17:
Friendly advice @casper255, unless you want to use it for troubleshooting, consider setting the `logtelegrams: true` to `false`.
If you don't, your HA instance will balloon massively, as many of the devices send out telegrams every few seconds.
The same is true for ` meterfilesaction: overwrite` as compared to `append`, which I am currently using—the log files can grow to 100s of MB. Will need to change it to overwrite as soon as I am done to save space.

convicte, 2023-02-17:
@KrzysztofHajdamowicz @thecem 
I've checked it already, and it fails as well:

[image]

So for the moment I'd like to keep this open, unless you'd like to close this one and open another issue for the `donotprobe:` array error?

I am fine with either option.

chpego, 2023-02-17:
In the conf section, there are only the values declared in the schema section of the config.json file; 
I think we have to see to have the possibility to extend the scheme to all available options of wmbusmeter

KrzysztofHajdamowicz, 2023-02-17:
> @KrzysztofHajdamowicz @thecem I've checked it already, and it fails as well:
> 
> [image]
> 
> So for the moment I'd like to keep this open, unless you'd like to close this one and open another issue for the `donotprobe:` array error?
> 
> I am fine with either option.

First, split `data_path`, `enable_mqtt_discovery`, `conf` and `meters` into separate inputs.
It should look like that:
[image]

Second, currently using multiple `donotprobe` entries is not supported.

convicte, 2023-02-17:
@KrzysztofHajdamowicz 
Thank you very much for the comment.
[image]
You may find that if you click on the three dots and switch to YAML mode, as compared to UI, yours will look like mine, and vice versa. It's just a matter of formatting in the window - the headers for `data_path`, `conf`, etc. are automatically alocated.
After I click `Edit in UI` it will be displayed parsed into sections as seen in your screenshot:
[image]

@chpego you hit the nail on the head there:
```
23-02-17 11:24:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'alarmshell' for Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters)
23-02-17 11:24:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'meterfiles' for Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters)
23-02-17 11:24:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'meterfilesaction' for Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters)
23-02-17 11:24:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'meterfilesnaming' for Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters)
23-02-17 11:24:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'meterfilestimestamp' for Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters)
23-02-17 11:24:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'ignoreduplicates' for Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters)
23-02-17 11:24:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'alarmexpectedactivity' for Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters)
23-02-17 11:24:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'alarmtimeout' for Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters)
```
Trying to use the config above constantly prints errors into the supervisor log - hence I'll have to switch back to 1.11.0 for now, until the schema can accommodate things like alarms.

chpego, 2023-02-17:
> @chpego you hit the nail on the head there:
> 
> ```
> 23-02-17 11:24:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'alarmshell' for Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters)
> 23-02-17 11:24:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'meterfiles' for Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters)
> 23-02-17 11:24:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'meterfilesaction' for Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters)
> 23-02-17 11:24:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'meterfilesnaming' for Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters)
> 23-02-17 11:24:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'meterfilestimestamp' for Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters)
> 23-02-17 11:24:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'ignoreduplicates' for Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters)
> 23-02-17 11:24:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'alarmexpectedactivity' for Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters)
> 23-02-17 11:24:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'alarmtimeout' for Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters)
> ```
> 
> Trying to use the config above constantly prints errors into the supervisor log - hence I'll have to switch back to 1.11.0 for now, until the schema can accommodate things like alarms, donotprobe arrays, etc.
> 
> I sincerely apologize for potentially sounding ungrateful, and hope it's understood that I appreciate everyone's efforts. Unfortunately, I am unable to assist in fixing it, but will continue to contribute however I can. I think I'll start creating separate bugs for each of the above to allow easier tracking of work in these areas.

I've made a draft PR https://github.com/weetmuts/wmbusmeters/pull/869

casper255, 2023-02-17:
@convicte thank you for the hint.

BIBOLV, 2023-02-17:
@convicte - firstly you can use rtl dongle serial to point to specific device that should be used `device: rtlwmbus[112233]` in that case other devices should not be touched and/or use `donotprobe=all` to prevent probing of other devices.
Secondly - as you already mentioned, please open ticket per issue and close this one, not to pile multiple issues in one ticket so that we would be able to track them properly.

convicte, 2023-02-17:
It's not exactly this simple if I am using an ESP32 based device to collect the telegrams and push them to the addon for interpretation. In this case, it's impossible to assign the serial address.
Historically, I've had situations where the specific ESP missing from the network would cause the add-on to sometimes grab an SDR tuned to 433Mhz collecting my weather station information. It could also sometimes cause conflicts with other devices - Google CORAL and Zigbee adapter, though this happened maybe once.

I am hereby closing this bug report and putting the issues in the respective bugs for each.

closed 2023-02-17 by convicte

BIBOLV, 2023-02-17:
Have you tried `donotprobe=all` and it did not work for you?

convicte, 2023-02-17:
> Have you tried `donotprobe=all` and it did not work for you?

At this time, I didn't know I can disable probe for all and still have the add-on look up the specific device listed above—that sounds counterintuitive.

So in short, I cannot say if it would work or not, but thank you for the idea.

BIBOLV, 2023-02-17:
Leave everything in your config as is just replace all `donotprobe` lines with one - `donotprobe=all`