wmbusmeters/wmbusmeters #870: donotprobe key does not accept an array - unable to list multiple USB devices

convicte, 2023-02-17:
### Type of request

Bug

### HAOS version

2023.02.04

### wmbusmeters version

1.11.0-2

### Your message goes here

The new schema will not work for more than one ``` donotprobe: ```.
In other words, ```donotprobe``` array is not accepted, 
This feature of the 1.11.0 schema was used to prevent the add-on from talking to other SDRs and USB devices, causing conflicts.

For the moment, up until 1.11.0-1 the ```donotprobe``` key could be listed several times, but that is not possible in the new schema

See below for the 1.10.0 schema config:

```yaml
data_path: /config/wmbusmeters
enable_mqtt_discovery: true
conf: >-
  loglevel=normal

  #device=rtlwmbus[77771111153705700]:c1 

  device=rtlwmbus:CMD(/usr/bin/nc -lku 9011)

  donotprobe=/dev/ttyUSB0

  donotprobe=/dev/bus/usb/001/005

  donotprobe=/dev/bus/usb/002/002

  donotprobe=/dev/bus/usb/001/004

  logtelegrams=false

  format=json

  logfile=/dev/stdout

  meterfiles=/config/wmbusmeters/logs/meter_readings

  meterfilesaction=append

  meterfilesnaming=name

  meterfilestimestamp=day

  shell=/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"

  alarmshell=/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$ALARM_TYPE
  $ALARM_MESSAGE"

  ignoreduplicates=true

  alarmexpectedactivity=mon-sun(00-23)

  alarmtimeout=3h
meters:
  - |-
    name=water_meter
    driver=izar
    id=4141d24a
    key=NOKEY
  - |-
    name=garden_water_meter
    driver=izar
    id=2141503d
    key=NOKEY
mqtt: {}

```
This is the layout I would envision it working under for 1.11.0-2 and beyond:

```yaml

data_path: /config/wmbusmeters
enable_mqtt_discovery: true
conf:
  loglevel: normal
  device: rtlwmbus:CMD(/usr/bin/nc -lku 9011)
  donotprobe: 
     - /dev/ttyUSB0
     - /dev/bus/usb/001/005
     - /dev/bus/usb/002/002
     - /dev/bus/usb/001/004
  logtelegrams: false
  format: json
  logfile: /dev/stdout
  shell: /wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"
  alarmshell: /wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$ALARM_TYPE $ALARM_MESSAGE"
  meterfiles: /config/wmbusmeters/logs/meter_readings
  meterfilesaction: append
  meterfilesnaming: name
  meterfilestimestamp: day
  ignoreduplicates: true
  alarmexpectedactivity: mon-sun(00-23)
  alarmtimeout: 3h
meters:
   - name: water_meter
     driver: izar
     id: "4141d24a"
     key: NOKEY
   - name: garden_water_meter
     driver: izar
     id: "2141503d"
     key: NOKEY
mqtt: {}
```

Please include array support for `donotprobe` in the same way it's currently supported for meters.

convicte, 2023-02-17:
[image]

See bug https://github.com/weetmuts/wmbusmeters/issues/860 for more detailed conversation on the subject

juliekoubova, 2023-03-05:
Please see PR #880

BIBOLV, 2023-03-13:
HA add-on has moved to separate repo - https://github.com/wmbusmeters/wmbusmeters-ha-addon
It has now pre-built container images and config interface changes - please test and close this issue if no other questions.

closed 2023-06-12 by github-actions[bot]