wmbusmeters/wmbusmeters #871: New config schema doesn't support all wmbusmeters features available previously

convicte, 2023-02-17:
### Type of request

Bug

### HAOS version

2023.02.04

### wmbusmeters version

1.11.0-2

### Your message goes here

Previously available variable can no longer be called in 1.11.0-1 and beyond:

```
23-02-17 11:24:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'alarmshell' for Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters)
23-02-17 11:24:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'meterfiles' for Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters)
23-02-17 11:24:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'meterfilesaction' for Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters)
23-02-17 11:24:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'meterfilesnaming' for Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters)
23-02-17 11:24:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'meterfilestimestamp' for Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters)
23-02-17 11:24:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'ignoreduplicates' for Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters)
23-02-17 11:24:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'alarmexpectedactivity' for Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters)
23-02-17 11:24:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'alarmtimeout' for Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters)
```

Work on resolving the issue is underway in this PR https://github.com/weetmuts/wmbusmeters/pull/869

Example config resulting in the errors above.
```yaml

data_path: /config/wmbusmeters
enable_mqtt_discovery: true
conf:
  loglevel: normal
  device: rtlwmbus:CMD(/usr/bin/nc -lku 9011)
  donotprobe:   /dev/ttyUSB0
  logtelegrams: false
  format: json
  logfile: /dev/stdout
  shell: /wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"
  alarmshell: /wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$ALARM_TYPE $ALARM_MESSAGE"
  meterfiles: /config/wmbusmeters/logs/meter_readings
  meterfilesaction: append
  meterfilesnaming: name
  meterfilestimestamp: day
  ignoreduplicates: true
  alarmexpectedactivity: mon-sun(00-23)
  alarmtimeout: 3h
meters:
   - name: water_meter
     driver: izar
     id: 4141d24a
     key: NOKEY
   - name: garden_water_meter
     driver: izar
     id: 2141503d
     key: NOKEY
mqtt: {}

```

convicte, 2023-02-17:
For more information - please see a longer discuss in this bug https://github.com/weetmuts/wmbusmeters/issues/860#issuecomment-1434734252

BIBOLV, 2023-03-13:
HA add-on has moved to separate repo - https://github.com/wmbusmeters/wmbusmeters-ha-addon
It has now pre-built container images and config interface changes - please test and close this issue if no other questions.

closed 2023-06-12 by github-actions[bot]