wmbusmeters/wmbusmeters #872: Can't start bone stock HA add-on - Invalid list for option 'meters'...

mediaNen, 2023-02-17:
**Type of request**
Bug

**HAOS version**
2023.02.05

**wmbusmeters version**
1.11.0-2

**Your message goes here:**
Installed the HA add-on from scratch, tried to start it up without changing anything - it gives the following error:
Invalid list for option 'meters' in Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters). (I get the option to go to the config tile where it shows the errors in greater detail when editing something and clicking save):

"Failed to save add-on configuration, Invalid list for option 'meters' in Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters). Got {'data_path': '/config/wmbusmeters', 'enable_mqtt_discovery': False, 'conf': {'loglevel': 'normal', 'device': 'auto:t1:c1', 'donotprobe': '/dev/ttyAMA0', 'logtelegrams': False, 'format': 'json', 'logfile': '/dev/stdout', 'shell': '/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"'}, 'meters': {'name': 'ElectricityMeter', 'driver': 'amiplus', 'id': '12345678', 'key': 'DEADBEEF1337DEADBEEF1337DEADBEEF'}, 'mqtt': {}}"

I have used this up till today on my RPI4 without any issues, but when I moved to a freshly installed VM the newly installed add-on will not start. I was using an older version of the wmbusmeters add-on which I guess is why it worked on the RPI4.

BIBOLV, 2023-02-17:
Configuration has changed in latest version - please check docs. 
If you can`t find where the issue is - post config here.

mediaNen, 2023-02-17:
It is setup with the new configuration - bone stock :)
The config is in yaml:

data_path: /config/wmbusmeters
enable_mqtt_discovery: false
conf:
  loglevel: normal
  device: auto:t1
  donotprobe: /dev/ttyAMA0
  logtelegrams: false
  format: json
  logfile: /dev/stdout
  shell: /wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"
meters:
  name: ElectricityMeter
  driver: amiplus
  id: "12345678"
  key: DEADBEEF1337DEADBEEF1337DEADBEEF
mqtt: {}

[image]

BIBOLV, 2023-02-17:
And what config gives you error?

mediaNen, 2023-02-17:
I am not sure, I can't find any library in config or ect folder in HA. Forgive me for beeing a newbie here.
Do I manually have to add files to the /config/ folder?
[image]

BIBOLV, 2023-02-17:
What config do you have on old instance, I guess you want to transfer it, or want to start from scratch?

mediaNen, 2023-02-17:
I guess I would rather start from scratch.
The old config was like this:

|-
  loglevel=normal
  device=im871a:c1,t1
  donotprobe=/dev/ttyAMA0
  logtelegrams=true
  format=json
  logfile=/dev/stdout
  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"

- |-
  name=WaterSensor
  driver=multical21
  id=76791307
  key=8EF8FAEC2103F0BECFA1487XXXXXXXX

I do not have a wmbusmeters folder under the /config/ folder - I had on my old PI4.

BIBOLV, 2023-02-17:
Try following, this should have you started:
```yaml
data_path: /config/wmbusmeters
enable_mqtt_discovery: false
conf:
  loglevel: normal
  device: im871a:c1,t1
  donotprobe: /dev/ttyAMA0
  logtelegrams: true
  format: json
  logfile: /dev/stdout
  shell: wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters:
  - name: WaterSensor
    driver: multical21
    id: "76791307"
    key: 8EF8FAEC2103F0BECFA1487XXXXXXXX
mqtt: {}

```

mediaNen, 2023-02-17:
I have no idea why that worked, but now it started, thank you!
There is still a small issue:
"/bin/sh: wmbusmeters/mosquitto_pub.sh: not found
(shell) /bin/sh exited with non-zero return code: 127"

[image]

BIBOLV, 2023-02-17:
`shell: /wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"`

mediaNen, 2023-02-17:
That did the trick, fantastic!
Thank you so much for your help. Been sitting for a couple of hours pulling my hair out. Could not figure out what to change.
Have a great evening!

closed 2023-02-17 by mediaNen

BIBOLV, 2023-02-17:
Take care 😉

infrat, 2023-02-23:
Btw. I had exactly the same issue with my fresh setup for wmbusmeters. Fixed with suggestion @BIBOLV