wmbusmeters/wmbusmeters #874: Can not save config and run addon

pepsonEL, 2023-02-19:
### Type of request

Enhancement

### HAOS version

HassOS 9.5

### wmbusmeters version

1.11.0-2 

### Your message goes here

Hi
I installed wmbus addon and connect by USB adapter with radio. I create config and try save it and run addon but i can not:

```
data_path: /config/wmbusmeters
enable_mqtt_discovery: false
conf:
  loglevel: normal
  device: auto:t1
  donotprobe: /dev/ttyUSB1
  logtelegrams: false
  format: json
  logfile: /dev/stdout
  shell: /wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"
meters:
  name: WaterMeter
  driver: apator162
  id: "00068526"
  key: 00000000000000000000000000000000
mqtt:
  host: 192.168.10.3
  port: 1883
  user: MyUser
  password: MyPassowrd
```

i get this error when try save:

Nie udało się zapisać konfiguracji dodatku, Invalid list for option 'meters' in Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters). Got {'data_path': '/config/wmbusmeters', 'enable_mqtt_discovery': False, 'conf': {'loglevel': 'normal', 'device': 'auto:t1', 'donotprobe': '/dev/ttyUSB1', 'logtelegrams': False, 'format': 'json', 'logfile': '/dev/stdout', 'shell': '/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"'}, 'meters': {'name': 'WaterMeter', 'driver': 'apator162', 'id': '00068526', 'key': 0}, 'mqtt': {'host': '192.168.10.3', 'port': 1883, 'user': 'MyUser', 'password': 'MyPassword'}}

thecem, 2023-02-19:
Meters needs an array:

```YAML
meters:
  - name: WaterMeter
     driver: apator162
     id: "00068526"
     key: 00000000000000000000000000000000
```

pepsonEL, 2023-02-19:
What you mean ?

thecem, 2023-02-19:
Put my example, in your config. With hyphen

pepsonEL, 2023-02-19:
What example ?

thecem, 2023-02-19:
[image]

pepsonEL, 2023-02-19:
ok i change
```
meters:
  - name: WaterMeter
    driver: apator162
    id: "00068526"
    key: 00000000000000000000000000000000
```

and also can not save

thecem, 2023-02-19:
If your key is only a number you to put it in quotation marks:

"12345678901234567890123456789012"

pepsonEL, 2023-02-19:
Ok good now saved and run. 
But in mqtt add me sensor but show only RSSI and state unknown and no show other information....

pepsonEL, 2023-02-19:
But how add adapter to found me my meters and show correct ID ?

pepsonEL, 2023-02-19:
now add in mqtt sensor also sensor name TOTAL but also show UNKNOWN

pepsonEL, 2023-02-19:
and in logs show me:
Your cul firmware has a bug that prevents longer telegrams from being received.

BIBOLV, 2023-02-22:
Hi, closing this issue as issue was solved and cul problem registered in https://github.com/weetmuts/wmbusmeters/issues/876, so this is dublicate. Please reopen if necessary.

closed 2023-02-22 by BIBOLV