wmbusmeters/wmbusmeters #876: Your cul firmware has a bug that prevents longer telegrams from being received.!

pepsonEL, 2023-02-19:
### Type of request

Enhancement

### HAOS version

HassOS 9.5

### wmbusmeters version

1.11.0-2 

### Your message goes here

Hi
I have a problem with nanoCUL and i can not found solution... Anybody help me please... I get this error in logs addon in HA

```
Started auto cul on /dev/ttyUSB1 listening on t1
(cul) Warning! Your cul firmware has a bug that prevents longer telegrams from being received.!
(cul) Please read: https://github.com/weetmuts/wmbusmeters/issues/390
(cul)         and: https://weetmuts.github.io/wmbusmeterswiki/nanoCUL.html
(cul) warning: the hex string is not proper! Ignoring telegram!
(cul) Warning! Your cul firmware has a bug that prevents longer telegrams from being received.!
(cul) Please read: https://github.com/weetmuts/wmbusmeters/issues/390
(cul)         and: https://weetmuts.github.io/wmbusmeterswiki/nanoCUL.html
(cul) warning: the hex string is not proper! Ignoring telegram!
(cul) Warning! Your cul firmware has a bug that prevents longer telegrams from being received.!
(cul) Please read: https://github.com/weetmuts/wmbusmeters/issues/390
(cul)         and: https://weetmuts.github.io/wmbusmeterswiki/nanoCUL.html
(cul) warning: the hex string is not proper! Ignoring telegram!
(cul) Warning! Your cul firmware has a bug that prevents longer telegrams from being received.!
(cul) Please read: https://github.com/weetmuts/wmbusmeters/issues/390
(cul)         and: https://weetmuts.github.io/wmbusmeterswiki/nanoCUL.html
(cul) warning: the hex string is not proper! Ignoring telegram!
(cul) Warning! Your cul firmware has a bug that prevents longer telegrams from being received.!
(cul) Please read: https://github.com/weetmuts/wmbusmeters/issues/390
(cul)         and: https://weetmuts.github.io/wmbusmeterswiki/nanoCUL.html
(cul) warning: the hex string is not proper! Ignoring telegram!
```

snapspizza, 2023-02-22:
Did you try to update the firmware of your nanoCUL, as described in the [wiki](https://github.com/wmbusmeters/wmbusmeters-wiki/blob/master/nanoCUL.md)?

pepsonEL, 2023-02-23:
Ok now start working show consumption water and show correct RSSI. Thanks for your help.

closed 2023-02-23 by pepsonEL