wmbusmeters/wmbusmeters #877: Sensus decryption key

Chiill34, 2023-02-20:
### Type of request

Question

### OS version

Debian

### wmbusmeters version

last

### Your message goes here

Hi,

I tried to use wbbusmaster but you need the key.
I don't know if Sensus use diffèrent keys.
Can i decrypt without asking Sensus?

It's this meter :
https://sensus.com/emea/products/640-water-meter/

Here are the telegrams I got.

Received telegram from: 00301583
          manufacturer: (SEN) Sensus Metering Systems, Germany (0x4cae)
          type: Water meter (0x07) encrypted
          ver: 0x7c
          device: rtlwmbus[00000001]
          rssi: 133 dBm
          driver: iperl

telegram=|_1E44AE4C831530007C077A4C001005C2B2F288A6F55826A19AB2EACE905BD6|+302
telegram=|_1E44AE4C831530007C077A4D0010059244DB725E39AE6AD9A4574859B0D2A2|+4894
telegram=|_1E44AE4C831530007C077A4E001005C28D851AE5C1D48CC2BDBAEDAFE1F025|+7702
telegram=|_1E44AE4C831530007C077A4F00100541DCC7ADE7EC96D9AF24A3F436361266|+12320
telegram=|_1E44AE4C831530007C077A5000100503463907AA3485C77039DFC233D83A44|+14283
telegram=|_1E44AE4C831530007C077A51001005266908AB898A45D93E6B36A3E7C01BD9|+19528
telegram=|_1E44AE4C831530007C077A5200100533BE9CA7403D60E64BCEF701F4D81A8C|+21107
telegram=|_1E44AE4C831530007C077A530010051977B92EB0C8EC1957BDA022217FA2DA|+27412
telegram=|_1E44AE4C831530007C077A54001005F3F31DA2A0BC975740055F65570A693B|+29021

Thanks for this work.

BIBOLV, 2023-02-20:
Hi, you need a decryption, without it you will not be able to decrypt and read message. You need to ask key from that entity that installed them or own them, not manufacturer, unless you bought them from manufacturer. Search in this issue archive - there are lot of discussions about this topic.

Murcin86, 2023-02-23:
@Chiill34 
iPerl have default key - please try this so maybe all Sensus works on the same
E6C88800DEB868C0D6A84880CE982840 - in my case that works !

Chiill34, 2023-02-25:
> @Chiill34 iPerl have default key - please try this so maybe all Sensus works on the same E6C88800DEB868C0D6A84880CE982840 - in my case that works !

You are my lord, it's working !!!!
Thanks a lot dude.

fguiet, 2023-03-19:
Hi,

I have got a brand new iPerl Sensus Water meter. So I wanted to know if I can receive some information.

[image]

Here is what wmbusmeters managed to receive:

```bash
(TPL) num encrypted blocks 1 (16 bytes and remaining unencrypted 0 bytes)
(telegram) TPL CI=7a ACC=31 STS=00 CFG=0510 (AES_CBC_IV nb=1 cntn=0 ra=0 hc=0)
Received telegram from: 00283922
          manufacturer: (SEN) Sensus Metering Systems, Germany (0x4cae)
                  type: Water meter (0x07) encrypted
                   ver: 0x68
                device: rtlwmbus[00000001]
                  rssi: 22 dBm
                driver: iperl
(wmbus) 000   : 1e length (30 bytes)
(wmbus) 001   : 44 dll-c (from meter SND_NR)
(wmbus) 002   : ae4c dll-mfct (SEN)
(wmbus) 004   : 22392800 dll-id (00283922)
(wmbus) 008   : 68 dll-version
(wmbus) 009   : 07 dll-type (Water meter)
(wmbus) 010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(wmbus) 011   : 31 tpl-acc-field
(wmbus) 012   : 00 tpl-sts-field (OK)
(wmbus) 013   : 1005 tpl-cfg 0510 (AES_CBC_IV nb=1 cntn=0 ra=0 hc=0 )
(wmbus) 015 CE: 48E6FD2055CA411092CFD330200702B9 encrypted
(wmbus) telegram from 00283922 ignored by all configured meters!
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2023-03-19 19:38:41.000;25;25;00283922;0x1e44ae4c2239280068077a3100100548e6fd2055ca411092cfd330200702b9<0A>"
(rtlwmbus) received full frame
```
I managed to decrypt the encrypted part with the magic key : E6C88800DEB868C0D6A84880CE982840

[image]

Thank you for this great software

g1za, 2024-04-18:
May I ask @Chiill34 @fguiet @Murcin86 if you own(ed) a personal (you bought) iPerl meter or anything provided by the water supplier? I'm trying to understand if the decryption key may work also if you do now own it.
Thanks!

fguiet, 2024-04-18:
Hi @g1za 

The iPerl meter was provided to me by my water supplier. I found the decryption key on the Internet and it proved to be working on my model.

Cheers

g1za, 2024-04-23:
@fguiet Unluckily I'm not as lucky as you :'(
So it really depends

smai86, 2024-11-04:
> @Chiill34 iPerl have default key - please try this so maybe all Sensus works on the same E6C88800DEB868C0D6A84880CE982840 - in my case that works !

I´ve got an iPerl installed a few weeks ago. the decryption key above sadly didn´t work for my iperl...

[image]