wmbusmeters/wmbusmeters #883: Problem with MQTT connection

koles142, 2023-02-24:
### Type of request

Bug

### HAOS version

9.5

### wmbusmeters version

1.11.0-2

### Your message goes here

Hi,
I have problem with add-on. Can't connect to the MQTT Broker. The same broker, with the same login and pass is working with z2m without any problem.

My MQTT config:
`host: 192.168.31.7
port: 1883
user: xxx
password: xxx`

Error in log of add-on:

`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[22:06:06] INFO: CONFIG_CONF ...
[22:06:06] INFO: {"loglevel":"normal","device":"rtlwmbus:868.9M:t1","donotprobe":"/dev/ttyAMA0","logtelegrams":false,"format":"json","logfile":"/dev/stdout","shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[22:06:06] INFO: CONFIG_METERS ...
[22:06:06] INFO: []
[22:06:06] INFO: Syncing wmbusmeters configuration ...
[22:06:06] INFO: Registering meters ...
[22:06:07] INFO: Generating MQTT configuration ... 
[22:06:07] INFO: Broker 192.168.31.7 will be used.
[22:06:07] ERROR: Got unexpected response from the API: Service not enabled
MQTT Discovery cleanup...
[22:06:08] INFO: Running wmbusmeters ...
Started config rtlwmbus[00000001] listening on t1 using fq 868.9M
No meters configured. Printing id:s of all telegrams heard!`

Have you any idea?

BIBOLV, 2023-02-25:
Hi,
Did you configure some meters and checked that no data is actually delivered to mqtt?

koles142, 2023-02-26:
Hi, I've add 4 meters. I have still in log the same error, but autodiscovery MQTT is working and I have sensors in HA. Now I'm waiting for data from meters to confirmation hat connectivity is working.

```yaml
[11:59:52] INFO: Syncing wmbusmeters configuration ...
[11:59:52] INFO: Registering meters ...
[11:59:52] INFO: Adding meter-0001 ...
[11:59:52] INFO: Adding meter-0002 ...
[11:59:52] INFO: Adding meter-0003 ...
[11:59:53] INFO: Adding meter-0004 ...
[11:59:53] INFO: Generating MQTT configuration ... 
[11:59:53] INFO: Broker 192.168.31.7 will be used.
[11:59:53] ERROR: Got unexpected response from the API: Service not enabled
MQTT Discovery ...
 Adding meter: Łazienka - zimna woda ...
  Add/update topic: homeassistant/sensor/wmbusmeters/04652559_rssi_dbm/config
  Add/update topic: homeassistant/sensor/wmbusmeters/04652559_timestamp/config
  Add/update topic: homeassistant/sensor/wmbusmeters/04652559_total_m3/config
 Adding meter: Łazienka - ciepła woda ...
  Add/update topic: homeassistant/sensor/wmbusmeters/04764452_rssi_dbm/config
  Add/update topic: homeassistant/sensor/wmbusmeters/04764452_timestamp/config
  Add/update topic: homeassistant/sensor/wmbusmeters/04764452_total_m3/config
 Adding meter: Kuchnia - zimna woda ...
  Add/update topic: homeassistant/sensor/wmbusmeters/04704320_rssi_dbm/config
  Add/update topic: homeassistant/sensor/wmbusmeters/04704320_timestamp/config
  Add/update topic: homeassistant/sensor/wmbusmeters/04704320_total_m3/config
 Adding meter: Kuchnia - ciepła woda ...
  Add/update topic: homeassistant/sensor/wmbusmeters/04807280_rssi_dbm/config
  Add/update topic: homeassistant/sensor/wmbusmeters/04807280_timestamp/config
  Add/update topic: homeassistant/sensor/wmbusmeters/04807280_total_m3/config
MQTT Discovery cleanup...
[11:59:58] INFO: Running wmbusmeters ...
Started config rtlwmbus[00000001] listening on t1 using fq 868.9M
```

koles142, 2023-02-26:
Everything working, thanks for help

closed 2023-02-26 by koles142