wmbusmeters/wmbusmeters #886: wmbusmeters.service son't start

Chiill34, 2023-02-25:
### Type of request

Bug

### OS version

Debian 11

### wmbusmeters version

last

### Your message goes here

Hi,
I can use wmbusmeters for listening telegrams but when i try to reload service, it's not working. He is loaded but not active..

root@JANET:/etc# systemctl status wmbusmeters.service
● wmbusmeters.service - "wmbusmeters service"
     Loaded: loaded (/lib/systemd/system/wmbusmeters.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sat 2023-02-25 18:52:07 CET; 28s ago
       Docs: https://github.com/weetmuts/wmbusmeters
             man:wmbusmeters(1)
    Process: 50396 ExecStartPre=/bin/mkdir -p /var/lib/wmbusmeters/meter_readings (code=exited, status=0/SUCCESS)
    Process: 50397 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/lib/wmbusmeters/meter_readings (code=exited, status=0/SUCCESS)
    Process: 50398 ExecStartPre=/bin/mkdir -p /var/log/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 50399 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/log/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 50400 ExecStartPre=/bin/mkdir -p /run/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 50401 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /run/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 50402 ExecStart=/usr/sbin/wmbusmetersd /run/wmbusmeters/wmbusmeters.pid (code=exited, status=0/SUCCESS)
   Main PID: 50403 (code=exited, status=1/FAILURE)

févr. 25 18:52:06 JANET systemd[1]: wmbusmeters.service: Main process exited, code=exited, status=1/FAILURE
févr. 25 18:52:06 JANET systemd[1]: wmbusmeters.service: Failed with result 'exit-code'.
févr. 25 18:52:07 JANET systemd[1]: wmbusmeters.service: Scheduled restart job, restart counter is at 3.
févr. 25 18:52:07 JANET systemd[1]: Stopped "wmbusmeters service".
févr. 25 18:52:07 JANET systemd[1]: wmbusmeters.service: Start request repeated too quickly.
févr. 25 18:52:07 JANET systemd[1]: wmbusmeters.service: Failed with result 'exit-code'.
févr. 25 18:52:07 JANET systemd[1]: Failed to start "wmbusmeters service".

Thanks !

BIBOLV, 2023-02-25:
Hi,
Please set loglevel to debug in configuration, restart and check wmbusmeters logs.

Chiill34, 2023-02-25:
Thx, didn't know this debug mod.
rtl_sdr was in /usr/local/bin and not in /usr/bin
It's ok now.

closed 2023-02-25 by Chiill34