wmbusmeters/wmbusmeters #888: (apator162) telegram contains a register with unknown size

lustefaniak, 2023-02-27:
### Type of request

Existing device improvement

### Meter description

Apator AT-WMBUS-16-2 HW:3 FW:3

### Logged telegrams from meter in decrypted format

```shell
3E4401060403820305077A090030852F2F0F9B5B229700000044C2DED310A25500007201C64A0000853C000094310000A0464B1904FFFFFFFFFFFFFFFF2ED6
3E4401062237340405077ADA0030852F2F0FA65B229700000044CED6D610A10701007201DBE60000ABC00000E39F0000A0B0508B04FFFFFFFFFFFFFFFFC15F
3E4401062237340405077AD80030852F2F0FA05B229700000042E1E9340010A10701007201DBE60000ABC00000E39F0000A0B0508B04FFFFFFFFFFFFFF8AF7
3E4401062237340405077AD70030852F2F0F9F5B229700000041555910A10701007201DBE60000ABC00000E39F0000A0B0508B04FFFFFFFFFFFFFFFFFF7EEB
```

weetmuts, 2023-04-29:
Its on the todo.

weetmuts, 2023-05-12:
Fix pushed! Please test!

sharo1k, 2023-06-04:
I do not want to open another ticket, but have 0x70 register with unknown size. It is historical value also.
one line of code fixes it: "case 0x70: return 1+1*4;" :)

weetmuts, 2023-06-04:
@sharo1k You have to publish the telegram here so that we can decode it. Use --logtelegrams

sharo1k, 2023-06-04:
Here is one:
4E4401063566040305077A300000852F2F0FA884F6970C0200A343181403C62314A0CE2C9203F00101000042409D1D0141545940C3CCD2F1F4F3**700F00000000**11FF0010986A0100FFFFFFFFFF43E4
This is brand new module, so no historical data is available.

Basically registers 0x70-0x7F are historical data for 1 to 16 months. First byte is day of month for saving the data followed by 4 bytes for each measurement point.

closed 2023-12-05 by github-actions[bot]