wmbusmeters/wmbusmeters #889: No function after update wmbusmeters

Benleni17, 2023-02-27:
### Type of request

Enhancement

### HAOS version

1.7

### wmbusmeters version

1.11.0-2

### Your message goes here

Hello,

no function after update wmbusmeters.

wmbusmeters does not start and gives the message:

Invalid dict for option 'conf' in Wmbusmeters (W-MBus to MQTT) (eeb738b7_wmbusmeters).

Whats the Problem?

thecem, 2023-02-27:
What for Version of HAOS?

you have to upgrade to 9.5 (latest)

Benleni17, 2023-02-27:
soory!

The Operating System is 9.5 but the another software level is:

Home Assistant 2023.1.7
Supervisor 2023.01.1
Operating System 9.5
Frontend 20230110.0 - latest

thecem, 2023-02-27:
Could you paste the complete log, with loglevel debug

Benleni17, 2023-02-27:
Here is the config:

data_path: /config/wmbusmeters
enable_mqtt_discovery: true
conf: >-
  loglevel=normal\ndevice=auto:c1\ndonotprobe=/dev/ttyAMA0\nlogtelegrams=true\nformat=json\nlogfile=/dev/stdout\nshell=/wmbusmeters/mosquitto_pub.sh
  "wmbusmeters/$METER_NAME" "$METER_JSON"\n
meters:
  - |-
    name=MainWater
    driver=multical21
    id=76406477
    key=AD2F041B407A23917A8D810217AA304B
mqtt: {}

Benleni17, 2023-02-27:
[image]

thecem, 2023-02-27:
The config has changed, your config is old style.

Benleni17, 2023-02-27:
it worked until the update. How do I have to change the configuration so that it works again. Can you help me?

Benleni17, 2023-02-27:
Thanks ! Its works. And now, whats new for my multival21?

How can I determine the daily requirement without the energy board?

thecem, 2023-02-27:
- Settings
- Integrations
- Helpers
- Add helper’s 
- consumption meter
- Entry entity: your total energy consumption entity from wmbusmeters 
- meter reset cycle: Daly 
- save

thecem, 2023-02-28:
Pls close the issue. Thx.