wmbusmeters/wmbusmeters #892: Can't set meters without options

Drealine, 2023-03-01:
### Type of request

Bug

### HAOS version

NA

### wmbusmeters version

NA

### Your message goes here

Hi,

Since last update, I can't set meters without it in the configuration. It say 👍 
Échec de l'enregistrement de la configuration du module complémentaire, Invalid list for option 'meters' in Wmbusmeters (W-MBus to MQTT) (cada3e3b_wmbusmeters). Got {'data_path': '/config/wmbusmeters', 'enable_mqtt_discovery': False, 'conf': {'loglevel': 'normal', 'device': 'auto:t1', 'donotprobe': '/dev/ttyAMA0', 'logtelegrams': False, 'format': 'json', 'logfile': '/dev/stdout', 'shell': '/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"'}, 'meters': {}, 'mqtt': {}}

chpego, 2023-03-01:
Same problem from here https://github.com/weetmuts/wmbusmeters/issues/874 ?
Could you post your config ?

tOOmki, 2023-03-10:
Same problem:
[edge] Wmbusmeters (W-MBus to MQTT)
Current version: 1.11.0-5-3

Default config

`data_path: /config/wmbusmeters
enable_mqtt_discovery: false
conf:
  loglevel: normal
  device: auto:t1
  donotprobe: /dev/ttyAMA0
  logtelegrams: false
  format: json
  logfile: /dev/stdout
  shell: /wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"
meters:
  name: ElectricityMeter
  driver: amiplus
  id: "12345678"
  key: DEADBEEF1337DEADBEEF1337DEADBEEF
mqtt: {}
`

thecem, 2023-03-10:
Pls post it like: https://github.com/orgs/wmbusmeters/discussions/896#discussioncomment-5263757

thecem, 2023-03-10:
You have configure the meters as an array!

```YAML
meters:
  - name: ElectricityMeter
    driver: amiplus
    id: "12345678"
    key: DEADBEEF1337DEADBEEF1337DEADBEEF
```

thecem, 2023-03-11:
pls close if fixed:
https://github.com/orgs/wmbusmeters/discussions/896#discussioncomment-5271928

BIBOLV, 2023-03-13:
HA add-on has moved to separate repo - https://github.com/wmbusmeters/wmbusmeters-ha-addon
It has now pre-built container images and config interface changes - please test and close this issue if no other questions.

closed 2023-06-12 by github-actions[bot]