wmbusmeters/wmbusmeters #898: Add driver for Sontex Supercal 739

raknjane, 2023-03-05:
### Type of request

New device support

### Meter description

Hi,

I am using a AMB8465 USB stick trying to read data from a Sontex Supercal 739 (wM-Bus EN13757-4) meter via (wmbus).

This is all new to me but I have managed to retrieve data.
I followed the HowToAddaNewMeter.txt which Simulation suggested to use "Best driver  : microclima 25/31" 

I did a test using the "supercom587" driver having the same result.

This driver does not appear to work correctly and below message show up in the debug log: 

```
(meter) B1102UFH: meter detection did not match the selected driver microclima! correct driver is: unknown!
(meter) Not printing this warning again for id: 12345678 mfct: (SON) Sontex, Switzerland (0x4dee) type: Heat meter (0x04) ver: 0x1b
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
```

Please find details below, not sure if more details is needed if so let me know.

Thank you very much.

https://sontex.ch/en/applications/building-automation/#supercal-739

### Logged telegrams from meter in decrypted format

```shell
Received telegram from: 12345678
          manufacturer: (SON) Sontex, Switzerland (0x4dee)
                  type: Heat meter (0x04)
                   ver: 0x1b
                device: amb8465[00000000]
                  rssi: -74 dBm
                driver: unknown!
telegram=|_4444EE4D850166281B047AFB000000046D3612E423820A6CE1F1040628010000840A060000000004144A110000840A14000000000259E30F025D2A0C023B1D01032C3A0100|+6

wmbusmeters --analyze simulation_12345678.txt
Auto driver  : not found!
Best driver  : microclima 25/31
Using driver : microclima 25/31
000   : 44 length (68 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : ee4d dll-mfct (SON)
004   : 85016628 dll-id (12345678)
008   : 1b dll-version
009   : 04 dll-type (Heat meter)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : fb tpl-acc-field
012   : 00 tpl-sts-field (OK)(OK)
013   : 0000 tpl-cfg 0000 ( )
015   : 04 dif (32 Bit Integer/Binary Instantaneous value)
016   : 6D vif (Date and time type)
017 C!: 3612E423 ("meter_datetime":"2023-03-04 18:54")
021   : 82 dif (16 Bit Integer/Binary Instantaneous value)
022   : 0A dife (subunit=0 tariff=0 storagenr=20)
023   : 6C vif (Date type G)
024 C?: E1F1
026   : 04 dif (32 Bit Integer/Binary Instantaneous value)
027   : 06 vif (Energy kWh)
028 C!: 28010000 ("total_energy_consumption_kwh":296)
032   : 84 dif (32 Bit Integer/Binary Instantaneous value)
033   : 0A dife (subunit=0 tariff=0 storagenr=20)
034   : 06 vif (Energy kWh)
035 C!: 00000000 ("consumption_at_set_date_20_kwh":0)
039   : 04 dif (32 Bit Integer/Binary Instantaneous value)
040   : 14 vif (Volume 10⁻² m³)
041 C!: 4A110000 ("total_volume_m3":44.26)
045   : 84 dif (32 Bit Integer/Binary Instantaneous value)
046   : 0A dife (subunit=0 tariff=0 storagenr=20)
047   : 14 vif (Volume 10⁻² m³)
048 C?: 00000000
052   : 02 dif (16 Bit Integer/Binary Instantaneous value)
053   : 59 vif (Flow temperature 10⁻² °C)
054 C!: E30F ("flow_temperature_c":40.67)
056   : 02 dif (16 Bit Integer/Binary Instantaneous value)
057   : 5D vif (Return temperature 10⁻² °C)
058 C!: 2A0C ("return_temperature_c":31.14)
060   : 02 dif (16 Bit Integer/Binary Instantaneous value)
061   : 3B vif (Volume flow l/h)
062 C!: 1D01 ("volume_flow_m3h":0.285)
064   : 03 dif (24 Bit Integer/Binary Instantaneous value)
065   : 2C vif (Power 10¹ W)
066 C!: 3A0100 ("power_kw":3.14)

{
    "media":"heat",
    "meter":"microclima",
    "name":"",
    "id":"12345678",
    "meter_datetime":"2023-03-04 18:54",
    "flow_temperature_c":40.67,
    "return_temperature_c":31.14,
    "status":"OK",
    "total_energy_consumption_kwh":296,
    "total_volume_m3":44.26,
    "volume_flow_m3h":0.285,
    "power_kw":3.14,
    "consumption_at_set_date_20_kwh":0,
    "set_date_20_date":"-204568456-02-29",
    "timestamp":"2023-03-04T18:22:06Z"
}
```

closed 2023-06-03 by github-actions[bot]

weetmuts, 2024-06-19:
It turned out that the microclima driver was a red herring. Create a new driver supercal.
Check the source here:
 https://github.com/wmbusmeters/wmbusmeters/blob/master/drivers/src/supercal.xmq

weetmuts, 2024-06-19:
Fix pushed.