wmbusmeters/wmbusmeters #903: Automated testing fail during debuild

petterreinholdtsen, 2023-03-05:
### Type of request

Bug

### OS version

Debian 11 Bullseye

### wmbusmeters version

github master

### Your message goes here

Running 'debuild' to built the .deb, but the self testing fail like this:

```
make[1]: Leaving directory '/home/pere/src/wmbusmeters-new'
   dh_auto_test
        make -j2 test VERBOSE=1
make[1]: Entering directory '/home/pere/src/wmbusmeters-new'
Building 1.11.0-131-gc4e943d with local changes
OK: test internals
OK: Test C1 meters
OK json: Test T1 meters
OK fields: Test T1 meters
OK json: Test S1 meters
OK fields: Test S1 meters
OK: Test mbus
OK: Test ANYID with explicit driver
OK: Test ANYID with auto driver
OK: Test ANYID with auto but no driver found!
OK: Test list-envs and list-fields
OK: Test list-envs and list-fields
OK json: Test non-compliant apator162 meters
OK fields: Test non-compliant apator162 meters
OK json: Test Izars meters
OK: Test listen to all izars
OK: Test aes encrypted telegrams
OK: Test that telegram really is encrypted
OK: Test that failed decryption warning only prints once.
OK: Test shell invocation
OK: Test shell in config file
OK: Test shell environment
OK: Test that normal meterfiles are written
OK: Test that meterfiles with name-id are written
OK: Test that meterfiles with id are written
OK: Test that meterfiles with timestamps are written
OK: Test that meterfiles with timestamps are written
OK: Test normal config files with C1 meters
OK: Test logfile
OK: Test single meter conf file matches any id
OK: Test listen and print any meter heard in logfile
OK: Test listen and print any meter heard on stdout
OK: Test detection of drivers for all telegrams
OK: Test listen to wildcard * id
OK: Test listen to wildcard suffix 8856* id
OK: Test listen to two comma separted ids
OK: Test listen to three comma separted ids
OK: Test listen with negated ids
OK: Test conversions of units using calculations
OK: Test calculations
OK: Test cmdline formulas --calculate_... 
OK: Test meter config formulas calculate_... 
OK: Test date calculations
OK: Test selected fields
OK: Test oneshot
OK: Test wrong keys
OK: Test config4 with added conversions
OK: Test cmdline --overridedevice and --useconfig
[... 210 lines cut ...]
OK: Test config with oneshot
OK: Test config with exitafter
OK: Test config override with oneshot
OK: Test config override with exitafter
RUNNING Test alarms ...
STDERR---------------------------------
[ALARM DeviceInactivity] 4 seconds of inactivity resetting simulations/simulation_alarm.txt simulation (timeout 4s expected mon-sun(00-23) now 2023-03-05 18:21)
(wmbus) successfully reset wmbus device
STDOUT---------------------------------
TMP/OUTPUT-----------------------------
METER =={"media":"cold water","meter":"multical21","name":"Water","id":"76348799","status":"DRY","total_m3":6.408,"target_m3":6.408,"flow_temperature_c":127,"external_temperature_c":19,"current_status":"DRY","time_dry":"22-31 days","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2023-03-05T17:21:54Z"}==
METER =={"media":"cold water","meter":"multical21","name":"Water","id":"76348799","status":"DRY","total_m3":6.408,"target_m3":6.408,"flow_temperature_c":127,"external_temperature_c":19,"current_status":"DRY","time_dry":"22-31 days","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2023-03-05T17:22:00Z"}==
TMP/ALARM------------------------------
ALARM_SHELL DeviceInactivity [ALARM DeviceInactivity] 4 seconds of inactivity resetting simulations/simulation_alarm.txt simulation (timeout 4s expected mon-sun(00-23) now 2023-03-05 18:21)
---------------------------------------
OK: Test alarms
Some tests failed!
make[1]: *** [Makefile:318: test] Error 1
make[1]: Leaving directory '/home/pere/src/wmbusmeters-new'
dh_auto_test: error: make -j2 test VERBOSE=1 returned exit code 2
make: *** [debian/rules:17: binary] Error 25
dpkg-buildpackage: error: debian/rules binary subprocess returned exit status 2
debuild: fatal error at line 1182:
dpkg-buildpackage -us -uc -ui failed
```

weetmuts, 2023-03-05:
Its the netcat test where we feed wmbusmeters using udp packages using netcat.
The test is disable on macosx anyway so maybe not necessary to run during the debuild?

petterreinholdtsen, 2023-03-05:
[Fredrik Öhrström]
> Its the netcat test where we feed wmbusmeters using udp packages using
> netcat.  The test is disable on macosx anyway so maybe not necessary
> to run during the debuild?

No idea.  Is there some dependency requirement I might be missing?
-- 
Happy hacking
Petter Reinholdtsen

petterreinholdtsen, 2023-03-05:
Could it be that the test expect a different 'netcat' tool than is installed?  There is both netcat-openbsd and netcat-traditional in the Debian archive, and they do not have identical behaviour.  If a specific edition is needed, d/control should be reflected to document this.

petterreinholdtsen, 2023-03-05:
The test seem to only work with netcat-openbsd.

PR #905 (2023-03-06): Corrected deb/control typos and misfeatures.
Moved runtime dependency adduser from source to binary dependency.

Added build dependency netcat-openbsd needed during testing.

Corrected Suggest->Suggests typo.

Fixes #903.
commit a0cf8696 (2023-03-06): Corrected deb/control typos and misfeatures.

Moved runtime dependency adduser from source to binary dependency.

Added build dependency netcat-openbsd needed during testing.

Corrected Suggest->Suggests typo.

Fixes #903.
closed 2023-03-06 by weetmuts