wmbusmeters/wmbusmeters #907: wmbus detection not working anymore

Mindkey75, 2023-03-09:
### Type of request

Bug

### OS version

2023.3.2

### wmbusmeters version

1.11.0-2

### Your message goes here

After updating to 1.11.0-2 this is comming in the log. When i try to roll back to the old config and old version of Wmbusmeter it works again. 

MQTT Discovery ...
 Adding meter: WaterSensor ...
  Add/update topic: homeassistant/sensor/wmbusmeters/23028459_flow_m3h/config
  Add/update topic: homeassistant/sensor/wmbusmeters/23028459_max_flow_m3h/config
  Add/update topic: homeassistant/sensor/wmbusmeters/23028459_max_flow_temperature_c/config
  Add/update topic: homeassistant/sensor/wmbusmeters/23028459_min_external_temperature_c/config
  Add/update topic: homeassistant/sensor/wmbusmeters/23028459_min_flow_m3h/config
  Add/update topic: homeassistant/sensor/wmbusmeters/23028459_min_flow_temperature_c/config
  Add/update topic: homeassistant/sensor/wmbusmeters/23028459_rssi_dbm/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/23028459_status_burst/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/23028459_status_dry/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/23028459_status_leak/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/23028459_status_reversed/config
  Add/update topic: homeassistant/sensor/wmbusmeters/23028459_target_m3/config
  Add/update topic: homeassistant/sensor/wmbusmeters/23028459_total_m3/config
MQTT Discovery cleanup...
[12:19:52] INFO: Running wmbusmeters ...
[ALARM SpecifiedDeviceNotFound] the device im871a:t1,c1 is not working
No wmbus device detected, waiting for a device to be plugged in.
[ALARM SpecifiedDeviceNotFound] the device im871a:t1,c1 is not working
[ALARM SpecifiedDeviceNotFound] the device im871a:t1,c1 is not working

This is my config:

data_path: /config/wmbusmeters
enable_mqtt_discovery: true
conf:
  loglevel: normal
  device: im871a:c1,t1
  donotprobe: /dev/ttyUSB2
  logtelegrams: true
  format: json
  logfile: /dev/stdout
  shell: /wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"
meters:
  - name: WaterSensor
    driver: flowiq2200
    id: "23028459"
    key: 53A8585F471BD955F2xxxxxxxxxxxxxxxx
mqtt: {}

Mindkey75, 2023-03-09:
this is the whole log.

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[12:19:46] INFO: CONFIG_CONF ...
[12:19:46] INFO: {"loglevel":"normal","device":"im871a:c1,t1","donotprobe":"/dev/ttyUSB2","logtelegrams":true,"format":"json","logfile":"/dev/stdout","shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[12:19:46] INFO: CONFIG_METERS ...
[12:19:46] INFO: [{"name":"WaterSensor","driver":"flowiq2200","id":"23028459","key":"53A8585F471BDxxxxxxxxxxxxxxxxx"}]
[12:19:46] INFO: Syncing wmbusmeters configuration ...
[12:19:46] INFO: Registering meters ...
[12:19:46] INFO: Adding meter-0001 ...
[12:19:46] INFO: Generating MQTT configuration ... 
[12:19:47] INFO: Broker core-mosquitto will be used.
MQTT Discovery ...
 Adding meter: WaterSensor ...
  Add/update topic: homeassistant/sensor/wmbusmeters/23028459_flow_m3h/config
  Add/update topic: homeassistant/sensor/wmbusmeters/23028459_max_flow_m3h/config
  Add/update topic: homeassistant/sensor/wmbusmeters/23028459_max_flow_temperature_c/config
  Add/update topic: homeassistant/sensor/wmbusmeters/23028459_min_external_temperature_c/config
  Add/update topic: homeassistant/sensor/wmbusmeters/23028459_min_flow_m3h/config
  Add/update topic: homeassistant/sensor/wmbusmeters/23028459_min_flow_temperature_c/config
  Add/update topic: homeassistant/sensor/wmbusmeters/23028459_rssi_dbm/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/23028459_status_burst/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/23028459_status_dry/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/23028459_status_leak/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/23028459_status_reversed/config
  Add/update topic: homeassistant/sensor/wmbusmeters/23028459_target_m3/config
  Add/update topic: homeassistant/sensor/wmbusmeters/23028459_total_m3/config
MQTT Discovery cleanup...
[12:19:52] INFO: Running wmbusmeters ...
[ALARM SpecifiedDeviceNotFound] the device im871a:t1,c1 is not working
No wmbus device detected, waiting for a device to be plugged in.
[ALARM SpecifiedDeviceNotFound] the device im871a:t1,c1 is not working
[ALARM SpecifiedDeviceNotFound] the device im871a:t1,c1 is not working
[ALARM SpecifiedDeviceNotFound] the device im871a:t1,c1 is not working
[ALARM SpecifiedDeviceNotFound] the device im871a:t1,c1 is not working
[ALARM SpecifiedDeviceNotFound] the device im871a:t1,c1 is not working
[ALARM SpecifiedDeviceNotFound] the device im871a:t1,c1 is not working
[ALARM SpecifiedDeviceNotFound] the device im871a:t1,c1 is not working
[ALARM SpecifiedDeviceNotFound] the device im871a:t1,c1 is not working
[ALARM SpecifiedDeviceNotFound] the device im871a:t1,c1 is not working
[ALARM SpecifiedDeviceNotFound] the device im871a:t1,c1 is not working

BIBOLV, 2023-03-10:
Could it be that im871 is under usb2 as you have `"donotprobe":"/dev/ttyUSB2"`? Try to remove `donotprobe` and set `loglevel` to debug.

Mindkey75, 2023-03-10:
I try this and it looks like i works fine. What do you Think i have to do from here

 Mvh Morten Henriksen

BIBOLV, 2023-03-10:
If there is no device that you do not want wmbusmeters to probe then remove `"donotprobe"` line from config, set loglevel back to normal and that's it.

Mindkey75, 2023-03-10:
Thank you 😊 now it is working

 Mvh Morten Henriksen

BIBOLV, 2023-03-10:
Closing as answered.

closed 2023-03-10 by BIBOLV