wmbusmeters/wmbusmeters #909: Itron Cyble HRF 868Mhz V5.2

jonathanD91, 2023-03-10:
### Type of request

Question

### OS version

Linux

### wmbusmeters version

Last oNe

### Your message goes here

Hello, I have an Itron Cyble HRF 868mhz v5.2 water meter.
when listening I do not receive any counter signal. 
the water meter is at 15metres. do you know if my meter is compatible?
THANKS

jonathanD91, 2023-03-11:
i think i have to use c2 protocol but how to install?

jonathanD91, 2023-03-11:
La ligne de debug ```(shell) return code 1
(shell) 2577 exited
(main) rtl_srd found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(serial) EVENT thread interrupted
(shell) output: >>><<<
(shell) return code 0
(shell) 2579 exited
(main) rtl_wmbus found in path
(main) rtlsdr device [00000001](tel:00000001) not currently used.
(serial) EVENT thread interrupted
(main) found specified device (rtlwmbus:868M) that matches detected device (:rtlwmbus[00000001]:0/0)
(main) opening rtlwmbus:868M
Started config rtlwmbus[00000001] listening on t1 using fq 868M
(rtlwmbus) using command: /usr/bin/rtl_sdr  -d 0 -f 868M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus
(rtlwmbus) opening [00000001](tel:00000001)
(serial) EVENT thread interrupted
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "/usr/bin/rtl_sdr  -d 0 -f 868M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus"
(serialcmd) opened /bin/sh pid 2585 fd 5 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop```

jonathanD91, 2023-03-18:
Hi, what i do to run with C2 protocol? I don’t understand…

jonathanD91, 2023-03-26:
Hi, anyone say??

weetmuts, 2023-03-26:
What kind of dongle for receiving/sending radio do you have?

Have you read this?
https://github.com/wmbusmeters/wmbusmeters/issues/346

and made sure its not the Radio protocol?
https://github.com/neutrinus/everblu-meters

jonathanD91, 2023-03-26:
I see in the github you are a configuration for protocol c2, in webmusters.conf bit i don’t say what i write inside

jonathanD91, 2023-04-14:
@weetmuts I still can't get the right configuration for the C2 protocol. The dongle I use is an RTL2332U dongle (Andoer mini portable DVB-T DAB. how should i do?

weetmuts, 2023-04-14:
Sorry, the rtl2332U cannot transmit to the meter. The C2 protocol means that you transmit a message to the meter that then responds with data. An im871a or amb8465 can transmit.

jonathanD91, 2023-04-14:
@weetmuts you confirm this key it’s not good for the protocol c2
Andoer Mini Portable Numérique USB 2.0 TV Stick DVB-T + DAB + FM RTL2832U Support de Puce SDR Tuner Récepteur https://amzn.eu/d/3bdju3j

Which key to buy? You are a model? 
thanks

closed 2023-07-14 by github-actions[bot]

jeedomsteph37, 2024-06-29:
> @weetmuts you confirm this key it’s not good for the protocol c2 Andoer Mini Portable Numérique USB 2.0 TV Stick DVB-T + DAB + FM RTL2832U Support de Puce SDR Tuner Récepteur https://amzn.eu/d/3bdju3j
> 
> Which key to buy? You are a model? thanks

Bonjour,
J'ai le même compteur, avez vous trouvé le bon récepteur et le bon protocole ?
Merci