wmbusmeters/wmbusmeters #913: Techem Gas Meter with incorrect values

almperez, 2023-03-11:
Hi guys,

A Techem Natural Gas meter was installed this week, and I'm trying to get the current consumed values, but I'm receiving some pretty weird values. My computer is running a Kali Linux with wmbusmeters version 1.11.0-146-ge9e6c32 and a Realtek Semiconductor Corp. RTL2838 DVB-T device.

The meter ID is correct, as I could see a label when the technician came here to install it, so I'm not getting the information from another device. The command line is:

`wmbusmeters --verbose --debug --oneshot auto:c1 rtlwmbus:916.4M Gas mkradio4 22250093 NOKEY`

Results:

`└─$ wmbusmeters --verbose --debug --oneshot auto:c1 rtlwmbus:916.4M Gas mkradio4 22250093 NOKEY
(wmbusmeters) version: 1.11.0-146-ge9e6c32
(config) using device: rtlwmbus:916.4M 
(config) number of meters: 1
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers

Usage:   -f frequency_to_tune_to [Hz]
        [-s samplerate (default: 2048000 Hz)]
        [-d device_index (default: 0)]
        [-g gain (default: 0 for auto)]
        [-p ppm_error (default: 0)]
        [-b output_block_size (default: 16 * 16384)]
        [-n number of samples to read (default: 0, infinite)]
        [-S force sync output (default: async)]
        filename (a '-' dumps samples to stdout)

<<<
(shell) return code 1
(shell) 455545 exited
(serial) EVENT thread interrupted
(main) rtl_srd found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(shell) output: >>><<<
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 455547 exited
(main) rtl_wmbus found in path
(main) rtlsdr device 00000001 not currently used.
(main) found specified device (rtlwmbus:916.4M) that matches detected device (:rtlwmbus[00000001]:0/0)
(main) opening rtlwmbus:916.4M
Started config rtlwmbus[00000001] listening on c1 using fq 916.4M
(rtlwmbus) using command: /usr/bin/rtl_sdr  -d 0 -f 916.4M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus
(rtlwmbus) opening 00000001
(serial) EVENT thread interrupted
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "/usr/bin/rtl_sdr  -d 0 -f 916.4M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus"
(serialcmd) opened /bin/sh pid 455553 fd 5 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
(serial) MAIN thread interrupted
(serial) EVENT thread interrupted
(bgshell) 455553 exited with return code 0
(serial) closing non working fd=5 "00000001"
(wmbus) disconnected 00000001 rtlwmbus
(serialcmd) closed /bin/sh pid=455553 fd=5 (rtlwmbus)
(serial) TIMER thread interrupted
(serial) EVENT thread interrupted
Lost 00000001 closing rtlwmbus[00000001]
(wmbus) closing....
(wmbus) deleted rtlwmbus
(serialcmd) closed /bin/sh pid=455553 fd=-1 (rtlwmbus)
(serial) EVENT thread interrupted
[... 114 lines cut ...]
(rtlwmbus) received full frame
(serial) TIMER thread interrupted
(wmbus) skipping already handled telegram leng=22.
(serial) event loop stopped!
(wmbus) deleted rtlwmbus
(serial) closing 1 devices
(shell) sending SIGTERM to process group 455605
(bgshell) sent sigterm, now waiting for child 455605 to exit.
(bgshell) 455605 terminated due to signal 15
(serialcmd) closed /bin/sh pid=455605 fd=5 (rtlwmbus)`

What is the difference between total_m3 and target_m3?
On the picture below, you can see the current metered values. I have left my oven turned on for a while so it went from 2.504 to 2.613, but metered values changed as follows:

`Gas     22250093        1105.4 m³       644.6 m³        2023-03-11 14:13.21
Gas     22250093        1105.4 m³       644.6 m³        2023-03-11 14:14.04
Gas     22250093        1105.4 m³       644.6 m³        2023-03-11 14:15.16
Gas     22250093        1105.4 m³       644.6 m³        2023-03-11 14:16.31
Gas     22250093        1156.6 m³       670.2 m³        2023-03-11 14:17.14
Gas     22250093        1156.6 m³       670.2 m³        2023-03-11 14:18.27
Gas     22250093        1156.6 m³       670.2 m³        2023-03-11 14:19.05
Gas     22250093        1156.6 m³       670.2 m³        2023-03-11 14:20.25
`

Thanks in advance. :)

almperez, 2023-03-11:
[image]
[image]

BIBOLV, 2023-03-11:
Hi, I would double check if those telegrams are from your meter after all. I'm bit suspicious because of FM in it's name as it could point to it not being wmbus compliant but rather some other modulation. Similar issue - https://github.com/wmbusmeters/wmbusmeters/issues/523

almperez, 2023-03-11:
Hi @BIBOLV 

Thanks for your reply. I'm sure that my meter is 22250093. What is the FM you mentioned?

If run the command `wmbusmeters --logtelegrams auto:c1 rtlwmbus:916.4M` I got this reply:

Received telegram from: 22250093
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Warm water (0x62)
                   ver: 0x50
                device: rtlwmbus[00000001]
                  rssi: 94 dBm
                driver: unknown!
telegram=|_15446850930025225062A209006B2E1A000000130000|+291

Can you suggest anything I can check or do? thanks

BIBOLV, 2023-03-11:
With FM I am referring to marking on your techem module - "FM radio 4". If you would check issue I refereed to then there are pictures of modules with two markings "FM radio 4" and "radio 4".
Also type of meter is reported as "Warm water", this of course could be wrongly configured module.
I you can, I would suggest to stop using gas for a bit and check received messages - if readings are raising then it is not your meter.

almperez, 2023-03-11:
I'm sure this is my meter... I left the wmbusmeters running the entire night and it did not rose. The only moment if changed was when I turned on the oven.
Also I took a picture of a label that the technician took with him, and the ID 22250093 was printed there. Also I can see by the RSSI value. Also when I unplug my antenna and take my laptop very near the Gas Meter, this is the only device that is getting telegrams.

In addition, I run `wmbusmeters  --logtelegrams auto:c1,t1,s1,c2,t2,s2 rtlwmbus:916.4M Gas mkradio4 22250093 NOKEY` and it returned:

Started config rtlwmbus[00000001] listening on s1,s2,t1,t2,c1,c2 using fq 916.4M
(meter) Gas: meter detection did not match the selected driver mkradio4! correct driver is: unknown!
(meter) Not printing this warning again for id: 22250093 mfct: (TCH) Techem Service (0x5068) type: Warm water (0x62) ver: 0x50
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|15446850930025225062A2_09006B2E1A000000130000|+87
Gas     22250093        1156.6 m³       670.2 m³        2023-03-11 19:37.15
^C                                                                            

Do you have any additional suggestion

BIBOLV, 2023-03-11:
Ok, it is pretty convincing that it is your meter 😉 
Then I would suggest to add calculation and subtract some amount to get actual value of your meter as output and then check in longer term that it is correct. 
This version could be added to existing mkradio4 driver but it will not fix that reported value does not correspond to your meter reading.

almperez, 2023-03-11:
hahaha :rofl: 

Will do it on the next days and keep monitoring it. 
Do you know what could be the difference between total and target m3?

BIBOLV, 2023-03-11:
Total is total consumption, target is consumption in previous period - usually month.

BIBOLV, 2023-03-12:
You can post 4-5 telegrams with coresponding readings from meter. 
Maybe @weetmuts will have some time some day to check if new Driver is necessary.

almperez, 2023-03-17:
Hi @BIBOLV 
As you requested:

Meter: 3,271
telegram=|15446850930025225062A2_09006E2E20000000190000|+54
Gas     22250093        1463.8 m³       823.8 m³        2023-03-14 12:23.28

Meter: 3,573 or 3,574 (between both values)
telegram=|15446850930025225062A2_09006F2E230000001C0000|+8
Gas     22250093        1617.4 m³       900.6 m³        2023-03-15 13:37.32

Meter: 4,003
telegram=|15446850930025225062A2_0900702E28000000021F00|+36
Gas     22250093        1079.8 m³       1028.6 m³       2023-03-16 14:25.38

Meter: 4,408
telegram=|15446850930025225062A2_0900712E2C000000061F00|+3
Gas     22250093        1284.6 m³       1131 m³ 2023-03-17 17:42.14

Hi @weetmuts,
Can you please take a look when you have some time? thanks :)

mineirinh0, 2023-05-22:
@almperez : did you get you system to read this aepio/techem meter? may you please share a little more?

closed 2023-08-23 by github-actions[bot]