wmbusmeters/wmbusmeters #917: Add new Device: DIEHL AERIUS Gas Meter

Stefan8615, 2023-03-15:
Hi All,

I have a gas meter from DIEHL (Model AERIUS) which is currently not supported:

(meter) Gas: meter detection could not find driver for id: 42093300 mfct: (DME) DIEHL Metering, Germany (0x11a5) type: Gas meter (0x03) ver: 0x30
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/ (meter) to add support for this unknown mfct,media,version combination

What is required to get this meter supported?

Thanks in Advance!

BIBOLV, 2023-03-15:
In order to log telegrams, set `logtelegrams=true` in configuration and gather at least 10 telegrams from meter. Check that telegrams are not encrypted, it can be verified in logs or by pasting them [here](https://wmbusmeters.org/). If telegram is encrypted you need to get encryption key, decrypt messages and them post them as we can not do anything with encrypted ones.

Stefan8615, 2023-03-15:
Thanks for your support. 
here are some readings:

telegram=|2E44A5110033094230037A060020052F2F_0C933E842784060A3B00000A5A5901C4016D3B37DF2CCC01933E24032606|+11
telegram=|5E44A5111630727070077AD50050052F2F_0C13314504004C1330010100CC101300000000CC201330010100426CDF2C0B3B00000002FD74AF12025A6100C4016D3B17EE23CC01135240040001FD085102FD170000EB013B0400F02F2F2F2F2F|+22
telegram=|2E44A5110033094230037A060020052F2F_0C933E912784060A3B60040A5A5901C4016D3B37DF2CCC01933E24032606|+31
telegram=|5E44A5111630727070077AD50050052F2F_0C13314504004C1330010100CC101300000000CC201330010100426CDF2C0B3B00000002FD74AF12025A6100C4016D3B17EE23CC01135240040001FD085202FD170000EB013B0400F02F2F2F2F2F|+37
telegram=|5E44A5111630727070077AD50050052F2F_0C13314504004C1330010100CC101300000000CC201330010100426CDF2C0B3B00000002FD74AF12025A6100C4016D3B17EE23CC01135240040001FD085302FD170000EB013B0400F02F2F2F2F2F|+51
telegram=|5E44A5111630727070077AD50050052F2F_0C13314504004C1330010100CC101300000000CC201330010100426CDF2C0B3B00000002FD74AF12025A6100C4016D3B17EE23CC01135240040001FD085402FD170000EB013B0400F02F2F2F2F2F|+65
telegram=|5E44A5111630727070077AD50050052F2F_0C13314504004C1330010100CC101300000000CC201330010100426CDF2C0B3B00000002FD74AF12025A6200C4016D3B17EE23CC01135240040001FD085502FD170000EB013B0400F02F2F2F2F2F|+79
telegram=|2E44A5110033094230037A060020052F2F_0C933E972784060A3B31080A5A5901C4016D3B37DF2CCC01933E24032606|+91
telegram=|5E44A5111630727070077AD50050052F2F_0C13314504004C1330010100CC101300000000CC201330010100426CDF2C0B3B00000002FD74AF12025A6200C4016D3B17EE23CC01135240040001FD085602FD170000EB013B0400F02F2F2F2F2F|+93
telegram=|5E44A5111630727070077AD50050052F2F_0C13314504004C1330010100CC101300000000CC201330010100426CDF2C0B3B00000002FD74AF12025A6200C4016D3B17EE23CC01135240040001FD085702FD170000EB013B0400F02F2F2F2F2F|+107
telegram=|5E44A5111630727070077AD50050052F2F_0C13314504004C1330010100CC101300000000CC201330010100426CDF2C0B3B00000002FD74AF12025A6200C4016D3B17EE23CC01135240040001FD085902FD170000EB013B0400F02F2F2F2F2F|+122
telegram=|5E44A5111630727070077AD50050052F2F_0C13314504004C1330010100CC101300000000CC201330010100426CDF2C0B3B00000002FD74AF12025A6300C4016D3B17EE23CC01135240040001FD085B02FD170000EB013B0400F02F2F2F2F2F|+136

Stefan8615, 2023-03-15:
The shorter telegrams are the Aerius. The longer ones in between are from a Hydrus which is working fine.

Stefan8615, 2023-03-16:
Thanks @BIBOLV !
I have the wmbusmeters Home Assistant AddOn. How can I test this new driver now?

BIBOLV, 2023-03-16:
When PR will be merged and addon built. but it will take some time - will let you know.

BIBOLV, 2023-03-16:
You can test driver now by installing latest version of edge ha addon.

Stefan8615, 2023-03-16:
Works perfectly - many thanks for the quick support!!!

closed 2023-03-16 by BIBOLV