wmbusmeters/wmbusmeters #921: Service does not start

chrigabr, 2023-03-18:
Hello, I'm on Rasperry PI Zero 2 W. I've been installing wmbusmeters several times. I can't start the service. I don't understand.

root@FramboiseN2F:/opt/wmbusmeters# systemctl status wmbusmeters.service
● wmbusmeters.service - "wmbusmeters service"
     Loaded: loaded (/lib/systemd/system/wmbusmeters.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sat 2023-03-18 07:30:41 UTC; 1min 53s ago
       Docs: https://github.com/wmbusmeters/wmbusmeters
             man:wmbusmeters(1)
    Process: 3734 ExecStartPre=/bin/mkdir -p /var/lib/wmbusmeters/meter_readings (code=exited, status=0/SUCCESS)
    Process: 3735 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/lib/wmbusmeters/meter_readings (code=exited, status=0/SUCCESS)
    Process: 3736 ExecStartPre=/bin/mkdir -p /var/log/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 3737 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/log/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 3738 ExecStartPre=/bin/mkdir -p /run/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 3739 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /run/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 3740 ExecStart=/usr/sbin/wmbusmetersd /run/wmbusmeters/wmbusmeters.pid (code=exited, status=203/EXEC)
        CPU: 352ms

Mar 18 07:30:41 FramboiseN2F systemd[1]: wmbusmeters.service: Scheduled restart job, restart counter is at 3.
Mar 18 07:30:41 FramboiseN2F systemd[1]: Stopped "wmbusmeters service".
Mar 18 07:30:41 FramboiseN2F systemd[1]: wmbusmeters.service: Start request repeated too quickly.
Mar 18 07:30:41 FramboiseN2F systemd[1]: wmbusmeters.service: Failed with result 'exit-code'.
Mar 18 07:30:41 FramboiseN2F systemd[1]: Failed to start "wmbusmeters service".

weetmuts, 2023-03-18:
How did you install wmbusmeters? What does /var/log/wmbusmeters/wmbusmeters.log say?

chrigabr, 2023-03-18:
I used 

git clone https://github.com/wmbusmeters/
./configure; make; sudo make install

var/log/wmbusmeters is empty ! :-( 

Dongle sees work:
root@FramboiseN2F:/opt/wmbusmeters# rtl_sdr -f 433M -s 1600000 - 2>/dev/null | rtl_wmbus -p s
T1;0;0;2023-03-18 07:42:49.000;21;47;000000FF;0x
T1;0;0;2023-03-18 07:45:12.000;25;25;FFFFFFFF;0x0affffffffffffffffffff
T1;0;0;2023-03-18 07:51:29.000;22;52;FFFFFF3A;0x14ffffff3affffffffffffb6ffffffffffffffffff
T1;0;0;2023-03-18 07:52:04.000;30;32;FFFFFFFF;0x8accffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffb1ff22ffffff09ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
T1;0;0;2023-03-18 07:52:09.000;61;18;FF1AFFFF;0x0affffffffff1affffffff
T1;0;0;2023-03-18 07:53:25.000;39;19;00000000;0x

Many thanks!

weetmuts, 2023-03-18:
and /var/log/syslog ?

chrigabr, 2023-03-18:
Then in /var/log/syslog:

Mar 18 07:30:35 FramboiseN2F systemd[1]: Starting "wmbusmeters service"...
Mar 18 07:30:36 FramboiseN2F systemd[3721]: wmbusmeters.service: Failed to execute /usr/sbin/wmbusmetersd: Permission denied
Mar 18 07:30:36 FramboiseN2F systemd[3721]: wmbusmeters.service: Failed at step EXEC spawning /usr/sbin/wmbusmetersd: Permission denied
Mar 18 07:30:36 FramboiseN2F systemd[1]: wmbusmeters.service: Control process exited, code=exited, status=203/EXEC
Mar 18 07:30:36 FramboiseN2F systemd[1]: wmbusmeters.service: Failed with result 'exit-code'.
Mar 18 07:30:36 FramboiseN2F systemd[1]: Failed to start "wmbusmeters service".
Mar 18 07:30:37 FramboiseN2F systemd[1]: wmbusmeters.service: Scheduled restart job, restart counter is at 1.
Mar 18 07:30:37 FramboiseN2F systemd[1]: Stopped "wmbusmeters service".
Mar 18 07:30:37 FramboiseN2F systemd[1]: Starting "wmbusmeters service"...
Mar 18 07:30:38 FramboiseN2F systemd[3731]: wmbusmeters.service: Failed to execute /usr/sbin/wmbusmetersd: Permission denied
Mar 18 07:30:38 FramboiseN2F systemd[3731]: wmbusmeters.service: Failed at step EXEC spawning /usr/sbin/wmbusmetersd: Permission denied
Mar 18 07:30:38 FramboiseN2F systemd[1]: wmbusmeters.service: Control process exited, code=exited, status=203/EXEC
Mar 18 07:30:38 FramboiseN2F systemd[1]: wmbusmeters.service: Failed with result 'exit-code'.
Mar 18 07:30:38 FramboiseN2F systemd[1]: Failed to start "wmbusmeters service".
Mar 18 07:30:39 FramboiseN2F systemd[1]: wmbusmeters.service: Scheduled restart job, restart counter is at 2.
Mar 18 07:30:39 FramboiseN2F systemd[1]: Stopped "wmbusmeters service".
Mar 18 07:30:39 FramboiseN2F systemd[1]: Starting "wmbusmeters service"...
Mar 18 07:30:39 FramboiseN2F systemd[3740]: wmbusmeters.service: Failed to execute /usr/sbin/wmbusmetersd: Permission denied
Mar 18 07:30:39 FramboiseN2F systemd[3740]: wmbusmeters.service: Failed at step EXEC spawning /usr/sbin/wmbusmetersd: Permission denied
Mar 18 07:30:39 FramboiseN2F systemd[1]: wmbusmeters.service: Control process exited, code=exited, status=203/EXEC
Mar 18 07:30:39 FramboiseN2F systemd[1]: wmbusmeters.service: Failed with result 'exit-code'.
Mar 18 07:30:39 FramboiseN2F systemd[1]: Failed to start "wmbusmeters service".
Mar 18 07:30:41 FramboiseN2F systemd[1]: wmbusmeters.service: Scheduled restart job, restart counter is at 3.
Mar 18 07:30:41 FramboiseN2F systemd[1]: Stopped "wmbusmeters service".
Mar 18 07:30:41 FramboiseN2F systemd[1]: wmbusmeters.service: Start request repeated too quickly.
Mar 18 07:30:41 FramboiseN2F systemd[1]: wmbusmeters.service: Failed with result 'exit-code'.
Mar 18 07:30:41 FramboiseN2F systemd[1]: Failed to start "wmbusmeters service".

weetmuts, 2023-03-18:
ls -l /usr/sbin/wmbusmetersd
ls -l /usr/sbin/wmbusmeters

says what?

weetmuts, 2023-03-18:
ls -l /usr/bin/wmbusmeters

chrigabr, 2023-03-18:
cgabriel@FramboiseN2F:~ $ ls -l /usr/sbin/wmbusmeters
ls: cannot access '/usr/sbin/wmbusmeters': No such file or directory

cgabriel@FramboiseN2F:~ $ ls -l /usr/sbin/wmbusmetersd
lrwxrwxrwx 1 root root 18 Mar 17 23:16 /usr/sbin/wmbusmetersd -> ../bin/wmbusmeters

cgabriel@FramboiseN2F:~ $ ls -l /usr/bin/wmbusmeters
total 1692
drwxr-xr-x  2 root root    4096 Mar 17 19:34 autoconf
drwxr-xr-x  3 root root   20480 Mar 17 21:10 build
-rw-r--r--  1 root root   39439 Mar 17 19:34 CHANGES
-rwxr-xr-x  1 root root     143 Mar 17 19:34 check.sh
-rw-r--r--  1 root root   12813 Mar 17 21:15 config.log
-rwxr-xr-x  1 root root  134752 Mar 17 19:34 configure
-rw-r--r--  1 root root    2035 Mar 17 19:34 configure.ac
drwxr-xr-x  3 root root    4096 Mar 17 19:34 deb
-rw-r--r--  1 root root   57074 Mar 17 19:34 DLMS_Flagids.csv
drwxr-xr-x  2 root root    4096 Mar 17 19:34 doc
drwxr-xr-x  2 root root    4096 Mar 17 19:34 docker
drwxr-xr-x  4 root root    4096 Mar 17 19:34 fuzz_testcases
drwxr-xr-x  3 root root    4096 Mar 17 19:34 ha-addon
-rw-r--r--  1 root root    4084 Mar 17 19:34 HowToAddaNewMeter.txt
-rwxr-xr-x  1 root root    1165 Mar 17 19:34 install.sh
-rw-r--r--  1 root root   35147 Mar 17 19:34 LICENSE
-rw-r--r--  1 root root   16392 Mar 17 19:34 Makefile
-rw-r--r--  1 root root   40177 Mar 17 19:34 README.md
-rw-r--r--  1 root root     114 Mar 17 19:34 repository.json
drwxr-xr-x  2 root root    4096 Mar 17 19:34 scripts
drwxr-xr-x  2 root root    4096 Mar 17 19:34 simulations
drwxr-xr-x  3 root root    4096 Mar 17 19:34 snap
drwxr-xr-x  2 root root    4096 Mar 17 19:34 src
drwxr-xr-x 14 root root    4096 Mar 17 19:34 tests
-rwxr-xr-x  1 root root    4551 Mar 17 19:34 test.sh
-rwxr-xr-x  1 root root    1651 Mar 17 19:34 uninstall.sh
drwxr-xr-x  2 root root    4096 Mar 17 19:34 utils
-rwxr-xr-x  1 root root 1265576 Mar 17 23:16 wmbusmeters
-rw-r--r--  1 root root   10277 Mar 17 19:34 wmbusmeters.1
-rw-r--r--  1 root root     690 Mar 17 19:34 wmbusmeters-admin.1

weetmuts, 2023-03-18:
you have managed to put the build directory in /usr/bin
do
mv /usr/bin/wmbumeters /home/yourhome
cd /home/yourhome/wmbusmeters
and redo
sudo make install

chrigabr, 2023-03-18:
OK! WMBUSMETERS starts now. coooool!

One more question :-) it seems that he does not receive anything. I think it comes from the frequency. My water meter is a SENSUS iPERL which communicates at 433Mhz. But I believe that the original WMBUSMETERS is in 868.95MHz. How can I change the frequency? THANKS

 cgabriel@FramboiseN2F:~ $ wmbusmeters auto:t1
Started auto rtlwmbus[MyDongleId] listening on t1
No meters configured. Printing id:s of all telegrams heard!

chrigabr, 2023-03-19:
Hello, after spending quite a bit of time on ISSUES I managed to find the command that works. On the other hand, you really have to be very, very patient... for once, almost 12 hours.

So with a Sensus iPERL meter here is the online command that works:
cgabriel@FramboiseN2F:~ $ wmbusmeters --logtelegrams --format=json auto:t1 watermeter iperl 00211276 E6C8                                                               8800DEB868C0D6A84880CE982840
Started auto rtlwmbus[MyDongleId] listening on t1
telegram=|1E44AE4C7612210068077AF30010052F2F_0413DFCF0300023B00002F2F2F2F|+5086
{"media":"water","meter":"iperl","name":"watermeter","id":"00211276","total_m3":249.823,"max_flow_m3h":0,                                                               "timestamp":"2023-03-19T20:53:48Z","device":"rtlwmbus[MyDongleId]","rssi_dbm":125}

Thanks for your help!

closed 2023-03-19 by chrigabr

weetmuts, 2023-03-20:
I happy it worked for you!