wmbusmeters/wmbusmeters #928: Sensus water meter

pbasista, 2023-03-25:
### Type of request

New device support

### Meter description

Hi, I have noticed a few encrypted telegrams (1 per hour) from an unknown Sensus water meter that is probably located somewhere nearby. I am unsure which specific device sends it. Wmbusmeters is unable to pick any driver for decoding its telegrams.

Maybe someone with a similar Sensus water meter and a decryption key could analyze its telegrams and add support for this particular meter. Perhaps there is a default key which these meters are typically configured to use. Issues #877 and #878 mention the default key `E6C88800DEB868C0D6A84880CE982840` for the Sensus iPERL meter. But it fails to decode these telegrams.

### Logged telegrams from meter in decrypted format

```shell
Encrypted telegrams:

2644AE4C2527020038387225270200AE4C0B07A8001005BCC346BC48D7B1583DCED6A1436F460F
2644AE4C2527020038387225270200AE4C0B07A9001005074B13C9D80D579829A4ABCE4FCA5B26
2644AE4C2527020038387225270200AE4C0B07AA0010050624511D53759201DFB318323F1AC858
2644AE4C2527020038387225270200AE4C0B07AB0010054B86549BA6EBE333C5D52152C47B820B
2644AE4C2527020038387225270200AE4C0B07AD0010058C725994DEA20F289CEFA52D0E6700E0
2644AE4C2527020038387225270200AE4C0B07AE001005508BD3169A57C36F02ECA2F179130CA7
2644AE4C2527020038387225270200AE4C0B07AF00100543BC3F56C590924A485381924FBE2C55
2644AE4C2527020038387225270200AE4C0B07B000100526B5BE8A5DF0D04EE452B4F03AD1BAE8
```

chrigabr, 2023-03-25:
Hello,

I also have a Sensus Iperl meter. Here is the command that works to see the decoded message:
wmbusmeters --logtelegrams --format=json auto:t1 watermeter iperl 00211276 E6C88800DEB868C0D6A84880CE982840

00211276 is my meter id.  Can you test?

I also receive very few messages. At best one per hour. So I wonder if I'm on the best frequency.

Christian GABRIEL

pbasista, 2023-03-25:
> I also have a Sensus Iperl meter.

The meter I have received the telegrams from is almost certainly not an iPERL meter. The reason is that, as I have mentioned in the issue's description, neither the driver's autodetection, nor the default decryption key work with these telegrams.

thecem, 2023-03-27:
@chrigabr which one do you have? This one?

https://www.xylem.com/de-de/brands/sensus/sensus-produkte/iperl-international-water-meter/

thecem, 2023-03-27:
@BIBOLV standard log is set in addon to `/dev/stdout`. Is there an option to log also in a file like `conf/wmbusmeters/logs/logfile.log` ?

Thanks!

chrigabr, 2023-03-27:
@thecem yes ! 
[image]

thecem, 2023-03-27:
ok, which frequenz do your meter use 433 or 868? What for a device do you use?

chrigabr, 2023-03-27:
@thecem I used 868MHz, but most messages have CRC errors. I think I'm going to have to invest in a IM871A-USB which apparently uses much less processor.

thecem, 2023-03-27:
so at the moment a sdr? which settings?

chrigabr, 2023-03-28:
Hi @thecem,

Below are my settings but the readings are random....
[image]
[image]

BIBOLV, 2023-03-31:
> @BIBOLV standard log is set in addon to `/dev/stdout`. Is there an option to log also in a file like `conf/wmbusmeters/logs/logfile.log` ?

Yes, you can change `logfile` parameter to for example `/config/wmbusmeters/logs/wmbusmeters.log`, but you need to create log file manually, otherwise addon will not start.

thecem, 2023-04-01:
thx. I assume to have the 433 version, but I was not able to receive any only one telegram. I use also a sdr with config like chrigabr.

@weetmuts: ist it possible to retrieve kind different format instead of wmbus, since it could be they sends only “sensusrf”.
Is there a possibility to get the information by using some like a spectrum analyser with the sdr?

thecem, 2023-04-01:
@BIBOLV you are right it dosn´t start any more since the path is wrong! I have created the file. But when the webserver is not runing to change the config how could I change the config???

thecem, 2023-04-01:
so the question is which value we have to config to point to \config\wmbusmeters\logs\wmbus.log ??

[image]

thecem, 2023-04-01:
@weetmuts I get the information that on freq 434.475M I should able to receive telegramms
So I cinfigured `rtlwmbus:CMD(rtl_sdr -f 434.475M -s 1600000 - 2>/dev/null | rtl_wmbus -p   s -a)` is this correct, since I get no telegramms, on 868,9 I get my heat energy meter, but nothing for the sensus ipearl.

weetmuts, 2023-04-01:
@thecem you have backslashes \ in the path. :-) try with normal slashes instead. /

thecem, 2023-04-01:
Thanks I did now I could log.

this was the mandatory point to log all telegramms over a long time since the HA log is not long enough.

thecem, 2023-04-01:
Could this be that the meter sends telegrams in diffrent formats like w-m-bus and sensus rf?

```
(main) regular reset of rtlwmbus  using CMD(rtl_sdr -f 434.475M -s 1600000 - 2>/dev/null | rtl_wmbus -p   s -a) will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) received ascii "T1;0;0;2023-04-01 08:28:51.000;7;33;00000000;0x<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2023-04-01 08:28:51.000;7;33;00000000;0x<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
```

BIBOLV, 2023-04-01:
> @BIBOLV you are right it dosn´t start any more since the path is wrong! I have created the file. But when the webserver is not runing to change the config how could I change the config???

Config file is accessible from supervisor and located in `/mnt/data/supervisor/addons/data/[imageid]_wmbusmeters-ha-addon`
In order to access supervisor, you need to connect monitor to your HA device.

thecem, 2023-04-08:
@BIBOLV A second device is not accepted, how to configure a second device?

I have to configure:
device=rtlwmbus:CMD(rtl_sdr -f 434.475M -s 1600000 - 2>/dev/null | rtl_wmbus -p   s -a)
and
device=/dev/serial/by-id/usb-Silicon_Labs_WiMOD_iM871A-usb_02759383-if00-port0:im871a:c1,t1

BIBOLV, 2023-04-08:
In one line with `;` delimiter - in your case it would be `device`=`rtlwmbus:CMD(rtl_sdr -f 434.475M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a);/dev/serial/by-id/usb-Silicon_Labs_WiMOD_iM871A-usb_02759383-if00-port0:im871a:c1,t1`

thecem, 2023-04-08:
Thx! 

I solved it with an alias `device=MAIN`

closed 2023-07-08 by github-actions[bot]

mlichtronix, 2025-12-05:
Sensus Pulse RF:
https://wmbusmeters.org/analyze/2644ae4c2527020038387225270200ae4c0B07B1001005e1dec5275cf03790d5f0fda8fB54ffde:auto:0E9FACC6A0E1EF1C3B27A49BAC216535