wmbusmeters/wmbusmeters #929: No

uwec, 2023-03-25:
### Type of request

Question

### HAOS version

Home Assistant 2023.3.6 Supervisor 2023.03.2 Operating System 9.5 Frontend 20230309.1 - latest

### wmbusmeters version

1.12.0-13

### Your message goes here

I have run the HA Addon correctly.

The configuration I do via the web interface:
[image]

In case of a misconfiguration , e.g. "device type = **A**uto:c1", wmbusmeter stops with an error in the log for sure.
`[16:04:27] INFO: Starting web configuration service.
[16:04:27] INFO: Running wmbusmeters ...
Not a valid device "Auto:c1"
s6-rc: info: service legacy-services: stopping
[15:04:27] INFO: Service restart after closing
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
`

After stop, also the web interface is stopped and no change via the web interface is possible anymore.

Using instead edting and correcting the wrong configured device in the "wmbusmeter.conf"  and save the file.
[image]

But after restarting wmbusmaster, somehow the edited wmbusmeter.conf is not (!) loaded, and the old wrong configuration is still considered  instead.

So no correction is possible, and only a uninstall and reinstall of Addon fixes the issue.

BIBOLV, 2023-03-25:
In such case `/data/options_custom.json` file should be edited manually - it contains configuration that both web interface uses and is being used for actual config generation.

uwec, 2023-03-25:
Thanks for the reply, but I do not see any path or file `/data/options_custom.json` in my wmbusmerter folder ?!  Only `/logs/meter_readings/`
`/etc/wmbusmeters.conf`
`/etc/wmbusmeters.d/`
`/etc/mqtt_discovery`

BIBOLV, 2023-03-25:
Looking from HA side it is located in `/mnt/data/supervisor/addons/data/c8a990ad_wmbusmeters-ha-addon` where `c8a990ad` is image id that could be different in each case.

uwec, 2023-03-26:
Seems like I need more deeper access to the system as like I have via the Terminal&SSH addon?! 
Because as well I do not see such folders.

Only possibility for my skills is then to use the work around w/ unistalling and reinstalling the wmbusmeters addon.

closed 2023-03-26 by uwec