wmbusmeters/wmbusmeters #934: Serial TTL Device is not detected - libmbus is working fine

NilsRo, 2023-03-29:
### Type of request

Bug

### OS version

Debian 11

### wmbusmeters version

https://github.com/wmbusmeters/wmbusmeters/commit/c865e88ccf26d23247b8cab114fe0f00068e6bed

### Your message goes here

Hi,

please excuse if I forgot something as I am very new to MBus and software around. I setup a new network for my heatmeters. In the beginning I thought the converter is broken but it seems that wmbusmeters cannot use it. [Converter](https://de.aliexpress.com/item/1005004667031804.html?spm=a2g0o.order_detail.order_detail_item.3.24db6368fkNlgQ&gatewayAdapt=glo2deu) bought.

It connects on /dev/ttyUSB0 as a simple serial converter: Bus 005 Device 004: ID 067b:23d3 Prolific Technology, Inc. USB-Serial Controller
I can use it with libmbus without issues which leads into the question if there is an issue with wmbusmeters.

Version does output: _

linmbus is working fine:
[image]

wmbusmeters.conf
```
loglevel=debug
# Remember to change auto here to the device you are going to use in production.
# device=auto:t1
device=MAIN=/dev/ttyUSB0:mbus:2400
logtelegrams=true
format=json
meterfiles=/var/lib/wmbusmeters/meter_readings
meterfilesaction=overwrite
logfile=/var/log/wmbusmeters/wmbusmeters.log
```

Logoutput:
```
(wmbusmeters) logging started 2023-03-29 13:02:09 using _
(wmbusmeters) version: _
(config) store meter files in: "/var/lib/wmbusmeters/meter_readings"
(config) using device: MAIN=/dev/ttyUSB0:mbus:2400
(config) number of meters: 0
(lookup) with file/hex "/dev/ttyUSB0" mbus
(lookup) driver: mbus
(main) opening MAIN=/dev/ttyUSB0:mbus:2400
[2023-03-29_13:02:09] Started config MAIN=mbus on /dev/ttyUSB0
(mbus) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE|PARENB) local() special_chars(2) modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B2400) input() output() control(CLOCAL|CREAD|CSIZE|PARENB) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 4 (mbus)
Sending NKE to mbus MAIN
(serial /dev/ttyUSB0) sent "1040004016"
(serial) EVENT thread interrupted
(main) regular reset of mbus  on /dev/ttyUSB0 will happen every 82800 seconds
(wmbus) no alarm (expected activity) for mbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(wmbusmeters) waiting for telegrams
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
[2023-03-29_13:02:12] (memory) rss 2932736 peak 2.79 MiB
(serial) stopping manager
(serial) MAIN thread interrupted
(serial) closing 1 devices
(wmbus) disconnected /dev/ttyUSB0 mbus
(serialtty) closed /dev/ttyUSB0 (mbus)
(serial) TIMER thread interrupted
(serial) event loop stopped!
(wmbusmeters) shutting down
(wmbus) deleted mbus
(serial) closing 1 devices
(wmbusmeters) logging started 2023-03-29 20:28:22 using _
(wmbusmeters) version: _
(config) store meter files in: "/var/lib/wmbusmeters/meter_readings"
(config) using device: MAIN=/dev/ttyUSB0:mbus:2400
(config) number of meters: 0
(lookup) with file/hex "/dev/ttyUSB0" mbus
(lookup) driver: mbus
(main) opening MAIN=/dev/ttyUSB0:mbus:2400
[2023-03-29_20:28:22] Started config MAIN=mbus on /dev/ttyUSB0
(mbus) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B2400) input() output() control(CLOCAL|CREAD|CSIZE|PARENB) local() special_chars(3) modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B2400) input() output() control(CLOCAL|CREAD|CSIZE|PARENB) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 4 (mbus)
Sending NKE to mbus MAIN
(serial /dev/ttyUSB0) sent "1040004016"
(serial) EVENT thread interrupted
(main) regular reset of mbus  on /dev/ttyUSB0 will happen every 82800 seconds
(wmbus) no alarm (expected activity) for mbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(wmbusmeters) waiting for telegrams
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
[2023-03-29_20:28:25] (memory) rss 3026944 peak 2.88 MiB
(serial) stopping manager
(serial) MAIN thread interrupted
(serial) EVENT thread interrupted
(serial) closing 1 devices
(serial) event loop stopped!
(serial) TIMER thread interrupted
(wmbus) disconnected /dev/ttyUSB0 mbus
(serialtty) closed /dev/ttyUSB0 (mbus)
(wmbusmeters) shutting down
(wmbus) deleted mbus
(serial) closing 1 devices
```

I had an issue with the baudrate with libmbus as my meters does support 2400baud only. But it is still set.

Many thanks!

NilsRo, 2023-03-29:
Solved myself. Used libmbus to read out the ids and created devices. I thought from the message that all devices are scanned in the first step as all telegrams are print out.

closed 2023-03-29 by NilsRo

weetmuts, 2023-03-30:
Yes, it is indeed a good idea to add the functionality to scan the mbus to wmbusmeters!
I just have not had the time to do it!

NilsRo, 2023-03-30:
Yes, misinterpret the message but would be nice for the initial setup. 

Actually libmbus could be easily used for it. So perhaps you link it in the docs so there is no need to do that work twice. 

And many thanks for the software, MQTT is running and after understanding everything it was a very easy setup!