wmbusmeters/wmbusmeters #942: Unsupp. Meter from Meßtechnik HMM-2 or BMT HYDROCAL-M4

killer0071234, 2023-04-13:
### Type of request

New device support

### Meter description

Hi,

I got a new heat meter with wmbus which is not supported by wmbusmeters.
It's from `Meßtechnik` the type `HMM-2`.
[Datasheet Meßtechnik](https://www.messtechnik.at/wordpress/wp-content/uploads/2022/12/HMM-2_manual_de.pdf)

The Meter seems to be a relabeled type from BMT / BMETERS acc. the output of the HA addon:
`
(meter) Heizung: meter detection did not match the selected driver c5isf! correct driver is: unknown!
(meter) Not printing this warning again for id: 05040538 mfct: (BMT) BMETERS, Italy (0x9b4) type: Heat/Cooling load meter (0x0d) ver: 0x1a
`

The driver `c5isf` returns the best result for me.

The original type could be `BMT / BMETERS HYDROCAL-M4`
[Datasheet BMT / BMETERS](https://www.bmeters.com/wp-content/uploads/2022/06/v1.2_HYDROCAL-M4_UM_EN.pdf)

BR Daniel

### Logged telegrams from meter in decrypted format

```shell
2C44B409380504051A0D8C208A7ACE000020_046D0CB6ED240C03267300000C13130800000F6300000000000000
2C44B409380504051A0D8C208B7ACF000020_046D10B6ED240C03287300000C13150800000F6300000000000000
2C44B409380504051A0D8C208C7AD0000020_046D13B6ED240C03397300000C13220800000F6300000000000000
2C44B409380504051A0D8C208D7AD1000020_046D13B6ED240C03397300000C13220800000F6300000000000000
2C44B409380504051A0D8C208E7AD2000020_046D16B6ED240C03477300000C13270800000F6300000000000000
2C44B409380504051A0D8C208F7AD3000020_046D1AB6ED240C03477300000C13270800000F6300000000000000
2C44B409380504051A0D8C20937AD7000020_046D27B6ED240C03477300000C13270800000F6300000000000000
2C44B409380504051A0D8C20947AD8000020_046D2AB6ED240C03477300000C13270800000F6300000000000000
2C44B409380504051A0D8C20957AD9000020_046D2EB6ED240C03477300000C13270800000F6300000000000000
2C44B409380504051A0D8C20967ADA000020_046D31B6ED240C03477300000C13270800000F6300000000000000
2C44B409380504051A0D8C20977ADB000020_046D34B6ED240C03477300000C13270800000F6300000000000000
2C44B409380504051A0D8C20987ADC000020_046D38B6ED240C03477300000C13270800000F6300000000000000
2C44B409380504051A0D8C20997ADD000020_046D3BB6ED240C03477300000C13270800000F6300000000000000
2C44B409380504051A0D8C209A7ADE000020_046D02B7ED240C03477300000C13270800000F6300000000000000
```

BIBOLV, 2023-04-22:
Hi,
Support implemented, please test.

killer0071234, 2023-05-16:
Thanks for the implementation!
It works like a charm!

I had added a new mqtt discovery file `hydrocalm3.json` to get the values autocreated in HA.
I will made a PR to add it.