wmbusmeters/wmbusmeters #944: Calculation errors

marcelstoer, 2023-04-17:
### Type of request

Question

### OS version

Raspian bullseye

### wmbusmeters version

1.12.0-20-gf1c6423

### Your message goes here

I have this data from my meter

```json
{
	"media": "warm water",
	"meter": "izar",
	"name": "MyGasMeter",
	"id": "875103bf",
	"total_m3": 11.367,
	"last_month_total_m3": 11.09,
	"last_month_measure_date": "2022-11-30",
	"remaining_battery_life_y": 2.5,
	"current_alarms": "underflow,submarine",
	"previous_alarms": "leakage,sensor_fraud",
	"transmit_period_s": 8,
	"timestamp": "2023-04-17T06:58:51Z",
	"device": "im871a[00103852]",
	"rssi_dbm": -93
}
```

Due to the issues described at https://github.com/wmbusmeters/wmbusmeters/issues/712#issuecomment-1510785251 I started experimenting with calculated fields.

- `--calculate_corrected_total_m3='total_m3'` works which tells me that data extraction from the `total_m3` field is fine
- `--calculate_corrected_total_m3='total_m3*1000'` yields "void FormulaImplementation::doMultiplication(): Assertion `op_stack_.size() >= 2' failed." --> which of the two factors is not on the stack?
- `--calculate_corrected_total_m3='total_m3 - 1'` yields "NumericFormula* FormulaImplementation::top2Op(): Assertion `op_stack_.size() > 1' failed." --> again, which factor is missing on the stack?
- `--calculate_corrected_total_m3='total_m3*last_month_total_m3'` does not fail but yields `"corrected_total_m3":null`

weetmuts, 2023-04-18:
Every constant value like 1 or 1000 must be followed by a unit. Bad error message indeed.

Try this instead:
--calculate_corrected_total_m3='total_m3*1000 factor' 
(there are two dimensionless units factor and counter. Do wmbusmeters --listunits)

Likewise add m3 to subtract cubic meters from total_m3. Automatic unit conversion works
so you can subtract 1 l (for 1 liter as well)

--calculate_corrected_total_m3='total_m3 - 1 m3'

This will never work since the result of the multiplication generates the unit m^6
which cannot be converted back to a readable unit, definitely not m3. :-)
--calculate_corrected_total_m3='total_m3*last_month_total_m3'

Operator precedence is not yet implemented so use parentheses if you mix additions and multiplications.

weetmuts, 2023-04-18:
I just pushed a fix so a good error message i printed and no assert is thrown for the missing unit for constant.

marcelstoer, 2023-04-18:
Thank you! Fixed by fa72502.

closed 2023-04-18 by marcelstoer