wmbusmeters/wmbusmeters #945: pollinterval broken

marcelstoer, 2023-04-17:
### Type of request

Bug

### OS version

Raspian bullseye

### wmbusmeters version

1.12.0-20-gf1c6423

### Your message goes here

In my setup the `pollinginterval` parameter appears to be ignored. I first added it to meter file as shown at https://github.com/wmbusmeters/wmbusmeters#example-wmbusmeterconf-file. When that didn't help I moved it to the global config like so:

```
loglevel=normal
# You can use auto:t1 to find the device you have connected to your system.
# But do not use auto here since it will cause unnecessary and slow probing of the serial ports.
device=/dev/ttyUSB0:im871a:c1,t1
# But do not probe this serial tty.
donotprobe=/dev/ttyACM2
logtelegrams=true
format=json
meterfiles=/var/lib/wmbusmeters/meter_readings
meterfilesaction=append
meterfilesnaming=name
meterfilestimestamp=day
logfile=/var/log/wmbusmeters/wmbusmeters.log
#shell=/usr/bin/mosquitto_pub -h localhost -t wmbusmeters/$METER_ID -m "$METER_JSON"
#alarmshell=/usr/bin/mosquitto_pub -h localhost -t wmbusmeters_alarm -m "$ALARM_TYPE $ALARM_MESSAGE"
#alarmtimeout=1h
#alarmexpectedactivity=mon-sun(00-23)
ignoreduplicates=true
pollinterval=60s
```

Yet, I still see an entry appended to the readings file every ~10s. Also, I restarted the daemon every time I changed the config.

BIBOLV, 2023-04-17:
Polling is meant for an C2/T2/S2 wireless meters or an wired m-bus meters.
C1/T1 meters will send messages without polling and will be received by wmbusmeters whenever they are sent.

marcelstoer, 2023-04-17:
Oh, that explains why I was confused to even _see_ a polling parameter 😕 Maybe I should read up on what those C1/T1/C2/T2/S2 mean, sorry.

Is there way I can tell wmbusmeters to ignore C1/T1 datagrams within a certain time window thereby achieving something similar as a "polling" interval?

weetmuts, 2023-04-17:
There have been discussions in a different issue on the concept of delaying the shell command, or only
executing the shell command with the collected data so far, once per hour.

https://github.com/wmbusmeters/wmbusmeters/issues/591

Do you want to join that issue and close this one?

marcelstoer, 2023-04-17:
Thanks Fredrik

closed 2023-04-17 by marcelstoer