wmbusmeters/wmbusmeters #947: Techem radio 4 meters

Dillton, 2023-04-19:
### Type of request

Question

### HAOS version

10

### wmbusmeters version

1.12

### Your message goes here

Hi.

I have the Techem radio 4 meters so I have configured the addon in HA with mkradio4 however in log is mentioned that its wrong driver and I shoud use mkradio4a but this driver is not present on my installation.

I found it on https://wmbusmeters.org/ in drivers list so I assume it must be somewhere.
The question is where can I get mkradio4a driver?

Thanks in advance.

BIBOLV, 2023-04-19:
Hi,
All drivers are present in ha-addon - in meter driver field just put in `mkradio4a`.

Dillton, 2023-04-19:
Ah sorry my bad its working it was related to mqtt_discovery.
There was mkradio4a missing so I have made a renamed copy of mkradio4 and seems its fine now.
Also it looks like I will need to get decryption keys :(

Anyway thank you for fast reaction :)

closed 2023-04-19 by Dillton