wmbusmeters/wmbusmeters #948: GLIBC not found after upgrade via snapd

raszuk, 2023-04-20:
### Type of request

Bug

### wmbusmeters version

Upgrade from 1.5 to 1.12

### Your message goes here

After successful error free upgrade to 1.12.0 via snapd I am not able to run wmbusmeter. 

Upgrade went smooth:

robert@nuc1eo:~/wmbus/wmbusmeters_2023_04$ sudo snap install wmbusmeters
wmbusmeters 1.12.0 from Fredrik Öhrström (oehrstroem) installed
robert@nuc1eo:~/wmbus/wmbusmeters_2023_04$

Now when trying to run it I am getting: 

robert@nuc1eo:/snap/wmbusmeters/current/usr/bin$ ./wmbusmeters -h
./wmbusmeters: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by ./wmbusmeters)
./wmbusmeters: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by ./wmbusmeters)
./wmbusmeters: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by ./wmbusmeters)
./wmbusmeters: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by ./wmbusmeters)
robert@nuc1eo:/snap/wmbusmeters/current/usr/bin$

Attempt to update glibc-source did not help as it comes with 2.31: 

robert@nuc1eo:/snap/wmbusmeters/2315/usr/bin$ sudo apt install glibc-source
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libfwupdplugin1 linux-headers-5.4.0-125 linux-headers-5.4.0-125-generic linux-headers-5.4.0-136 linux-headers-5.4.0-136-generic linux-image-5.4.0-125-generic
  linux-image-5.4.0-136-generic linux-modules-5.4.0-125-generic linux-modules-5.4.0-136-generic linux-modules-extra-5.4.0-125-generic
  linux-modules-extra-5.4.0-136-generic
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  glibc-source
0 upgraded, 1 newly installed, 0 to remove and 100 not upgraded.
Need to get 18,3 MB of archives.
After this operation, 24,1 MB of additional disk space will be used.
Get:1 http://pl.archive.ubuntu.com/ubuntu focal-updates/universe amd64 glibc-source all 2.31-0ubuntu9.9 [18,3 MB]
Fetched 18,3 MB in 0s (54,1 MB/s)
Selecting previously unselected package glibc-source.
(Reading database ... 277955 files and directories currently installed.)
Preparing to unpack .../glibc-source_2.31-0ubuntu9.9_all.deb ...
Unpacking glibc-source (2.31-0ubuntu9.9) ...
Setting up glibc-source (2.31-0ubuntu9.9) ...
robert@nuc1eo:/snap/wmbusmeters/2315/usr/bin$

BIBOLV, 2023-04-22:
wmbusmeter in snap is intended to be used as a service - not executed directly. Please check [docs](https://wmbusmeters.github.io/wmbusmeters-wiki/SNAP.html) and verify that you are able to start as service.

raszuk, 2023-04-28:
I get it now. 

One question before we close this one - which is the proper wmbusmeters.conf to use as they are two ? 

/snap/wmbusmeters/current/etc
or
/var/snap/wmbusmeters/common/etc

Same for meter definitions ? Where should they reside in snap world ? 

Many thx

BIBOLV, 2023-04-29:
`/var/snap/wmbusmeters/common/etc/` for configurations and `/var/snap/wmbusmeters/common/logs/` for log files. Meter files goes in `/var/snap/wmbusmeters/common/etc/wmbusmeters.d/` - as in standard wmbusmeters instalation.

raszuk, 2023-04-29:
Cool thx ! So maybe it would be better to get rid of /snap/wmbusmeters/current/etc to avoid confusion ?

raszuk, 2023-04-30:
Btw after migrating to snapd based wmbusmeters I noticed that only 2 out of 15 meters are read. I did not touch radio nor meter files. In log there is no trace that those missing 13 meters are ignored.  The two are read fine every 15 sec. 

Interesting enough only 2 read are those with AES key to be non zero. Rest 13 have AES key = all zeros. Is there some new code in wmbusmeters which now ignored telegrams from meters with no encryption ? 

Many thx !

BIBOLV, 2023-04-30:
IMO both are being mounted by snap automatically.

closed 2023-04-30 by raszuk