wmbusmeters/wmbusmeters #950: Error compile

fpalladino, 2023-04-28:
### Type of request

Bug

### OS version

CentOS Linux release 7.9.2009

### wmbusmeters version

1.12.00

### Your message goes here

Hi,
configure script works and during make compile appare the below error

Thanks in advance

```
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/util.cc -MMD -c -o build/util.o
src/util.cc: In function ‘std::string strdate(double)’:
src/util.cc:1368:16: error: call of overloaded ‘isnan(double&)’ is ambiguous
     if (isnan(v)) return "null";
                ^
src/util.cc:1368:16: note: candidates are:
In file included from /usr/include/features.h:375:0,
                 from /usr/include/signal.h:28,
                 from src/util.h:21,
                 from src/util.cc:18:
/usr/include/bits/mathcalls.h:235:1: note: int isnan(double)
 __MATHDECL_1 (int,isnan,, (_Mdouble_ __value)) __attribute__ ((__const__));
```

weetmuts, 2023-04-29:
Can you see if the problem goes away if you change all isnan(...) to ::isnan(...)
there seems to be three calls to isnan.

fpalladino, 2023-04-29:
Hi, 
now compile but I have another error.
I tried to install lusb but yum not found the package

```
$ sudo yum install lusb
Plugin abilitati:fastestmirror
Loading mirror speeds from cached hostfile
 * base: it1.mirror.vhosting-it.com
 * epel: it1.mirror.vhosting-it.com
 * extras: it1.mirror.vhosting-it.com
 * updates: it1.mirror.vhosting-it.com
Nessun pacchetto lusb disponibile.
Errore: Niente da fare

```

```
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/driver_waterstarm.cc -MMD -c -o build/driver_waterstarm.o
echo 'R"MANUAL(' > build/short_manual.h
sed -n '/wmbusmeters version/,/```/p' README.md \
| grep -v 'wmbusmeters version' \
| grep -v '```' >> build/short_manual.h
echo ')MANUAL";' >> build/short_manual.h
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/main.cc -c -E > build/main.o.src
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Wno-unused-function -Ibuild src/main.cc -MMD -c -o build/main.o
g++ -o build/wmbusmeters.g build/aes.o build/aescmac.o build/bus.o build/cmdline.o build/config.o build/dvparser.o build/formula.o build/mbus_rawtty.o build/metermanager.o build/meters.o build/manufacturer_specificities.o build/printer.o build/rtlsdr.o build/serial.o build/shell.o build/sha256.o build/threads.o build/translatebits.o build/util.o build/units.o build/wmbus.o build/wmbus_amb8465.o build/wmbus_im871a.o build/wmbus_cul.o build/wmbus_rtlwmbus.o build/wmbus_rtl433.o build/wmbus_simulator.o build/wmbus_rawtty.o build/wmbus_rc1180.o build/wmbus_utils.o build/lora_iu880b.o  build/driver_sharky774.o build/driver_bfw240radio.o build/driver_ultrimis.o build/driver_em24.o build/driver_itron.o build/driver_kamheat.o build/driver_ehzp.o build/driver_unismart.o build/driver_fhkvdataiii.o build/driver_apatoreitn.o build/driver_vario451mid.o build/driver_esyswm.o build/driver_qheat.o build/driver_dme_07.o build/driver_supercom587.o build/driver_mkradio4.o build/driver_enercal.o build/driver_lansendw.o build/driver_multical21.o build/driver_topaseskr.o build/driver_izar.o build/driver_hydroclima.o build/driver_watertech.o build/driver_abbb23.o build/driver_ebzwmbe.o build/driver_rfmamb.o build/driver_aventieswm.o build/driver_unknown.o build/driver_microclima.o build/driver_c5isf.o build/driver_lse_07_17.o build/driver_omnipower.o build/driver_qwater.o build/driver_ev200.o build/driver_sontex868.o build/driver_fhkvdataiv.o build/driver_gransystems.o build/driver_piigth.o build/driver_minomess.o build/driver_sharky.o build/driver_amiplus.o build/driver_ei6500.o build/driver_apator08.o build/driver_lansensm.o build/driver_q400.o build/driver_ultraheat.o build/driver_engelmann-faw.o build/driver_hydrodigit.o build/driver_lse_08.o build/driver_mkradio3.o build/driver_lansenpu.o build/driver_emerlin868.o build/driver_munia.o build/driver_pollucomf.o build/driver_qcaloric.o build/driver_auto.o build/driver_apator172.o build/driver_vario451.o build/driver_eurisii.o build/driver_kampress.o build/driver_flowiq2200.o build/driver_compact5.o build/driver_mkradio4a.o build/driver_hydrus.o build/driver_aventieshca.o build/driver_sensostar.o build/driver_vario411.o build/driver_qualcosonic.o build/driver_lansenth.o build/driver_qheat_55_us.o build/driver_elf.o build/driver_rfmtx1.o build/driver_evo868.o build/driver_qsmoke.o build/driver_tsd2.o build/driver_gwfwater.o build/driver_iperl.o build/driver_apator162.o build/driver_hydrocalm3.o build/driver_cma12w.o build/driver_nemo.o build/driver_weh_07.o build/driver_waterstarm.o build/main.o  -lrtlsdr -lusb-1.0 -lpthread
/usr/bin/ld: impossibile trovare -lusb-1.0
collect2: error: ld returned 1 exit status
make: *** [build/wmbusmeters.g] Errore 1
```

weetmuts, 2023-05-12:
Perhaps: 
sudo yum install libusb1

?

fpalladino, 2023-05-12:
I have already installed libusbx-1.0.21-1.el7.x86_64.

fpalladino, 2023-05-12:
I'm moved on docker solution and It works fine. Thanks again

weetmuts, 2023-05-12:
Ok! Closing this now.

closed 2023-05-12 by weetmuts