wmbusmeters/wmbusmeters #951: Kein Datemempfang Multival21

Benleni17, 2023-04-29:
### Type of request

Question

### HAOS version

10.1

### wmbusmeters version

1.12.0

### Your message goes here

Good Evening,

Wmbusmeters does not receive current data from multival 21 water meters. In MQTT there is only one payload, but it is outdated. I just updated the core. Is that the reason? What can I do?

`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[21:34:29] INFO: CONFIG_CONF ...
[21:34:29] INFO: {"loglevel":"normal","device":"auto:c1","donotprobe":"/dev/ttyAMA0","logtelegrams":true,"format":"json","logfile":"/dev/stdout","shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[21:34:29] INFO: CONFIG_METERS ...
[21:34:29] INFO: [{"name":"MainWater","driver":"multical21","id":"76406477","key":"AD2F041B407A23917A8D810217AA304B"}]
[21:34:29] INFO: Syncing wmbusmeters configuration ...
[21:34:29] INFO: Registering meters ...
[21:34:29] INFO: Adding meter-0001 ...
[21:34:30] INFO: Generating MQTT configuration ... 
[21:34:32] INFO: Broker core-mosquitto will be used.
MQTT Discovery ...
 Adding meter: MainWater ...
  Add/update topic: homeassistant/sensor/wmbusmeters/76406477_external_temperature_c/config
  Add/update topic: homeassistant/sensor/wmbusmeters/76406477_flow_temperature_c/config
  Add/update topic: homeassistant/sensor/wmbusmeters/76406477_max_flow_m3h/config
  Add/update topic: homeassistant/sensor/wmbusmeters/76406477_rssi_dbm/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/76406477_status_burst/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/76406477_status_dry/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/76406477_status_leak/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/76406477_status_reversed/config
  Add/update topic: homeassistant/sensor/wmbusmeters/76406477_target_m3/config
  Add/update topic: homeassistant/sensor/wmbusmeters/76406477_total_m3/config
MQTT Discovery cleanup...
[21:34:43] INFO: Running wmbusmeters ...
Started auto im871a[00103512] on /dev/ttyUSB0 listening on c1`

MQQT:

`
 / 

{
  "media": "cold water",
  "meter": "multical21",
  "name": "MainWater",
  "id": "76406477",
  "status": "OK",
  "total_m3": 108.866,
  "target_m3": 101.651,
  "flow_temperature_c": 2,
  "external_temperature_c": 7,
  "current_status": "",
  "time_dry": "",
  "time_reversed": "",
  "time_leaking": "",
  "time_bursting": "",
  "timestamp": "2023-04-27T15:55:10Z",
  "device": "im871a[00103512]",
  "rssi_dbm": -47`

Greetz
Benedikt

BIBOLV, 2023-04-30:
Set `logtelegrams=true` and also you can enable debug in order to check what is actually received.

Benleni17, 2023-05-01:
Good Morning,

the value logtelegrams= true ist active. But i doesent see any logs at protokolls.

BIBOLV, 2023-05-01:
Hi,
Set also `loglevel=debug` - could be that you do not receive any more telegrams and one you see in MQTT is old one.

Benleni17, 2023-05-01:
`(serial) received binary ""
(amb8465/3665) reading response... 2
(serial) received binary ""
(amb8465/3665) reading response... 3
(serial) received binary ""
(amb8465/3665) are you there? no.
(serialtty) closed /dev/ttyUSB0 (detect amb8465/amb3665)
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(127,3,253) modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(127,176,9,100,227,127,248) modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect im871a)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary "A581100833001301F13A000053D9"
(im871a) checkIM871AFrame "A581100833001301F13A000053D9"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 16
(im871a) got crc16 d953 expected d953
(im871a) received full frame
(device info bytes) "33001301F13A0000"
(im871a/im170a) info: type=im871a mode=collector firmware=13 hci=01 uid=00003af1
(serial /dev/ttyUSB0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary "A5810617FF000700B32512351000010001FF07023200010000000063C9"
(im871a) checkIM871AFrame "A5810617FF000700B32512351000010001FF07023200010000000063C9"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 c963 expected c963
(im871a) received full frame
(serialtty) closed /dev/ttyUSB0 (detect im871a)
(serial) EVENT thread interrupted
(device config bytes) "FF000700B32512351000010001FF070232000100000000"
(im871a/im170a) config: other link_mode=cab id=00103512 media=00 version=01 c_field=00 auto_rssi=01
(im871a/im170a) are you there? yes 00103512 im871a firmware: 13
(main) opening c1
Started auto im871a[00103512] on /dev/ttyUSB0 listening on c1
(im871a) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(127,4,253) modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(127,128,1,100,227,127,248) modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 4 (im871a)
(im871a) get config
(serial /dev/ttyUSB0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary "A5810617FF000700B32512351000010001FF07023200010000000063C9"
(im871a) checkIM871AFrame "A5810617FF000700B32512351000010001FF07023200010000000063C9"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 c963 expected c963
(im871a) received full frame
(im871a) get config completed
(im871a) got device id 00103512
(main) regular reset of im871a  on /dev/ttyUSB0 will happen every 82800 seconds
(im871a) get device info
(serial /dev/ttyUSB0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary "A581100833001301F13A000053D9"
(im871a) checkIM871AFrame "A581100833001301F13A000053D9"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 16
(im871a) got crc16 d953 expected d953
(im871a) received full frame
(im871a) device info completed
(im871a) device info: type=im871a mode=collector firmware=13 hci=01 uid=00003af1
(im871a) set config to set link mode 07
(serial /dev/ttyUSB0) sent "A5010306000207300100"
(serial) EVENT thread interrupted
(serial) received binary "A58104009CF7"
(im871a) checkIM871AFrame "A58104009CF7"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 4
(im871a) got crc16 f79c expected f79c
(im871a) received full frame
(im871a) set config completed
(wmbus) no alarm (expected activity) for im871a
(main) device /dev/ttyUSB1 not currently used, detect contents...
(serial) check if /dev/ttyUSB1 can be accessed
(serial) tty /dev/ttyUSB1 can be accessed
(serial) EVENT thread interrupted
(serialtty) device /dev/ttyUSB1 is already in use and locked.
(amb8465/3665) could not open tty /dev/ttyUSB1 for detection
(serial) EVENT thread interrupted
(serialtty) device /dev/ttyUSB1 is already in use and locked.
(im871a) could not open tty /dev/ttyUSB1 for detection
(serial) EVENT thread interrupted
(serialtty) device /dev/ttyUSB1 is already in use and locked.
(serial) EVENT thread interrupted
(serialtty) device /dev/ttyUSB1 is already in use and locked.
(serial) EVENT thread interrupted
(serialtty) device /dev/ttyUSB1 is already in use and locked.
(iu880b) could not open tty /dev/ttyUSB1 for detection
(main) ignoring /dev/ttyUSB1, it does not respond as any of the supported wmbus devices.
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop`

BIBOLV, 2023-05-01:
That is ok - now you need to wait for telegrams from meter.

closed 2023-08-01 by github-actions[bot]