wmbusmeters/wmbusmeters #952: Help with getting the device encryption key

sanchosk, 2023-04-29:
I would like to find out how can I get the encryption key for the Topas ES KR.
The message arrives, I am using the correct driver, but it seems to be encrypted and I cannot get the data out.
Any suggestions for newbie on how to get the key?
I doubt I'll be able to get it from local water company...

BIBOLV, 2023-04-30:
Getting key is only option - please check [this](https://github.com/wmbusmeters/wmbusmeters/issues/446) .

SaSa1983, 2023-05-08:
There's no chance that the developer cab help you to get the required key.
In my case I got mine by sending a request to the local city.

sanchosk, 2023-05-08:
Actually, the developer did offered a key that worked :)

-- Sent from /e/ Mail.

closed 2023-08-08 by github-actions[bot]