wmbusmeters/wmbusmeters #953: (cul) dll C1 (frame b) crcs failed check! Ignoring telegram!

FG148, 2023-05-02:
### Type of request

Question

### OS version

Linux raspberry

### wmbusmeters version

1.12.0-31-gd135ed9

### Your message goes here

Hello,
got WMBus working on a raspberry pi and a Nano CUL 868 after updating the firmware with nanoCUL_r568_mbus_c1t1_bufsize300.hex (increased buffersize).
We got a Engelmann heat meter (https://www.engelmann.de/de/sensostar-u_/) and Lorenz water meters (https://www.lorenz-meters.de/produkte/wohnungswasserzaehler/) installed by a service provider.
I got the AES-Keys for all 3 devices provided, but no communication details.

On C1 I am frequently receiving telegrams, not on T1 when I start wmbusmeters (wmbusmeters --debug --logtelegrams --verbose /dev/ttyUSB1:cul:c1).
But the telgrams are not processed further: 
(cul) dll C1 (frame b) crcs failed check! Ignoring telegram!

Could anybody help? Output and a view telegrams attached.

Thank you very much,
Florian

```
(wmbusmeters) version: 1.12.0-31-gd135ed9
(config) using device: /dev/ttyUSB1:cul:c1
(config) number of meters: 0
(lookup) with file/hex "/dev/ttyUSB1" cul
(serial) check if /dev/ttyUSB1 can be accessed
(serial) tty /dev/ttyUSB1 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB1 speed(B38400) input() output() control(CLOC                                                                                                             AL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB1 speed(B38400) input() output() control(CLOC                                                                                                             AL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB1 fd 3 (detect cul)
(serial /dev/ttyUSB1) sent "560D0A"
(serial) EVENT thread interrupted
(serial) received binary "0A5620312E3637206E616E6F43554C3836385F723536380D0A"
(cul) probe response "<0A>V 1.67 nanoCUL868_r568<0D><0A>"
(serialtty) closed /dev/ttyUSB1 (detect cul)
(cul) are you there? yes
(serial) EVENT thread interrupted
(main) opening /dev/ttyUSB1:cul:c1
Started config cul on /dev/ttyUSB1 listening on c1
(cul) on /dev/ttyUSB1
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB1 speed(B38400) input() output() control(CLOC                                                                                                             AL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB1 speed(B38400) input() output() control(CLOC                                                                                                             AL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB1 fd 3 (cul)
(main) regular reset of cul  on /dev/ttyUSB1 will happen every 82800 seconds
(cul) set link mode c
(serial /dev/ttyUSB1) sent "6272630D0A"
(serial) EVENT thread interrupted
(serial) received binary "434D4F44450D0A"
(cul) checkCULFrame "CMODE<0D><0A>"
(cul) no leading 'b' so it is text and no frame
(cul) received "CMODE"(serial /dev/ttyUSB1) sent "5832310D0A"
(wmbus) no alarm (expected activity) for cul
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) EVENT thread interrupted

(serial) received binary "625941"
(cul) checkCULFrame "bYA"
(cul) no eol found yet, partial frame
[... 67 lines cut ...]
(cul) checkCULFrame RSSI_RAW=-80
(cul) checkCULFrame LQI=106
(wmbus) trimming frame B "AD44C5146412052500047ACE00902529289022CDA92575BBAE3BB618688F9820AB325DB336D6ADC9A6B2B12520F39830C9788F5503F2C7F75F34E150A9D994F735C2F7D126A74C7ED3EAC87893094F1BDF4E7F630433F916C948C3A7EFAA1000FA2EE596467B4B9DBDD8C602B487A7F396F0F513CF05288A8DE62EF24BBB458F1C129D51E30673FB46F4E3B404CDA8E9413F"
(wmbus) ff b dll crc first 0-126 458f ok
(wmbus) ff b dll crc (calculated c47a) did not match (expected 413f) for bytes 128-144!
(cul) dll C1 (frame b) crcs failed check! Ignoring telegram!
(cul) error in received message.
(serial) received binary "625941443434433531343634313230353235303030343741443030303930323545314345343933323434433736443631383435"
(cul) checkCULFrame "bYAD44C5146412052500047AD0009025E1CE493244C76D61845"
(cul) no eol found yet, partial frame
(serial) received binary "413946464142334235433630323438454230433338324446443742443336333246344132384445464544423734463844373135413632463741443833443842463937393931373638464443313231384244323830363941303138323232373444394438354234393438353634453734303141453932313742"
(cul) checkCULFrame "bYAD44C5146412052500047AD0009025E1CE493244C76D61845A9FFAB3B5C60248EB0C382DFD7BD3632F4A28DEFEDB74F8D715A62F7AD83D8BF97991768FDC1218BD28069A01822274D9D85B4948564E7401AE9217B"
(cul) no eol found yet, partial frame
(serial) received binary "30393432463842383939364441413932464335443739434137454634393743414146324535383843343445413734443930444242434330373146304642"
(cul) checkCULFrame "bYAD44C5146412052500047AD0009025E1CE493244C76D61845A9FFAB3B5C60248EB0C382DFD7BD3632F4A28DEFEDB74F8D715A62F7AD83D8BF97991768FDC1218BD28069A01822274D9D85B4948564E7401AE9217B0942F8B8996DAA92FC5D79CA7EF497CAAF2E588C44EA74D90DBBCC071F0FB"
(cul) no eol found yet, partial frame
(serial) received binary "3538413846344645343734333941423336314441393338414333333744363343343233433331343335464439323232444132353139383931343734303545314346320D0A"
(cul) checkCULFrame "bYAD44C5146412052500047AD0009025E1CE493244C76D61845A9FFAB3B5C60248EB0C382DFD7BD3632F4A28DEFEDB74F8D715A62F7AD83D8BF97991768FDC1218BD28069A01822274D9D85B4948564E7401AE9217B0942F8B8996DAA92FC5D79CA7EF497CAAF2E588C44EA74D90DBBCC071F0FB58A8F4FE47439AB361DA938AC337D63C423C31435FD9222DA251989147405E1CF2<0D><0A>"
(cul) checkCULFrame RSSI_RAW=-14
(cul) checkCULFrame LQI=14
(wmbus) trimming frame B "AD44C5146412052500047AD0009025E1CE493244C76D61845A9FFAB3B5C60248EB0C382DFD7BD3632F4A28DEFEDB74F8D715A62F7AD83D8BF97991768FDC1218BD28069A01822274D9D85B4948564E7401AE9217B0942F8B8996DAA92FC5D79CA7EF497CAAF2E588C44EA74D90DBBCC071F0FB58A8F4FE47439AB361DA938AC337D63C423C31435FD9222DA251989147405E"
(wmbus) ff b dll crc first 0-126 8ac3 ok
(wmbus) ff b dll crc (calculated 4974) did not match (expected 405e) for bytes 128-144!
(cul) dll C1 (frame b) crcs failed check! Ignoring telegram!
```

weetmuts, 2023-05-02:
I think your telegram hit the 300 byte serial port byte buffer which sends the data to the computer. 

According to https://github.com/wmbusmeters/wmbusmeters-wiki/blob/master/nanoCUL.md
the maximum telegram length 148 bytes.

The telegram data starts with 0xad hex which means 173, so the telegram should have 173 bytes in addition to the length byte. However the actual number of bytes are only 145, so it got chopped due to the limitations of the cul software. This will case the crc of the remaining part to fail.

weetmuts, 2023-05-02:
Since the maximum wmbus telegram length is 255 bytes+1 for the length byte, it seems like the cul:s limitation to 148 is pretty severe. But of course many meters send short telegrams on c1, so it just happens to work for a lot of users. Someone with skill should rewrite the cul firmware... :-)

BIBOLV, 2023-05-05:
I would recommend purpose built devices like [imst](https://shop.imst.de/wireless-modules/usb-radio-products/10/im871a-usb-wireless-m-bus-usb-adapter-868-mhz?c=14) or amber.

FG148, 2023-05-11:
Thank you very much for your contructive comments! 
With the recommended IMST-Hardware it works perfectly!

dkadioglu, 2023-06-09:
> The described setup based on Arduino every (https://github.com/DecksLabs/every-culfw/tree/main/docs) should cope with the 173 byte message (buffer size 1500 bytes).

I would also like to know, if the Nano Every should be able to read messages from Engelmann Heatmeters, in my case SensoStar Q. I successfully set up the Arduino Nano Every with the mentioned every-culfw and receive now longer telegrams of the SensoStar Q than with the regular nanoCUL. One example:

`(wmbus) trimming frame B "AD44C5148919152300047A8B0090256D7E4BF066F399EE875BC25A70C447A9D318C4B69D6D58AF8239358DF7F14913D1D8994B273859A83736ACD15E9F85A18086C419618AF4547B6A577B58B7BEC69440933088BF68819AB2635BC4266B94E364B12FD791A1F2BB5E24FDEF30E063409CCDF88C62E89D23523B45465D95DDD7120839DADB20F97BE0FC364BC07B7907AC3840898F6049AF96A0CC1814FADD41F403FD0C05010002FD0B0111077C"
(wmbus) ff b dll crc first 0-126 ddd7 ok
(wmbus) ff b dll crc (calculated fd7e) did not match (expected 077c) for bytes 128-172!
(cul) dll C1 (frame b) crcs failed check! Ignoring telegram!
(cul) error in received message.
`

But I still have the CRC errors. As far as I understand it, the message should have a length of 173 (it also starts with 0xAD), so the Nano Every should easily cope with it. Or am I wrong here?

dkadioglu, 2023-07-08:
Since Thursday, I have the key for the two Engelmann Sensostar Q. However, I still have the issue, that the telegrams seem to not be received correctly. In the following some received telegrams, decoded by wmbusmeters.org:

```
ad46c5148919152300047a7a00900531f949095B5126d9f7fdf14d68dB4ef048320d1099cc5691B2c0010B61ca169f6019B19076029B27eBc076daafB4028B592e92daB31128af294cc052376e3c264cB518765dBadf9efffe63da18c54Bdc53a7caac30d6f6ede756e18f53df0efa258ceea8744ed11683a6B1a1667cB6B90BB9ae59c8767ed312d37a5435ce13796dd97340e8da78298c06e5330d336d5ac59403fd0c05010002fd0B11113dc08019

Auto driver  : sensostar
Best driver  : microclima 63/102
Using driver : sensostar 00/00
000   : ad length (173 bytes)
001   : 46 dll-c (from meter SND_IR)
002   : c514 dll-mfct (EFE)
004   : 89191523 dll-id (23151989)
008   : 00 dll-version
009   : 04 dll-type (Heat meter)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : 7a tpl-acc-field
012   : 00 tpl-sts-field (OK)
013   : 9005 tpl-cfg 0590 (AES_CBC_IV nb=9 cntn=0 ra=0 hc=0 )
015   : 2f2f decrypt check bytes (OK)
017   : 04 dif (32 Bit Integer/Binary Instantaneous value)
018   : 6D vif (Date and time type)
019 C!: 2928E627 ("meter_timestamp":"2023-07-06 08:41")
023   : 04 dif (32 Bit Integer/Binary Instantaneous value)
024   : 06 vif (Energy kWh)
025 C!: 26050000 ("total_kwh":1318)
029   : 04 dif (32 Bit Integer/Binary Instantaneous value)
030   : 13 vif (Volume l)
031 C!: 20390200 ("total_water_m3":145.696)
035   : 01 dif (8 Bit Integer/Binary Instantaneous value)
036   : FD vif (Second extension FD of VIF-codes)
037   : 17 vife (Error flags (binary))
038 C!: 00 ("current_status":"OK")
039   : 42 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
040   : 6C vif (Date type G)
041 C!: DF2C ("reporting_date":"2022-12-31")
043   : 44 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
044   : 06 vif (Energy kWh)
045 C!: B2020000 ("energy_consumption_at_reporting_date_kwh":690) ("consumption_1_months_ago_kwh":690)
049   : 84 dif (32 Bit Integer/Binary Instantaneous value)
050   : 01 dife (subunit=0 tariff=0 storagenr=2)
051   : 06 vif (Energy kWh)
052 C!: 26050000 ("consumption_2_months_ago_kwh":1318)
056   : 84 dif (32 Bit Integer/Binary Instantaneous value)
057   : 02 dife (subunit=0 tariff=0 storagenr=4)
058   : 06 vif (Energy kWh)
059 C!: 26050000 ("consumption_4_months_ago_kwh":1318)
063   : 84 dif (32 Bit Integer/Binary Instantaneous value)
064   : 03 dife (subunit=0 tariff=0 storagenr=6)
065   : 06 vif (Energy kWh)
066 C!: 26050000 ("consumption_6_months_ago_kwh":1318)
070   : 84 dif (32 Bit Integer/Binary Instantaneous value)
071   : 04 dife (subunit=0 tariff=0 storagenr=8)
072   : 06 vif (Energy kWh)
073 C!: F0040000 ("consumption_8_months_ago_kwh":1264)
077   : 84 dif (32 Bit Integer/Binary Instantaneous value)
078   : 05 dife (subunit=0 tariff=0 storagenr=10)
079   : 06 vif (Energy kWh)
080 C!: 4F040000 ("consumption_10_months_ago_kwh":1103)
084   : 84 dif (32 Bit Integer/Binary Instantaneous value)
085   : 06 dife (subunit=0 tariff=0 storagenr=12)
086   : 06 vif (Energy kWh)
087 C!: 8A030000 ("consumption_12_months_ago_kwh":906)
091   : 84 dif (32 Bit Integer/Binary Instantaneous value)
092   : 07 dife (subunit=0 tariff=0 storagenr=14)
093   : 06 vif (Energy kWh)
094 C!: B2020000 ("consumption_14_months_ago_kwh":690)
098   : 84 dif (32 Bit Integer/Binary Instantaneous value)
099   : 08 dife (subunit=0 tariff=0 storagenr=16)
100   : 06 vif (Energy kWh)
101 C?: CF010000
[... 412 lines cut ...]
    "id":"23151989",
    "meter_timestamp":"2023-07-06 08:57",
    "total_kwh":1318,
    "total_water_m3":145.696,
    "current_status":"OK",
    "reporting_date":"2022-12-31",
    "energy_consumption_at_reporting_date_kwh":690,
    "consumption_1_months_ago_kwh":690,
    "consumption_2_months_ago_kwh":1318,
    "consumption_4_months_ago_kwh":1318,
    "consumption_6_months_ago_kwh":1318,
    "consumption_8_months_ago_kwh":1264,
    "consumption_10_months_ago_kwh":1103,
    "consumption_12_months_ago_kwh":906,
    "consumption_14_months_ago_kwh":690,
    "timestamp":"2023-07-06T07:58:16Z"
}

Using: wmbusmeters: 1.13.1-35-g53c589e
53c589ed111f9faf2b5d26049e18b545f5f47d19

Do you know how to improve the decoding? Open pre-filled issue ticket
```

Is there anything I can do about this? As I am using a NanoCUL Every, I thought that the size limitation should not be an issue anymore. The weird thing is, that even with an RTL-SDR, I am not able to successfully receive those telegrams. Really weird is, that I even cannot see those partial messages with CRC errors, starting with AD, in the debug output.

dkadioglu, 2023-07-27:
Can I provide any further info to help analyzing this?

closed 2023-10-29 by github-actions[bot]

fry2k, 2024-12-25:
@dkadioglu did you found a solution for the nano every with the Engelmann SensoStar? I have now the exact same problem. With the nano every I can fetch the data from my Lorenz water meter (which wasn't possible with the non every). But for the SensoStar I still see crcs failed check

`(wmbus) trimming frame B "AD44C5145060152300047A2C009025373D4E8C44AF2A5FE40A6BBFFE915F6EF2CF1487129C7FB40FA1F04720E8B27138C42E670B0F48F7054F2F65ECDC6D7616B1CD941033659EF62249610C92893F46BA5EBA574B568D753CB32BE252B0753C897BDA8CBFCD14AAB7225FA6D50956A4114000DCDC82D2BD6E18746E7F027FE381E63F84420F4F64E1FD052C76214663BAA07EE568AB6A4F2C7A09E43E6D1686B903FD0C05010002FD0B21118B50"
(wmbus) ff b dll crc first 0-126 7fe3 ok
(wmbus) ff b dll crc (calculated 30fa) did not match (expected 8b50) for bytes 128-172!
[2024-12-25_21:00:43] (cul) dll C1 (frame b) crcs failed check! Ignoring telegram!`

dkadioglu, 2025-01-01:
> @dkadioglu did you found a solution for the nano every with the Engelmann SensoStar? I have now the exact same problem. With the nano every I can fetch the data from my Lorenz water meter (which [...]

Hi @fry2k , a happy new year. Unfortunately not, as I gave up on the Nano Every. Now I use the iM871a from IMST.

fry2k, 2025-01-10:
Hi @dkadioglu, thanks, I wish you a happy new year too! I bought now the iu891a because the im871a isn't available any more.
With the iu891a I can read my water meter but I don't see any telegrams from the SensoStar U. I have debug, verbose and log telegrams enabled. This is really confusing for me.
I replugged my nanoCUL every and I saw directly the crcs errors from the SensoStar. Analyzing this broken telegram shows the ID from my SensoStar meter.

I use the latest master from this repo for wmbusmeters. I use the c1 protocol on both dongles. The dongle are 1m away from the meters.

weetmuts, 2025-01-10:
The CUL is weird, if you specify c1, then it will listen to both c1 and t1. But iu891a only listens to c1 if you specify c1.
So specify c1,t1 for iu891a. The SensoStar is probably transmitting using T1.

dkadioglu, 2025-01-14:
Hi @fry2k , there is an issue with Engelmann devices, which has been investigated by IMST. They have implemented a fix in the firmware for that. However, currently this is still an unreleased beta and I am not sure, if I am allowed to share it with you. As soon as I have some further info, I will post it here. So nothing wmbusmeters related in the end.

dartrax, 2026-01-11:
Hi @dkadioglu and @fry2k, do you have any news on this issue?

The fix you've mentioned: Is that implememented in the Firmware of the IMST Stick, so we could call it a workaround?
Or was it implemented by Engelmann into their meters with the help of IMST?

We seem to face the same problem and I'd like to check what it takes to implement such a workaround into the cul or wmbusmeters, but we first need to understand the problem and possible existing fixes.

dkadioglu, 2026-01-18:
Sorry for my late answer. From my perspective and understanding, Engelmann 's implementatiln of WMBUS is "special", which leads to receiving issues with the sticks from IMST. Unfortunately, I still don't have any news regarding an official release of the beta firmware. I ask them regularly. What I can say is, that I can receive WMBUS messages from my two Engelmann SensoStar meters without issues since then. So for sure no issue in wmbusmeters, I think.

dkadioglu, 2026-01-18:
> Was this telegram received using the everyCUL? Are the information decoded as expected? Wouldn'd that mean that (only) the crc check is the culprit?

Yes, these telegrams should have been received using the everyCUL, as I bought the IMST one later. As far as I remember, the information was correct but not complete. Sorry, I forgot the details over time.

fry2k, 2026-01-19:
@dartrax I checked my old mails again and found this explanation from IMST which could be helpful. The answer was in German. I'm not translating it to not lose anything important.

Wir haben einige Engelmann Geräte mit einem Spektrum-Analysator vermessen und festgestellt, dass zwischen der Preamble und dem Syncword eine zusätzliche Null implementiert ist, die nicht dem wM-BuS-Standard entspricht.

And I have to correct my message from last week. With the beta firmware I was receiving telegrams from the sensostar. But I wasn't able to decode them. Maybe the AES key I received is wrong. Going to double check this.

```[2025-02-21_20:04:16] (meter) FernwaermeRh8: for me? 23156050.M=EFE.V=00.T=04 in 23156050
[2025-02-21_20:04:16] (meter) FernwaermeRh8: yes for me
[2025-02-21_20:04:16] (meter) FernwaermeRh8(1) auto  handling telegram from 23156050.M=EFE.V=00.T=04
[2025-02-21_20:04:16] (meter) FernwaermeRh8 23156050.M=EFE.V=00.T=04 "A944C5145060152300047AA5009025F33B57A868BDFF478546C31DEB6D8502CD31EC1B32FDA7738252B82DC93B210DC4F48CBBEE2013E27FC13E5ABB1E89AACA4AC920A5557823371961A16DB8D8F8B6706C1131779E56BD0E5736BEA24B587FC9576612AC44BD6C197C0BD8B8EDC189117FBC4F637A76B89FF0A77DE66A628DAA8F1002D87DBAAF95DBE1F053E46F59584D400F0AD9C03D5A52908E72DB1103FD0C05010002FD0B2111"
[2025-02-21_20:04:16] (wmbus) parseDLL @0 170
[2025-02-21_20:04:16] (telegram) DLL L=a9 C=44 (from meter SND_NR) M=14c5 (EFE) A=23156050 VER=00 TYPE=04 (Heat meter) (driver sensostar) DEV=iu891a[00202043] RSSI=-41
[2025-02-21_20:04:16] (wmbus) parseELL @10 160
[2025-02-21_20:04:16] (wmbus) parseNWL @10 160
[2025-02-21_20:04:16] (wmbus) parseAFL @10 160
[2025-02-21_20:04:16] (wmbus) parseTPL @10 160
[2025-02-21_20:04:16] (TPL) num encrypted blocks 9 (144 bytes and remaining unencrypted 11 bytes)
[2025-02-21_20:04:16] (TPL) AES CBC IV decrypting "F33B57A868BDFF478546C31DEB6D8502CD31EC1B32FDA7738252B82DC93B210DC4F48CBBEE2013E27FC13E5ABB1E89AACA4AC920A5557823371961A16DB8D8F8B6706C1131779E56BD0E5736BEA24B587FC9576612AC44BD6C197C0BD8B8EDC189117FBC4F637A76B89FF0A77DE66A628DAA8F1002D87DBAAF95DBE1F053E46F59584D400F0AD9C03D5A52908E72DB1103FD0C05010002FD0B2111"
[2025-02-21_20:04:16] (TPL) IV C514506015230004A5A5A5A5A5A5A5A5
[2025-02-21_20:04:16] (TPL) decrypted  "0FB7002EBB98608E8F12EC433B243875EF85504213CEED29B21A75B8FC39463E56FA8D178CC65AFB70077A9557F6283A61415AA86DD783BA46C33D3C5D102F18AF5A29F079358862FDD53042CCC871E356FA1A7E8AF8F299CCE838D682A6347C3881C0A2BC73DE1B601206A5BA75A2BC4AFAB7BA6E22E6E9C40364D89B27AE7A202C293D5CDEDBB064ED97B126EDC6D9"
[2025-02-21_20:04:16] (TPL) appended   "0FB7002EBB98608E8F12EC433B243875EF85504213CEED29B21A75B8FC39463E56FA8D178CC65AFB70077A9557F6283A61415AA86DD783BA46C33D3C5D102F18AF5A29F079358862FDD53042CCC871E356FA1A7E8AF8F299CCE838D682A6347C3881C0A2BC73DE1B601206A5BA75A2BC4AFAB7BA6E22E6E9C40364D89B27AE7A202C293D5CDEDBB064ED97B126EDC6D903FD0C05010002FD0B2111"
[2025-02-21_20:04:16] (wmbus) WARNING!! decrypted content failed check, did you use the correct decryption key? Permanently ignoring telegrams from id: 23156050 mfct: (EFE) Engelmann Sensor, Germany (0x14c5) type: Heat meter (0x04) ver: 0x00```

fry2k, 2026-02-02:
I can also confirm this issue as fixed. I was in contact with my provider and they send me the wrong AES key. I'm able to decode the telegrams with the new AES key. Everything works perfect with my every cul now. Thanks a lot.