wmbusmeters/wmbusmeters #955: Engelmann Heat cost allocators HCA e2

Skysurfer-14, 2023-05-09:
### Type of request

New device support

### Meter description

**Manufacturer**: Engelmann [https://www.engelmann.de/en/hca-e2/](url)
**Model**: HCA e2
**Radio mode**: C-mode OMS 4 Mode 5 with own AES key, Radio time: 7 am - 7 pm daily

For more information see #820 

### Logged telegrams from meter in decrypted format

```shell
(serial) received binary "A5C203CE44C5149126412331087A6B0BC02562235F3EA78C0A58BBB4284B7A9AEBE11943EB52740634A29C9E08CD2C3D90717360D2E342A39DACE7A849D03DCA80F4C424C0DE1A79A2A40F0EAC886C9417FA084BDFB8BE21A8E904AA41946E51D1B5E2D929127C6D1470EAF093497B7574CFB22FF13A866812FD780D9EBE9EB4DE2E8CADEF92F01FBBFFD42B1D7FE0FD8093D76F6015029EF9F6F96E11C4094C79D0CB547DAF69FD7F7CC98F57E6CB4CF147ADD5BAF247E03EF7AFE51EAF58AC905E756C17B077E58FA2E7B3C359BA8867A2938934"
(im871a) checkIM871AFrame "A5C203CE44C5149126412331087A6B0BC02562235F3EA78C0A58BBB4284B7A9AEBE11943EB52740634A29C9E08CD2C3D90717360D2E342A39DACE7A849D03DCA80F4C424C0DE1A79A2A40F0EAC886C9417FA084BDFB8BE21A8E904AA41946E51D1B5E2D929127C6D1470EAF093497B7574CFB22FF13A866812FD780D9EBE9EB4DE2E8CADEF92F01FBBFFD42B1D7FE0FD8093D76F6015029EF9F6F96E11C4094C79D0CB547DAF69FD7F7CC98F57E6CB4CF147ADD5BAF247E03EF7AFE51EAF58AC905E756C17B077E58FA2E7B3C359BA8867A2938934"
(im871a) has_timestamp=0 has_rssi=1 has_crc16=1
(im871a) endpoint 2
(im871a) msgid 3
(im871a) rssi 147 (-49 dBm)
(im871a) got crc16 3489 expected 3489
(im871a) received full frame
(wmbus) parseDLL @0 207
(wmbus) parseELL @10 197
(wmbus) parseNWL @10 197
(wmbus) parseAFL @10 197
(wmbus) parseTPL @10 197
(meter) ExampleMeter: for me? 23412691 in 23412691
(meter) ExampleMeter: yes for me
(meter) ExampleMeter(1) auto  handling telegram from 23412691
(meter) ExampleMeter 23412691 "CE44C5149126412331087A6B0BC02562235F3EA78C0A58BBB4284B7A9AEBE11943EB52740634A29C9E08CD2C3D90717360D2E342A39DACE7A849D03DCA80F4C424C0DE1A79A2A40F0EAC886C9417FA084BDFB8BE21A8E904AA41946E51D1B5E2D929127C6D1470EAF093497B7574CFB22FF13A866812FD780D9EBE9EB4DE2E8CADEF92F01FBBFFD42B1D7FE0FD8093D76F6015029EF9F6F96E11C4094C79D0CB547DAF69FD7F7CC98F57E6CB4CF147ADD5BAF247E03EF7AFE51EAF58AC905E756C17B077E58FA2E7B3C359BA8867A2"
(wmbus) parseDLL @0 207
(telegram) DLL L=ce C=44 (from meter SND_NR) M=14c5 (EFE) A=23412691 VER=31 TYPE=08 (Heat Cost Allocator) (driver unknown!) DEV=im871a[00103765] RSSI=-49
(wmbus) parseELL @10 197
(wmbus) parseNWL @10 197
(wmbus) parseAFL @10 197
(wmbus) parseTPL @10 197
(TPL) num encrypted blocks 12 (192 bytes and remaining unencrypted 0 bytes)
(TPL) AES CBC IV decrypting "62235F3EA78C0A58BBB4284B7A9AEBE11943EB52740634A29C9E08CD2C3D90717360D2E342A39DACE7A849D03DCA80F4C424C0DE1A79A2A40F0EAC886C9417FA084BDFB8BE21A8E904AA41946E51D1B5E2D929127C6D1470EAF093497B7574CFB22FF13A866812FD780D9EBE9EB4DE2E8CADEF92F01FBBFFD42B1D7FE0FD8093D76F6015029EF9F6F96E11C4094C79D0CB547DAF69FD7F7CC98F57E6CB4CF147ADD5BAF247E03EF7AFE51EAF58AC905E756C17B077E58FA2E7B3C359BA8867A2"
(TPL) IV C5149126412331086B6B6B6B6B6B6B6B
(TPL) decrypted  "2F2F046D1829FE23036E000000426C0000436E00000082016CBF2C83016E000000C2016E000082026E0000C2026E000082036E0000C2036E000082046E0000C2046E000082056E0000C2056E000082066E0000C2066E000082076E0000C2076E000082086E0000C2086E000082096E0000C2096E0000820A6E0000C20A6E0000820B6E0000C20B6E0000820C6E0000C20C6E0000820D6E0000C20D6E0000820E6E0000C20E6E0000820F6E0000C20F6E0000317F08346D302EFB232F2F2F2F2F"
(telegram) TPL CI=7a ACC=6b STS=0b CFG=25c0 (synchronous AES_CBC_IV nb=12 cntn=0 ra=0 hc=0)
telegram=|CE44C5149126412331087A6B0BC0252F2F_046D1829FE23036E000000426C0000436E00000082016CBF2C83016E000000C2016E000082026E0000C2026E000082036E0000C2036E000082046E0000C2046E000082056E0000C2056E000082066E0000C2066E000082076E0000C2076E000082086E0000C2086E000082096E0000C2096E0000820A6E0000C20A6E0000820B6E0000C20B6E0000820C6E0000C20C6E0000820D6E0000C20D6E0000820E6E0000C20E6E0000820F6E0000C20F6E0000317F08346D302EFB232F2F2F2F2F|+73253
(auto) 000   : ce length (206 bytes)
(auto) 001   : 44 dll-c (from meter SND_NR)
(auto) 002   : c514 dll-mfct (EFE)
(auto) 004   : 91264123 dll-id (23412691)
(auto) 008   : 31 dll-version
(auto) 009   : 08 dll-type (Heat Cost Allocator)
(auto) 010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(auto) 011   : 6b tpl-acc-field
(auto) 012   : 0b tpl-sts-field (ALARM PERMANENT_ERROR)
(auto) 013   : c025 tpl-cfg 25c0 (synchronous AES_CBC_IV nb=12 cntn=0 ra=0 hc=0 )
(auto) 015   : 2f2f decrypt check bytes (OK)
(auto) 017   : 04 dif (32 Bit Integer/Binary Instantaneous value)
(auto) 018   : 6D vif (Date and time type)
(auto) 019 C?: 1829FE23
(auto) 023   : 03 dif (24 Bit Integer/Binary Instantaneous value)
(auto) 024   : 6E vif (Units for H.C.A.)
(auto) 025 C?: 000000
(auto) 028   : 42 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
(auto) 029   : 6C vif (Date type G)
(auto) 030 C?: 0000
(auto) 032   : 43 dif (24 Bit Integer/Binary Instantaneous value storagenr=1)
(auto) 033   : 6E vif (Units for H.C.A.)
(auto) 034 C?: 000000
(auto) 037   : 82 dif (16 Bit Integer/Binary Instantaneous value)
(auto) 038   : 01 dife (subunit=0 tariff=0 storagenr=2)
(auto) 039   : 6C vif (Date type G)
[... 108 lines cut ...]
(auto) 176 C?: 0000
(auto) 178   : C2 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
(auto) 179   : 0E dife (subunit=0 tariff=0 storagenr=29)
(auto) 180   : 6E vif (Units for H.C.A.)
(auto) 181 C?: 0000
(auto) 183   : 82 dif (16 Bit Integer/Binary Instantaneous value)
(auto) 184   : 0F dife (subunit=0 tariff=0 storagenr=30)
(auto) 185   : 6E vif (Units for H.C.A.)
(auto) 186 C?: 0000
(auto) 188   : C2 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
(auto) 189   : 0F dife (subunit=0 tariff=0 storagenr=31)
(auto) 190   : 6E vif (Units for H.C.A.)
(auto) 191 C?: 0000
(auto) 193   : 31 dif (8 Bit Integer/Binary Value during error state)
(auto) 194   : 7F vif (Manufacturer specific)
(auto) 195 C?: 08
(auto) 196   : 34 dif (32 Bit Integer/Binary Value during error state)
(auto) 197   : 6D vif (Date and time type)
(auto) 198 C?: 302EFB23
(auto) 202   : 2F skip
(auto) 203   : 2F skip
(auto) 204   : 2F skip
(auto) 205   : 2F skip
(auto) 206   : 2F skip
```

BIBOLV, 2023-05-09:
I created PR with file provided in https://github.com/wmbusmeters/wmbusmeters/issues/820.
Regarding, what to do to be able to use in ha-addon - once PR will be accepted, driver will available in ha-addon for manual integration. If you want automatic integration - template should be added to https://github.com/wmbusmeters/wmbusmeters-ha-addon/tree/main/wmbusmeters-ha-addon-edge/mqtt_discovery.

closed 2023-08-09 by github-actions[bot]

Skysurfer-14, 2024-11-16:
Hi TimMe1602,

I think it's 868.95 MHz. The HCAe2 are using mode C1, for details see [Wireless M-Bus](https://radiocrafts.com/technologies/wireless-m-bus-technology-overview/).
Hope this helps!