wmbusmeters/wmbusmeters #958: Bmeters GSD8-I with IWM-TX5 wmbus module

maacaron, 2023-05-11:
### Type of request

New device support

### Meter description

Manufacturer: Bmeters
Model: GSD8-I
Link: https://www.bmeters.com/en/products/gsd8-i/
WM-BUS module: IWM-TX5
Link: https://www.bmeters.com/en/products/iwm-tx5/
Datasheet: https://www.bmeters.com/wp-content/uploads/2022/01/IWM-TX5_ENG_v1.1.pdf
> As soon as the radio operating mode is activated, the device sends a radio telegram according to Wireless
M-BUS, T1 mode (one-way transmission).

Telegram frame:
```
T1;1;1;2023-05-11 08:35:13.000;129;119;22917370;0x5144b4097073912218078c00247a0308400571e9615249ede52eaae09f61908f027c3877f3330ae9079528b23173ce124bcc255393e60b173c0a9f274c42dd92e4b23c14e8a41f042903358df01dd9268ad4
```
When trying to analyze telegram on [wmbusmeters.org](https://wmbusmeters.org/):
```
Auto driver  : not found!
Best driver  : unknown 00/00
Using driver : unknown 00/00
000   : 51 length (81 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : b409 dll-mfct (BMT)
004   : 70739122 dll-id (22917370)
008   : 18 dll-version
009   : 07 dll-type (Water meter)
010   : 8c ell-ci-field (ELL: Extended Link Layer I (2 Byte))
011   : 00 ell-cc (slow_resp)
012   : 24 ell-acc
013   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
014   : 03 tpl-acc-field
015   : 08 tpl-sts-field (PERMANENT_ERROR)
016   : 4005 tpl-cfg 0540 (AES_CBC_IV nb=4 cntn=0 ra=0 hc=0 )
018 CE: 71E9615249EDE52EAAE09F61908F027C3877F3330AE9079528B23173CE124BCC255393E60B173C0A9F274C42DD92E4B23C14E8A41F042903358DF01DD9268AD4 encrypted

{
    "media":"water",
    "meter":"unknown",
    "name":"",
    "id":"22917370",
    "timestamp":"1970-01-01T00:00:00Z"
}
```

### Logged telegrams from meter in decrypted format

```shell
// I'm trying to get encryption key but maybe someone has it already
T1;1;1;2023-05-11 09:05:20.000;118;123;22917370;0x5144b4097073912218078c00427a2108400515eb40de487a17e9154919ebd8200cfe528e5f3b5329d9536ebee743e7d2a74401d9230f2ee47daed0610a1dff81909bf4152cb479630ffb1034f90586b02148
T1;1;1;2023-05-11 09:05:20.000;122;144;22917370;0x5144b4097073912218078c00427a2108400515eb40de487a17e9154919ebd8200cfe528e5f3b5329d9536ebee743e7d2a74401d9230f2ee47daed0610a1dff81909bf4152cb479630ffb1034f90586b02148
T1;1;1;2023-05-11 09:06:20.000;117;120;22917370;0x5144b4097073912218078c00437a22084005bc4b8ed0143a1c9bd0348862ed2ef87a42b1b7b78a130fdc8d7e30e5b80a177f5887f3415a2bc5118a85982b1e189c34c5431d155417f729cf5a1045fb5c3dc6
T1;1;1;2023-05-11 09:06:20.000;123;153;22917370;0x5144b4097073912218078c00437a22084005bc4b8ed0143a1c9bd0348862ed2ef87a42b1b7b78a130fdc8d7e30e5b80a177f5887f3415a2bc5118a85982b1e189c34c5431d155417f729cf5a1045fb5c3dc6
T1;1;1;2023-05-11 09:07:20.000;131;144;22917370;0x5144b4097073912218078c00447a23084005b5a71c4b25cf79bd76119b631b085929c4be0c07159bdcfe82482982b27d57ac1a0289afdde8f9f41ca078d1dd44428975a5d6ed08219b281fdb34ddeaf53240
T1;1;1;2023-05-11 09:07:20.000;124;131;22917370;0x5144b4097073912218078c00447a23084005b5a71c4b25cf79bd76119b631b085929c4be0c07159bdcfe82482982b27d57ac1a0289afdde8f9f41ca078d1dd44428975a5d6ed08219b281fdb34ddeaf53240
T1;1;1;2023-05-11 09:08:21.000;121;119;22917370;0x5144b4097073912218078c00457a24084005c301ee824a951cbcac26fa9649e7f6e8488b674c160b09c32560d8c4ebd34ec21cd9ccb2647f3dab8482328d25bad0fb0a3ba255d52bdb02539af7e62f66e69d
T1;1;1;2023-05-11 09:08:21.000;123;126;22917370;0x5144b4097073912218078c00457a24084005c301ee824a951cbcac26fa9649e7f6e8488b674c160b09c32560d8c4ebd34ec21cd9ccb2647f3dab8482328d25bad0fb0a3ba255d52bdb02539af7e62f66e69d
T1;1;1;2023-05-11 09:09:21.000;121;125;22917370;0x5144b4097073912218078c00467a25084005e4f091fdbead5f1837babd1899a30332e8c06def8d5c395d36e09d2443f898e8c57d04780c45b656fc23a15fa21d5c60f7ae1b66e2af61745dad668000a3d951
T1;1;1;2023-05-11 09:09:21.000;123;142;22917370;0x5144b4097073912218078c00467a25084005e4f091fdbead5f1837babd1899a30332e8c06def8d5c395d36e09d2443f898e8c57d04780c45b656fc23a15fa21d5c60f7ae1b66e2af61745dad668000a3d951
T1;1;1;2023-05-11 09:10:21.000;126;121;22917370;0x5144b4097073912218078c00477a260840054e801ad210cd301c535b2cd528e4b3a53b1e2e0f1eebf1d6723c02bc57dbca2a9a6f884aafcdd38a69ec5da3b5fd854564a287504535b33aebd17b4e9bdee76b
T1;1;1;2023-05-11 09:10:21.000;124;137;22917370;0x5144b4097073912218078c00477a260840054e801ad210cd301c535b2cd528e4b3a53b1e2e0f1eebf1d6723c02bc57dbca2a9a6f884aafcdd38a69ec5da3b5fd854564a287504535b33aebd17b4e9bdee76b
```

weetmuts, 2023-05-12:
Thanks, do you have the decryption key for the meter?

weetmuts, 2023-05-12:
Thanks! The all-zeroes key works. I have pushed a new driver for this adapter. Please test!

./build/wmbusmeters --format=json 5144b4097073912218078c00247a0308400571e9615249ede52eaae09f61908f027c3877f3330ae9079528b23173ce124bcc255393e60b173c0a9f274c42dd92e4b23c14e8a41f042903358df01dd9268ad4 MyWater iwmtx5 22917370 00000000000000000000000000000000

{
    "media":"water",
    "meter":"iwmtx5",
    "name":"MyWater",
    "id":"22917370",
    "meter_datetime":"2023-05-11 10:38:24",
    "total_m3":0.025,
    "status":"PERMANENT_ERROR",
    "timestamp":"2023-05-12T09:14:30Z"
}

weetmuts, 2023-05-12:
You can also view the new decode here:
https://wmbusmeters.org/analyze/5144b4097073912218078c00247a0308400571e9615249ede52eaae09f61908f027c3877f3330ae9079528b23173ce124bcc255393e60b173c0a9f274c42dd92e4b23c14e8a41f042903358df01dd9268ad4:auto:00000000000000000000000000000000

maacaron, 2023-05-15:
Hi, thanks for adding driver! Regarding to decoding on webpage it looks fine. But I have an error on home assistant addon:
```
(meter) WaterMeterGarden: meter detection did not match the selected driver ! correct driver is: unknown!
(meter) Not printing this warning again for id: 22917370 mfct: (BMT) BMETERS, Italy (0x9b4) type: Water meter (0x07) ver: 0x18
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
/run.sh: line 109:   258 Segmentation fault      (core dumped) /wmbusmeters/wmbusmeters --useconfig=$CONFIG_DATA_PATH
s6-rc: info: service legacy-services: stopping
2023/05/15 12:48:46 [notice] 83#83: signal 15 (SIGTERM) received from 81, exiting
2023/05/15 12:48:46 [notice] 268#268: exiting
2023/05/15 12:48:46 [notice] 268#268: exit
2023/05/15 12:48:46 [notice] 83#83: signal 17 (SIGCHLD) received from 268
2023/05/15 12:48:46 [notice] 83#83: worker process 268 exited with code 0
2023/05/15 12:48:46 [notice] 83#83: exit
```
BUT! I installed edge version of an addon and it works! So I'm waiting for update in non-edge addon version :)

maacaron, 2023-07-03:
@weetmuts tested 1.14.0 and everything works fine! :)

closed 2023-07-03 by maacaron