wmbusmeters/wmbusmeters #959: (apator162) telegram contains a register (88) with unknown size.

travnick, 2023-05-12:
### Type of request

Existing device improvement

### Meter description

apator162, key is null.

It decodes as
```
Auto driver  : apator162
Best driver  : apator162 04/77
Using driver : apator162 00/00
000   : 3c length (60 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 0106 dll-mfct (APA)
004   : 85619401 dll-id (01946185)
008   : 05 dll-version
009   : 07 dll-type (Water meter)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : 28 tpl-acc-field
012   : 00 tpl-sts-field (OK)
013   : 3085 tpl-cfg 8530 (bidirectional AES_CBC_IV nb=3 cntn=0 ra=0 hc=0 )
015   : 2f2f decrypt check bytes (OK)
017 C?: 0F manufacturer specific data 064CB59718020043A0068300055A2D69610156BB0C101B1208007101A605080035D50700A030ECB903FFFF625DF341
025 C?: *** 43-A006
028 C?: *** 83-00055A2D69610156BB0C
039 C!: *** 10-1b120800 total consumption (528.923000 m3)
044 C?: *** 71-01A605080035D50700
054 C?: *** A0-30ECB903

{
    "media":"water",
    "meter":"apator162",
    "name":"",
    "id":"01946185",
    "total_m3":528.923,
    "timestamp":"2023-05-12T06:40:56Z"
}
```

but logs warning mentioned below

### Logged telegrams from meter in decrypted format

```shell
Started config hextty on stdin listening on any
(TPL) warning: decryption received less bytes than expected for decryption! Got 46 bytes but expected at least 48 bytes since num encr blocks was 3.
(TPL) warning: decryption received non-multiple of 16 bytes! Got 46 bytes shrinking message to 32 bytes.
(apator162) telegram contains a register (88) with unknown size.
Please open an issue at https://github.com/wmbusmeters/wmbusmeters/
and report this telegram: 3C4401068561940105077A280030852F2F0F064CB59718020043A0068300055A2D69610156BB0C101B1208007101A60AC5AA6DE6A5F0880E9ADD08393C
```
```

weetmuts, 2023-05-12:
Fix pushed! Please test!

travnick, 2023-05-12:
You are lightning-fast!
The mentioned telegram results with
```
Started config hextty on stdin listening on any
(wmbus) WARNING!! telegram should have been fully encrypted, but was not! id: 01946185 mfct: (APA) Apator, Poland (0x601) type: Water meter (0x07) ver: 0x05
(meter) newly created meter (apator01946185 01946185 apator162) did not handle telegram!
```

And there are new ones:
```
Started config hextty on stdin listening on any
(TPL) warning: decryption received less bytes than expected for decryption! Got 46 bytes but expected at least 48 bytes since num encr blocks was 3.
(TPL) warning: decryption received non-multiple of 16 bytes! Got 46 bytes shrinking message to 32 bytes.
(apator162) telegram contains a register (52) with unknown size.
Please open an issue at https://github.com/wmbusmeters/wmbusmeters/
and report this telegram: 3C4401068561940105077A430030852F2F0F9B4CB59718020043A00684A867624584F433500000102C1208007101A61BAC935F170B2C52E209991FAEA8

Started config hextty on stdin listening on any
(TPL) warning: decryption received less bytes than expected for decryption! Got 46 bytes but expected at least 48 bytes since num encr blocks was 3.
(TPL) warning: decryption received non-multiple of 16 bytes! Got 46 bytes shrinking message to 32 bytes.
(apator162) telegram decoding fails since last register (83 size 10) does not
align with telegram size 48 > 44.
Please open an issue at https://github.com/wmbusmeters/wmbusmeters/
and report this telegram: 3C4401068561940105077A440030852F2F0F9C4CB59718020043A0068300055A2D69610156BB0C102C1208007101A6EA2C578342287A836CE2B1A2902B

Started config hextty on stdin listening on any
(TPL) warning: decryption received less bytes than expected for decryption! Got 46 bytes but expected at least 48 bytes since num encr blocks was 3.
(TPL) warning: decryption received non-multiple of 16 bytes! Got 46 bytes shrinking message to 32 bytes.
(apator162) telegram contains a register (48) with unknown size.
Please open an issue at https://github.com/wmbusmeters/wmbusmeters/
and report this telegram: 3C4401068561940105077A450030852F2F0F9C4CB59718020043A00684A867624584F433500000102C1208007101A6FF198892D7B0F34895B3D5CDB9F7

Started config hextty on stdin listening on any
(TPL) warning: decryption received less bytes than expected for decryption! Got 46 bytes but expected at least 48 bytes since num encr blocks was 3.
(TPL) warning: decryption received non-multiple of 16 bytes! Got 46 bytes shrinking message to 32 bytes.
(apator162) telegram contains a register (4b) with unknown size.
Please open an issue at https://github.com/wmbusmeters/wmbusmeters/
and report this telegram: 3C4401068561940105077A460030852F2F0F9D4CB59718020043A0068300055A2D69610156BB0C102C1208007101A616534A7FE7E95E4BD362E2F04936

Started config hextty on stdin listening on any
(TPL) warning: decryption received less bytes than expected for decryption! Got 46 bytes but expected at least 48 bytes since num encr blocks was 3.
(TPL) warning: decryption received non-multiple of 16 bytes! Got 46 bytes shrinking message to 32 bytes.
(apator162) telegram decoding fails since last register (75 size 25) does not
align with telegram size 63 > 44.
Please open an issue at https://github.com/wmbusmeters/wmbusmeters/
and report this telegram: 3C4401068561940105077A480030852F2F0F9E4CB59718020043A0068300055A2D69610156BB0C102C1208007101A60CA04BE92971BD750AB938CC1D44
```

weetmuts, 2023-05-12:
Clearly it is unhappy with the decryption process. How do you feed wmbusmeters? Are you decrypting using an all-zeroes key?

travnick, 2023-05-12:
yes, it's all zeros. 

I have a `exec_wmbusmeters` wrapper (to make `OpenHAB` happy):
```
#!/bin/sh

echo $1 | wmbusmeters --useconfig=/etc
```
and just execute it as `exec_wmbusmeters 3C4401068561940105077A280030852F2F0F064CB59718020043A0068300055A2D69610156BB0C101B1208007101A60AC5AA6DE6A5F0880E9ADD08393C`

cat /etc/wmbusmeters.conf 
```
loglevel=normal
# Remember to change auto here to the device you are going to use in production.
device=stdin:hex
logtelegrams=false
format=json
meterfiles=/var/lib/wmbusmeters/meter_readings
meterfilesaction=append
logfile=/var/log/wmbusmeters/wmbusmeters.log
shell=echo "$METER_JSON"
```

weetmuts, 2023-05-12:
Can you capture some content with this change:
echo $1 > /tmp/content
echo $1 | wmbusmeters --useconfig=/etc

The print the content of /tmp/content here.

travnick, 2023-05-12:
```
3c4401068561940105077a8600308542d72912e7b3f2b7c0fef9d416b35d056531d6a7dfd2326805d41e2d3bc02b31951783fe45eeaf54adc544317a32751b33cc
3c4401068561940105077a87003085d3f3e6fca8a1cfee597852da658e1242a2899787491b7c3bf2f82f4c539e0d7ce8952f099eb0bc03759f979f0608cb422e9a
3c4401068561940105077a8900308566b6bdeab04d2a5cb493f4711da192cc2ddae12ff56ddb9e19caf2a89cb832e652a27d60e46a0b7f484bbd0b1a9d19a2c10f
3c4401068561940105077a8a00308580758919a70bb281b9bc4ddd11a2eb653ce78818012261ef139c17bac2998a55ba04e44e8adfd2abf4731d4742d504729d61
3c4401068561940105077a8b003085c320441ca50445d5795282434b267ddcac9db191c808d32e4e0e5748be4302f308adb09fa7313d32d8eaead40665cdc43363
3c4401068561940105077a8c00308566487596e3069266923e2474f3cfe4071bdb2187f69e039a6f4234c534deb68a5fd0230e8d9d2ba0f9038552a2b76a365ad2
3c4401068561940105077a920030850df89644450f81b79271fd7d96bc927232ae3726685c3a0c46ddb6338cd74b75ecd40b0c108d66aa2286301e1276f87c4693
3c4401068561940105077a930030859acbe9e47be8709dc8d67d43708a48cf0063ee88175d62e0b247450ee20dfd3e4e4e56f34a3be647e795df731b286b3b2ddf
3c4401068561940105077a950030858540431f83c7365bba1b291cd4583c6d6a4cbfe24f347b74a7625cee7571e632f2da5a964e327b320e73ff804bdb19e0af9e
3c4401068561940105077a960030856fd8cf7cb118d472c91dbf7e747b65794ca11a7267f2f77fbe2182625243208d2dce8d1f9f6dd78c3b24a49ea04a1755fe32
3c4401068561940105077a98003085810246e600c7e67ee8786fd37edf589bd883ee34d1e4f9f0a6197c93884f432254415b50c85e172881b304b2a0a5e9fd15c7
3c4401068561940105077a990030855d85204d51b9b30d03c796901cc5475eb6d4bcce48445509ec91599629588c48fe3a18bd87cc06146546d7bd0effce97241c
3c4401068561940105077a9b00308511075255079675e0cdd8b2cddd395e8f3b1ac11339264f066ad0904eed7b50ac50198c1f81463be9208387fc7e4c8d005bd4
3c4401068561940105077a9c003085cec5e3f08dd2e0659e61ec9ebb19bbb7f24a9618276ebc0d9254cf3b2e1295197541ee8c6d3f0a375ee4f2cefff9ac0bbb06
```

weetmuts, 2023-05-12:
Your telegrams have the wrong length byte 3c (two chars too short) and has 2 trailing garbage bytes (crcs?). How do you collect them?

Your telegrams should start with 3e. The stdin:hex driver uses the length byte to figure out how many bytes there are and cuts the telegram short, which means that the decryption fails because it require multiples of 16 bytes. I have pushed some warnings to make it easier to see when this is happening.

Store the hex: 3c4401068561940105077a8600308542d72912e7b3f2b7c0fef9d416b35d056531d6a7dfd2326805d41e2d3bc02b31951783fe45eeaf54adc544317a32751b33cc
into the file: content

Now run:
wmbusmeters stdin:hex Apa apator162 01946185 00000000000000000000000000000000 < content

Started config hextty on stdin listening on any
(TPL) warning: aes-cbc-iv decryption received less bytes than expected for decryption! Got 46 bytes but expected at least 48 bytes since num encr blocks was 3.
(TPL) warning: decryption received non-multiple of 16 bytes! Got 46 bytes shrinking message to 32 bytes.
(apator162) telegram contains a register (54) with unknown size.
Please open an issue at https://github.com/wmbusmeters/wmbusmeters/
and report this telegram: 3C4401068561940105077A860030852F2F0F496CB59718020043A0068300055A2D69610156BB0C10321208007101A6951783FE45EEAF54ADC544317A32
Apa	01946185	528.946 m³	2023-05-12 19:59.03

I.e. the register warning is just a side effect of the bad length. If you change the first byte 3c to 3e it will decode ok.

You can also supply the telegram on the command line like this:
wmbusmeters.g 3c4401068561940105077a8600308542d72912e7b3f2b7c0fef9d416b35d056531d6a7dfd2326805d41e2d3bc02b31951783fe45eeaf54adc544317a32751b33cc Apa apator162 01946185 00000000000000000000000000000000
(wmbus) telegram length byte (the first) 0x3c (60) is probably wrong. Expected 0x40 (64) based on the length of the telegram.
Apa	01946185	528.946 m³	2023-05-12 20:02.33

Here the command line does not need the length byte, since it uses all the bytes supplied, but now, with my latest fix pushed, it will warn. Note that the length is not 0x40 but 0x3e since if you run debug you will see that it finds 2 unencrypted bytes at the end. Since this meter does not use unencrypted bytes after the encrypted bytes, these are probably crcs from your  collector.

travnick, 2023-06-05:
@weetmuts I'm facing now strange issue.

In the `/tmp/content` file, there is `3c4401068561940105077a3f0030859ae0ded79ca9da11921710eef77c56591a80c4ad49a46b9a939f9330bc5eab55dc0fd3fa6ed13ef2d7cd8a4e17049575a473` but `wmbusmetters` logs

```
Started config hextty on stdin listening on any
(TPL) warning: aes-cbc-iv decryption received less bytes than expected for decryption! Got 46 bytes but expected at least 48 bytes since num encr blocks was 3.
(TPL) warning: decryption received non-multiple of 16 bytes! Got 46 bytes shrinking message to 32 bytes.
(apator162) telegram contains a register (f2) with unknown size.
Please open an issue at https://github.com/wmbusmeters/wmbusmeters/
and report this telegram: 3C4401068561940105077A3F0030852F2F0F5C45369718020043B80684A867627A7CF53451000010C23A080071013BDC0FD3FA6ED13EF2D7CD8A4E1704
```

Note that `3C4401068561940105077A3F0030852F2F0F5C45369718020043B80684A867627A7CF53451000010C23A080071013BDC0FD3FA6ED13EF2D7CD8A4E1704` is truncated and partially changed comparing to the original message.

```
wmbusmeters --version
wmbusmeters: 1.13.1-35-g53c589e
53c589ed111f9faf2b5d26049e18b545f5f47d19
```

closed 2023-09-05 by github-actions[bot]