wmbusmeters/wmbusmeters #960: logrotate with docker

jens62, 2023-05-13:
### Type of request

Question

### OS version

QTS 5.0.1.2376 (QNAP NAS)

### wmbusmeters version

1.12.0-16

### Your message goes here

I am looking for some hints how to configure `logrotate` in the docker enviroment.

Can anybody give me a hand?

BIBOLV, 2023-05-13:
There are multiple options
- Disable logging if there is not some specific need for logs. In case you would need them - can enable back later.
- Use logrotate functionality on QNAP, not sure if there is such, maybe there are some 3rd party packages that would enable such functionality.
- Write script that would do rotation - it should be possible to write scripts and schedule execution on QNAP?!
- Use specific [container](https://hub.docker.com/r/blacklabelops/logrotate) that would provide such functionality.

jens62, 2023-05-19:
Thanks for your reply.
I thought is missed something, because it is mentioned in the documentation that `make install` installs a. o. `/etc/logrotate.d/wmbusmeters`. I expected that it is a matter of configuration to use `logrote` in the docker environment as well.

I like docker because it reduces dependencies and complexity.
I want to avoid having additional dependencies/connections/relations to the wmbusmeters container from outside the container.

I looked into the wmbusmeters container with `docker exec -it wmbusmeters sh` and realized that `logrotate` is not part of it.
`cat /etc/os-release` gave me `Alpine Linux v3.14`. 

With 
```
apk update
apk upgrade
apk add logrotate
```
I added `logrotate` to the container.

I created a  `/wmbusmeters_data/etc/logrotate.conf `according to [scripts/prepare_logfiles.sh](https://github.com/wmbusmeters/wmbusmeters/blob/master/scripts/prepare_logfiles.sh):

```
cat  /wmbusmeters_data/etc/logrotate.conf

/wmbusmeters_data/logs/meter_readings/*
/wmbusmeters_data/logs/*.log {
    su root root
    rotate 12
    daily
    compress
    missingok
    postrotate
        /usr/bin/killall -HUP /wmbusmeters/wmbusmeters
    endscript
}

```

`logrotate` needs a cron daemon.
The standard cron daemon appears to be not the best solution inside a docker container.
I decided to use [supercronic](https://github.com/aptible/supercronic) and installed it with

```
export SUPERCRONIC_URL='https://github.com/aptible/supercronic/releases/download/v0.2.24/supercronic-linux-amd64'
export SUPERCRONIC='supercronic-linux-amd64'
export SUPERCRONIC_SHA1SUM='6817299e04457e5d6ec4809c72ee13a43e95ba41'

curl -fsSLO "$SUPERCRONIC_URL" \
 && echo "${SUPERCRONIC_SHA1SUM}  ${SUPERCRONIC}" | sha1sum -c - \
 && chmod +x "$SUPERCRONIC" \
 && mv "$SUPERCRONIC" "/usr/local/bin/${SUPERCRONIC}" \
 && ln -s "/usr/local/bin/${SUPERCRONIC}" /usr/local/bin/supercronic
```

I created a crontab for `logrotate`:

```
cat /wmbusmeters_data/etc/supercronic_crontab

0 * * * * /usr/sbin/logrotate /wmbusmeters_data/etc/logrotate.conf --state /wmbusmeters_data/etc/logrotate-state
```

Finally I adjusted` /wmbusmeters/docker-entrypoint.sh` to

```
#!/bin/sh

[ ! -d /wmbusmeters_data/logs/meter_readings ] && mkdir -p /wmbusmeters_data/logs/meter_readings
[ ! -d /wmbusmeters_data/etc/wmbusmeters.d ] && mkdir -p /wmbusmeters_data/etc/wmbusmeters.d
[ ! -f /wmbusmeters_data/etc/wmbusmeters.conf ] && echo -e "loglevel=normal\ndevice=auto:t1\ndonotprobe=/dev/ttyAMA0\nlogtelegrams=false\nformat=json\nmeterfiles=/wmbusmeters_data/logs/meter_readings\nmeterfilesaction=overwrite\nlogfile=/wmbusmeters_data/logs/wmbusmeters.log" > /wmbusmeters_data/etc/wmbusmeters.conf

/usr/local/bin/supercronic /wmbusmeters_data/etc/supercronic_crontab &
/wmbusmeters/wmbusmeters --useconfig=/wmbusmeters_data

```
in oder to run the `supercronic` cron daemon in the background.

This solution works for me.
I am aware that I will have an issue when I update the container from the repository - my changes will be lost.
Maybe the container maintainers decide to add `logrotate`, `logrotate.conf`, `supercronic`, a `supercronic_crontab` and an adjusted  `docker-entrypoint.sh` to the next wmbusmeters container version ;-)

BIBOLV, 2023-05-20:
I will close second opened issue regarding this topic - let`s keep discussion and tracking under one issue.

There is indeed logrotate configs included, but that all is for standalone version and was not included in docker. Another reason logrotate not being in docker is that IMO it is much simpler to manage logs from host side and leave docker only for app purposes, but I understand why this is issue for you. You have already done great job on making necessary changes and testing them so I do not see big issue if we would add it to container with few buts:
- Can you test if logrotate config is removed if it will not cause issues for docker functionality. Idea is - if someone would like to disable logrotate functionality withing docker, how it should be done correctly that does not compromise container functionality.
- I`m not fond of including some specific version of software in docker file - it would mean that constant followup and updates would be needed, It will make thing even more complex to build for multiple architectures.

 supercronic is included as package starting from Alpline 3.15, so let`s give some time for multiarch/alpine repo to update their base image to more latest one and then we could implement this properly. If you notice this happening you are more then welcome to create PR, meanwhile I will mark this issue as planned.

Regarding your local changes - you can use `wmbusmeters/wmbusmeters:release-1.13.1` docker image for example, if everything is working for you and you do not need latest changes, that way your local container would not be recreated and you would not lose local changes.