wmbusmeters/wmbusmeters #961: rtlwmbus listening on none - Is my custom_config correct ?

Pascal66, 2023-05-13:
### Type of request

Question

### OS version

OS: Linux 6.1.25

### wmbusmeters version

1.13.1-20-g6a14bb8

### Your message goes here

```
s6-rc: info: service legacy-services successfully started
[13:26:18] INFO: CONFIG_CONF ...
loglevel=debug device=rtlwmbus:CMD(rtl_sdr -f 169.4M -g 37.2 -s 1600000 - 2>/dev/null | rtl_wmbus -p   s -a) donotprobe=/dev/ttyAMA0;/dev/ttyACM0;/dev/ttyUSB0 logtelegrams=true format=json logfile=/dev/stdout shell=/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"
[13:26:18] INFO: 
[13:26:18] INFO: CONFIG_METERS ...
[13:26:18] INFO: [{"name":"ExampleMeter","driver":"amiplus","id":"12345678","key":"NOKEY"}]
[13:26:18] INFO: Syncing wmbusmeters configuration ...
[13:26:19] INFO: Registering meters ...
[13:26:19] INFO: Adding meter-0001 ...
[13:26:19] INFO: Generating MQTT configuration ... 
[13:26:21] INFO: Broker core-mosquitto will be used.
[13:26:22] INFO: MQTT Discovery cleanup...
[13:26:23] INFO: Starting web configuration service.
[13:26:23] INFO: Running wmbusmeters ...
(config) "device" "rtlwmbus:CMD(rtl_sdr -f 169.4M -g 37.2 -s 1600000 - 2>/dev/null | rtl_wmbus -p   s -a)"
(wmbusmeters) version: 1.13.1-20-g6a14bb8
(config) using device: rtlwmbus:CMD(rtl_sdr -f 169.4M -g 37.2 -s 1600000 - 2>/dev/null | rtl_wmbus -p   s -a) 
(config) number of meters: 1
(serial) expecting devices to work
(serial) expecting devices to work
(lookup) with cmd "rtlwmbus:CMD(rtl_sdr -f 169.4M -g 37.2 -s 1600000 - 2>/dev/null | rtl_wmbus -p   s -a)"
(main) opening rtlwmbus:CMD(rtl_sdr -f 169.4M -g 37.2 -s 1600000 - 2>/dev/null | rtl_wmbus -p   s -a)
Started config rtlwmbus listening on none using CMD(rtl_sdr -f 169.4M -g 37.2 -s 1600000 - 2>/dev/null | rtl_wmbus -p   s -a)
(rtlwmbus) using command: rtl_sdr -f 169.4M -g 37.2 -s 1600000 - 2>/dev/null | rtl_wmbus -p   s -a
(rtlwmbus) opening cmd_0
(bgshell) exec background "/bin/sh"
(serial) EVENT thread interrupted
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -f 169.4M -g 37.2 -s 1600000 - 2>/dev/null | rtl_wmbus -p   s -a"
(serialcmd) opened /bin/sh pid 193 fd 3 (rtlwmbus)
(main) regular reset of rtlwmbus  using CMD(rtl_sdr -f 169.4M -g 37.2 -s 1600000 - 2>/dev/null | rtl_wmbus -p   s -a) will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
[13:26:26] INFO: Starting Nginx...
2023/05/13 13:26:26 [notice] 73#73: using the "epoll" event method
2023/05/13 13:26:26 [notice] 73#73: nginx/1.22.1
2023/05/13 13:26:26 [notice] 73#73: OS: Linux 6.1.25
2023/05/13 13:26:26 [notice] 73#73: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2023/05/13 13:26:26 [notice] 73#73: start worker processes
2023/05/13 13:26:26 [notice] 73#73: start worker process 210
```

BIBOLV, 2023-05-13:
What meter you want to read? If it is indeed 169MHz then support for N modes are not yet implemented in rtl-wmbus. You can check [this](https://github.com/wmbusmeters/wmbusmeters/issues/321) or other issues related to this topic.

Pascal66, 2023-05-14:
It's a french Maddelana water.
BTW, it's more to check the command line I used for now.
I dont know why it say : "Started config rtlwmbus listening on none.."
Is my raphael usb key is recognized ? as is show when I use "auto:t1" instead of CMD
I saw all related issue, and seems mode N was implemented ?
Can it be preimplemented in the futur with mode "auto:N" ?

weetmuts, 2023-07-02:
Note that there is a 169MHz hardware dongle that works with wmbusmeters: amb3665
(Search for Amber AMB 3665-M)
Make sure you buy a matching suitable antenna as well.

weetmuts, 2023-07-02:
If it is N2 then we need to send a wakeup message to the meter. Assuming you have the meter key, then this can be implemented.

weetmuts, 2023-07-02:
it says "listening on none" because rtlwmbus always listens to everything c1 and t1.  So I have not bothered to make the informational message print a better string.

closed 2023-10-01 by github-actions[bot]