wmbusmeters/wmbusmeters #965: Incorrect handling of negative values

demetz, 2023-05-20:
### Type of request

Bug

### OS version

Linux

### wmbusmeters version

1.13.1

### Your message goes here

Negative values are incorrectly parsed.

Example: 
https://wmbusmeters.org/analyze/3b44a7320556637104047afa2000202f2f0c06774202000c14399956000b2d0200f00b3b3018000a5a51030a5e520302fd170000066d0c080af42500

is parsed as
```
029   : 0B dif (6 digit BCD Instantaneous value)
030   : 2D vif (Power 10² W)
031 C!: 0200F0 ("power_kw":220000.2)
```
however, this should be `-200W`

I've seen this for both `power_kw` as well as for `flow_m3h`

demetz, 2023-05-20:
I believe this is linked to https://github.com/wmbusmeters/wmbusmeters/pull/480

weetmuts, 2023-05-20:
Yes, there is a simple fix for the values that can be negative, replace:

 VifScaling::Auto
with 
 VifScaling::AutoSigned

Do you want to do a pull request?

weetmuts, 2023-05-20:
Thanks! I discovered some error flags and status bits in your telegram so I added the status field to the ultraheat driver.  Code pushed.

Do you have a manual for your meter?  Currently your tpl status has a mfct specific bit set.
We can add better translation of the bit, which is currently reported as UNKNOWN_20

demetz, 2023-05-22:
I've requested them from the manufacturer...

pvagner, 2023-05-22:
Hello,
I have got Lansen [LAN-WMBUS-CX-TH](https://www.lansensystems.com/assortment/sensors/air-quality/lan-wmbus-cx-th/) temperature and humidity sensor.
The wmbus frames are encrypted with a key. When reading negative values these are reported as very high numbers.
See the analyze output below.
Can you please advice how to receive correctly formatted negative temperature values in the output please?
I have rebuilt the app with the recent patch.

```
./build/wmbusmeters --analyze=<key_redacted> 2e44333041000600091b7aa700202585a14580592a8ac1413733152d98c9257c74660f5831bc7771136765b81575c7 
Auto driver  : not found!
Best driver  : lansenth 09/11
Using driver : lansenth 09/11
000   : 2e length (46 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 3330 dll-mfct (LAS)
004   : 41000600 dll-id (00060041)
008   : 09 dll-version
009   : 1b dll-type (Room sensor (eg temperature or humidity))
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : a7 tpl-acc-field
012   : 00 tpl-sts-field (OK)(OK)
013   : 2025 tpl-cfg 2520 (synchronous AES_CBC_IV nb=2 cntn=0 ra=0 hc=0 )
015   : 2f2f decrypt check bytes (OK)
017   : 02 dif (16 Bit Integer/Binary Instantaneous value)
018   : 65 vif (External temperature 10⁻² °C)
019 C!: DBF9 ("current_temperature_c":639.63)
021   : 42 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
022   : 65 vif (External temperature 10⁻² °C)
023 C!: FC04 ("average_temperature_1h_c":12.76)
025   : 82 dif (16 Bit Integer/Binary Instantaneous value)
026   : 01 dife (subunit=0 tariff=0 storagenr=2)
027   : 65 vif (External temperature 10⁻² °C)
028 C!: 6109 ("average_temperature_24h_c":24.01)
030   : 01 dif (8 Bit Integer/Binary Instantaneous value)
031   : FB vif (First extension FB of VIF-codes)
032   : 1B vife (Relative Humidity 10^0 %)
033 C!: 2C ("current_relative_humidity_rh":44)
034   : 41 dif (8 Bit Integer/Binary Instantaneous value storagenr=1)
035   : FB vif (First extension FB of VIF-codes)
036   : 1B vife (Relative Humidity 10^0 %)
037 C!: 23 ("average_relative_humidity_1h_rh":35)
038   : 81 dif (8 Bit Integer/Binary Instantaneous value)
039   : 01 dife (subunit=0 tariff=0 storagenr=2)
040   : FB vif (First extension FB of VIF-codes)
041   : 1B vife (Relative Humidity 10^0 %)
042 C!: 29 ("average_relative_humidity_24h_rh":41)
043   : 02 dif (16 Bit Integer/Binary Instantaneous value)
044   : 23 vif (On time days)
045 C?: BB00

{
    "media":"room sensor",
    "meter":"lansenth",
    "name":"",
    "id":"00060041",
    "status":"OK",
    "current_temperature_c":639.63,
    "current_relative_humidity_rh":44,
    "average_temperature_1h_c":12.76,
    "average_relative_humidity_1h_rh":35,
    "average_temperature_24h_c":24.01,
    "average_relative_humidity_24h_rh":41,
    "timestamp":"2023-05-22T12:37:09Z"
}
```

demetz, 2023-08-11:
according to the manufacturer, the values of the status byte are to be decoded as:
Bit 0..4: according to  EN 13757-3 (can be found in [OMS Specification](https://www.bsi.bund.de/SharedDocs/Downloads/DE/BSI/Publikationen/TechnischeRichtlinien/TR03109/TR-03109-1_Anlage_Feinspezifikation_Drahtlose_LMN-Schnittstelle.pdf?__blob=publicationFile&v=1) in section 4.2.3.2)
Bit 5: 1 = negative power
Bit 6: 1 = negative flow
Bit 7: 1 = negative temperature difference

closed 2023-11-10 by github-actions[bot]

weetmuts, 2023-11-12:
@demetz  On which page did you find that reference?

I have now fixed the code so that bcd values are always detected as negative if they are marked with nybble F.

weetmuts, 2024-05-06:
https://github.com/wmbusmeters/wmbusmeters/issues/1238