wmbusmeters/wmbusmeters #967: Receiving no telegrams - config wrong?

sebihimself, 2023-05-20:
### Type of request

Question

### HAOS version

2023.5.3 10.1

### wmbusmeters version

1.13.1-23

### Your message goes here

Hi, I'm trying to run wmbusmeters on my home assistant but I just received 1 single telegram in 24h although running rtl_433 on my mac I see there should be constantly incoming. Any idea what is wrong?

https://pastebin.com/7xaXJ4Mh

thecem, 2023-05-20:
What for meter?
Which frequency?

log telegrams true?

sebihimself, 2023-05-20:
It is a techem (3x water 1x heat) at 868.95 and I tried a lot of frequencies between 868.90-868.99. I was only able to receive one telegram once at 868.92:

`(rtlwmbus) checkRTLWMBusFrame "C1;1;1;2023-05-19 18:54:40.000;128;136;22316317;0x564468501763312295377263142474a85cfe06040040053b03a652e3b9b182afcd0672461a8a28332d9a6e3083b671a8015bd38f8539e8e48ce53889f82bceb9c2aa76b4ebabfb85431298eae0edadfce5ba3c05138ce1<0A>"
(rtlwmbus) received full frame
(meter) no meter handled  checking 0 templates.
(wmbus) parseDLL @0 87
(wmbus) parseELL @10 77
(wmbus) parseNWL @10 77
(wmbus) parseAFL @10 77
(wmbus) parseTPL @10 77
(wmbus) parseDLL @0 87
(telegram) DLL L=56 C=44 (from meter SND_NR) M=5068 (TCH) A=22316317 VER=95 TYPE=37 (Radio converter (meter side)) (driver mkradio4a) DEV=rtlwmbus[00000001] RSSI=128
(wmbus) parseELL @10 77
(wmbus) parseNWL @10 77
(wmbus) parseAFL @10 77
(wmbus) parseTPL @10 77
(TPL) num encrypted blocks 4 (64 bytes and remaining unencrypted 0 bytes)
(telegram) TPL CI=72 ACC=04 STS=00 CFG=0540 (AES_CBC_IV nb=4 cntn=0 ra=0 hc=0) ID=63142474 MFT=a85c VER=fe TYPE=06 (Warm Water (30°C-90°C) meter)
Received telegram from: 22316317
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Radio converter (meter side) (0x37) encrypted
                   ver: 0x95
      Concerning meter: 74241463
          manufacturer: (WEH) E Wehrle, Germany (0x5ca8)
                  type: Warm Water (30°C-90°C) meter (0x06) encrypted
                   ver: 0xfe
                device: rtlwmbus[00000001]
                  rssi: 128 dBm
                driver: mkradio4a
(wmbus) 000   : 56 length (86 bytes)
(wmbus) 001   : 44 dll-c (from meter SND_NR)
(wmbus) 002   : 6850 dll-mfct (TCH)
(wmbus) 004   : 17633122 dll-id (22316317)
(wmbus) 008   : 95 dll-version
(wmbus) 009   : 37 dll-type (Radio converter (meter side))
(wmbus) 010   : 72 tpl-ci-field (EN 13757-3 Application Layer (long tplh))
(wmbus) 011   : 63142474 tpl-id (74241463)
(wmbus) 015   : a85c tpl-mfct (WEH)
(wmbus) 017   : fe tpl-version
(wmbus) 018   : 06 tpl-type (Warm Water (30°C-90°C) meter)
(wmbus) 019   : 04 tpl-acc-field
(wmbus) 020   : 00 tpl-sts-field (OK)
(wmbus) 021   : 4005 tpl-cfg 0540 (AES_CBC_IV nb=4 cntn=0 ra=0 hc=0 )
(wmbus) 023 CE: 3B03A652E3B9B182AFCD0672461A8A28332D9A6E3083B671A8015BD38F8539E8E48CE53889F82BCEB9C2AA76B4EBABFB85431298EAE0EDADFCE5BA3C05138CE1 encrypted
telegram=|_564468501763312295377263142474A85CFE06040040053B03A652E3B9B182AFCD0672461A8A28332D9A6E3083B671A8015BD38F8539E8E48CE53889F82BCEB9C2AA76B4EBABFB85431298EAE0EDADFCE5BA3C05138CE1|+31
(wmbus) telegram from 22316317,74241463 ignored by all configured meters!
`

so I'm wondering why there are not more incoming?

closed 2023-08-20 by github-actions[bot]