wmbusmeters/wmbusmeters #970: Adapting izar for Python

n3myy, 2023-05-27:
### Type of request

Question

### OS version

CentOS/Windows

### wmbusmeters version

1.13.1

### Your message goes here

Great work with this project, it has helped me so much gaining understanding of mbus format.

I am working on a project to decrypt and parse DIEHL water meters telegrams sent to me via AMQ and I would like to automate this using Python (I am not clued up on C as I would like to be).

I require a function in python to call which I can provide the telegram and key and return the parsed json response.

Example: print(izar_driver('telegramhexhere', 'keyhere'))

Would it be possible to use your izar driver for this process? Is there a way to adapt this to python? Or if you have any reference for me to write my own into Python that would be a huge help.

Thanks in advance.

BIBOLV, 2023-05-27:
I would suggest to use some existing wmbus python project to process telegrams [like](https://github.com/gewv-tu-dresden/pyWirelessMbus) this or [this](https://github.com/CBrunsch/scambus), or you can search around - I'm sure there are more. Then you can proceed with specific meter decoding implementation, but izar meter use PRIOS proprietary protocol, so it will be even more complex than standard meter that uses OMS standard.

weetmuts, 2023-05-28:
You can also invoke the wmbusmeters binary from python. 

Something like:
import subprocess
subprocess.run(["wmbusmeters", "--format=json", "214412345678.......telegramhex", "MeterName", "izar", "12345678", "NOKEY" ])
 
and then catching the stdout which will be the json.

n3myy, 2023-06-03:
Thanks both, in the end if figured out the LFSR ecryption and that was enough for me map out the bytes I needed.

Thanks again for your help!

closed 2023-06-03 by n3myy