wmbusmeters/wmbusmeters #976: Apator08 on Home Assistant addon

filipsio, 2023-06-07:
### Type of request

Bug

### HAOS version

2023.5.3

### wmbusmeters version

1.11.0 

### Your message goes here

Hi.
I'm trying to make work Apator-08 water add-on. I manage to read telegram, however on HA add on there is missing apator08.json file to parse it correctly, only apator-162.json is available which does not parse value correctly.

My add-on config:
```
conf: |-
  loglevel=normal
  device=rtlwmbus:868.9M:t1
  logtelegrams=true
  format=json
  meterfilesaction=overwrite
  meterfiles=/config/wmbusmeters/logs/meter_readings/
  logfile=/config/wmbusmeters/wmbusmeters.log
  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters:
  - |-
    name=GardeningWater
    driver=apator162
    id=000c46a0
    key=00000000000000000000000000000000
  - |-
    name=GardeningWater2
    driver=apator162
    id=00072fd2
    key=00000000000000000000000000000000
```

When driver is set to apator08
```
 Adding meter: GardeningWater ...
  File /config/wmbusmeters/etc/mqtt_discovery/apator08.json not found.
```

Telegrams:

```
(meter) GardeningWater2: meter detection did not match the selected driver apator162! correct driver is: apator08
(meter) Not printing this warning again for id: 00072fd2 mfct: (APT) Unknown (0x8614) type: Gas meter (0x03) ver: 0x03
telegram=|89441486D22F07000303A0_6F482800D861002A1F080607061734B090FE0106FF05694828006948280069482800694828006948280069482800694828006948280069482800694828006948280069482800030000000037B50E1F000000002000000000000023E8B5B3102418090A12031527000000280000000000002C0000000000002F000000000000|+73
(apator162) telegram contains a register (1f) with unknown size.
Please open an issue at https://github.com/weetmuts/wmbusmeters/
and report this telegram: 89441486D22F07000303A06F482800D861002A1F080607061734B090FE0106FF05694828006948280069482800694828006948280069482800694828006948280069482800694828006948280069482800030000000037B50E1F000000002000000000000023E8B5B3102418090A12031527000000280000000000002C0000000000002F000000000000

(apator162) telegram contains a register (21) with unknown size.
Please open an issue at https://github.com/weetmuts/wmbusmeters/
and report this telegram: 89441486D22F07000303A06F482800D861003321080607061734B090FE0106FF05694828006948280069482800694828006948280069482800694828006948280069482800694828006948280069482800030000000037B50E1F000000002000000000000023ECB7B3102418090A12031527000000280000000000002C0000000000002F000000000000

(meter) GardeningWater: meter detection did not match the selected driver apator162! correct driver is: apator08
(meter) Not printing this warning again for id: 000c46a0 mfct: (APT) Unknown (0x8614) type: Gas meter (0x03) ver: 0x03
telegram=|73441486A0460C000303A0_501B0100D861348D180B0206FF05A8110100A8110100A8110100A9110100AA110100A8110100A8110100A8110100A8110100A8110100A8110100A8110100030000000037600D20000000000000242A1A060F0515280000000000002C0000000000002F000000000000|+228
(apator162) telegram contains a register (18) with unknown size.
Please open an issue at https://github.com/weetmuts/wmbusmeters/
and report this telegram: 73441486A0460C000303A0501B0100D861348D180B0206FF05A8110100A8110100A8110100A9110100AA110100A8110100A8110100A8110100A8110100A8110100A8110100A8110100030000000037600D20000000000000242A1A060F0515280000000000002C0000000000002F000000000000
```

filipsio, 2023-06-07:
While working with this issue, version 1.13.1  from  ha-addon repository behaves the same.

When using apator162 driver
`(apator162) telegram contains a register (05) with unknown size.
Please open an issue at https://github.com/wmbusmeters/wmbusmeters/
and report this telegram: 89441486D22F07000303A06F482800D8610019050F0607061734B090FE0106FF05694828006948280069482800694828006948280069482800694828006948280069482800694828006948280069482800030000000037B50E1F0000000020000000000000230527B5102418090A12031527000000280000000000002C0000000000002F000000000000
`
driver apator08 is still not usable
`[15:38:07] INFO:  Adding meter: GardeningWater ...
[15:38:07] INFO:   File /config/wmbusmeters/etc/mqtt_discovery/apator08.json not found.`

BIBOLV, 2023-06-07:
Hi,
For meter template creation in HA we relay on community - if someone is using specific meter, then he can create template, test, create PR and afterwards look after it. [There](https://github.com/wmbusmeters/wmbusmeters-ha-addon/tree/main/wmbusmeters-ha-addon-edge/mqtt_discovery) is a list of current meter templates, so if you are up for it, you can create template, test and then create PR, if not - you can still use manual integration/configuration described [here](https://wmbusmeters.github.io/wmbusmeters-wiki/HASS.html).

filipsio, 2023-06-07:
For anybody having that problem in a future as well. Copy/paste apator162.yaml in a mqtt discovery directory to as apator08.yaml. Eventually change content about meter version etc. Volume is parsed correctly. 
I answered myself ...  :)

BIBOLV, 2023-06-08:
Cool, you are welcome to create PR so other would be able to use it.