wmbusmeters/wmbusmeters #979: How to debug snap

corny, 2023-06-15:
### Type of request

Question

### wmbusmeters version

1.13.1

### Your message goes here

I'm unable to configure wmbusmasters to publish received telegrams to MQTT. I figured out that CLI arguments to `snap run wmbusmeters` are not passed to the process. Adding `loglevel=debug` to the config file does not help. How do I increase the debugging output then?

My config file at /var/snap/wmbusmeters/common/etc/wmbusmeters.conf
```
device=auto:t1
loglevel=debug
logfile=syslog
logtelegrams=true
format=json
shell=/usr/bin/mosquitto_pub -h localhost -t wmbusmeters/$METER_ID -m "$METER_JSON"
```

Then I restart the snap with `snap restart wmbusmeters` and I watch the journal `journalctl -fu snap.wmbusmeters\*`

corny, 2023-06-15:
Is it possible that lines in the configuration file starting with `#` break the config?

BIBOLV, 2023-06-16:
`#` should not break anything, but you can try without them. 
After restart checked status of service - is it started without errors?
Change `logfile` to original config - syslog IMO will not work in snap.

closed 2023-09-15 by github-actions[bot]