wmbusmeters/wmbusmeters #981: Can not access ttyUSB0

corny, 2023-06-16:
### Type of request

Bug

### wmbusmeters version

1.13.1

### Your message goes here

If I start wmbusmeters via snapd then it can not access `/dev/ttyUSB0`. If build the binary and run it as root everything is working fine. What is missing here?

```
Jun 16 09:11:52 ha wmbusmeters[6851]: (main) device /dev/ttyUSB0 not currently used, detect contents...
Jun 16 09:11:52 ha wmbusmeters[6851]: (serial) check if /dev/ttyUSB0 can be accessed
Jun 16 09:11:52 ha wmbusmeters[6851]: (serial) you do not have the correct permissions to open the tty /dev/ttyUSB0 and you do not share the same access group.
```

```
$ snap connections
Interface       Plug                        Slot             Notes
network         wmbusmeters:network         :network         -
network-bind    wmbusmeters:network-bind    :network-bind    -
raw-usb         wmbusmeters:raw-usb         :raw-usb         manual
system-observe  wmbusmeters:system-observe  :system-observe  manual
```

```
$ ls -la /dev/ttyUSB0 
crwxrwxrwx 1 root dialout 188, 0 16. Jun 09:12 /dev/ttyUSB0
```

BIBOLV, 2023-06-16:
Do you use `device=auto:t1` or something else as device config?

corny, 2023-06-16:
I use `device=im871a:t1` as device config.

BIBOLV, 2023-06-17:
What OS do you use?

corny, 2023-06-18:
I use Raspberry Pi OS based on Debian 11.

closed 2023-09-18 by github-actions[bot]