wmbusmeters/wmbusmeters #983: New driver

pawlosck, 2023-06-20:
### Type of request

New device support

### Meter description

Probably, this is this model
https://plentia.es/wp-content/uploads/2015/06/PLENTIA_GESTION_&_TECHEM_Contador_de_Agua_web.pdf

(meter) MyWater: meter detection could not find driver for id: XXXXXXXX mfct: (MAD) Maddalena, Italy (0x3424) type: Warm Water (30°C-90°C) meter (0x06) ver: 0xfe
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) newly created meter (MyWater XXXXXXXX auto) did not handle telegram!

=================

(meter) MyWater: meter detection could not find driver for id: XXXXXXXX mfct: (MAD) Maddalena, Italy (0x3424) type: Water meter (0x07) ver: 0xfe
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) newly created meter (MyWater XXXXXXXX auto) did not handle telegram!

### Logged telegrams from meter in decrypted format

```shell
How to get this info?
I am running this command

mbusmeters --format=json rtlwmbus:868.95M:t1,c1 MyWater auto '*' NOKEY

I ran command
rtl_sdr -f 868.95M -s 1.6e6 - | rtl_wmbus
to find more info, but I can't find my water meter
```

BIBOLV, 2023-06-20:
Add `--logtelegrams` to wmbusmeters cmd in order to log telegrams.

pawlosck, 2023-06-20:
There are a lof of lines started with "telegram=|".  Which is important?

/wmbusmeters --logtelegrams --format=json rtlwmbus:868.95M:t1,c1 MyWater auto '*' NOKEY

BIBOLV, 2023-06-21:
Replace `'*'` with meter ID to receive only telegrams from your meter.

weetmuts, 2023-06-21:
You need to post the telegram=|...| content here, otherwise we cannot add the meter. Thanks!

pawlosck, 2023-06-21:
Ok, I understand, but I don't know what to do to get this value. Maybe meter doesn't use official protocol? Is it possible to check it?

pawlosck, 2023-06-21:
I see other message. Maybe you can help me.

(wmbus) WARNING! no key to decrypt payload! Permanently ignoring telegrams from id: XXXXXXXX mfct: (TCH) Techem Service (0x5068) type: Radio converter (meter side) (0x37) ver: 0x95
(meter) newly created meter (MyWater YYYYYYYY mkradio4a) did not handle telegram!

BIBOLV, 2023-06-21:
Check your meter and verify that it has radio3 or radio4 marking on it - it would mean that radio module is installed on your meter. Then check your meter ID also on meter and configure wmbusmeters with that ID and wait for one full working day to check if you receive any messages from it.

pawlosck, 2023-06-21:
I found this issue https://github.com/wmbusmeters/wmbusmeters/issues/523
I have the same meter.
My meter uses mkradio4a driver and transmission is encrypted, so probably my problem can't be solved.

Received telegram from: XXXXXXXX
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Radio converter (meter side) (0x37) encrypted
                   ver: 0x95
      Concerning meter: YYYYYYYY
          manufacturer: (MAD) Maddalena, Italy (0x3424)
                  type: Warm Water (30°C-90°C) meter (0x06) encrypted
                   ver: 0xfe
                device: rtlwmbus[00000001]
                  rssi: 146 dBm
                driver: mkradio4a

closed 2023-09-20 by github-actions[bot]